emsesp/EMS-ESP #517: Climate error in HA - config issue or missing data at start-up?

glitter-ball, 2020-09-24:
**Question**
I've been getting a persistent error in the HA log:

`2020-09-21 21:14:50 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from `
[image]

That's all I get - rest is truncated, sadly. It seems to produce a burst (16 this time) and then stop. I've had a look in my climate.yaml file and am struggling to see what the issue is, if there is one. Alternatively, is this an issue with HA looking for data which hasn't come in over MQTT yet? When I start MQTT Explorer I can wait a while for thermostat data - e.g. just checked and first thermostat telegram came in 34 seconds after the first boiler one.

HA tiles look OK...
[image]

**Device information**
```
/config/climate.yaml

- platform: mqtt
  name: Heating
  modes:
    - "auto"
    - "heat"
    - "off"
  mode_command_topic: "ems-esp/thermostat_cmd_mode1"
  temperature_command_topic: "ems-esp/thermostat_cmd_temp1"
   
  mode_state_topic: "ems-esp/thermostat_data1"
  current_temperature_topic: "ems-esp/thermostat_data1"
  temperature_state_topic: "ems-esp/thermostat_data1"

  mode_state_template: "{{ value_json.mode }}"
  current_temperature_template: "{{ value_json.currtemp }}"
  temperature_state_template: "{{ value_json.seltemp }}"
   
  temp_step: 0.5
  
- platform: mqtt
  name: Hot Water
  modes:
    - "auto"
    - "off"
  min_temp: 40
  max_temp: 60
  temp_step: 1
  
  current_temperature_topic: "ems-esp/boiler_data"
  temperature_state_topic: "ems-esp/boiler_data"
  mode_state_topic: "ems-esp/boiler_data"

  current_temperature_template: "{{ value_json.wWCurTmp }}"
  temperature_state_template: "{{ value_json.wWSelTemp }}"
  mode_state_template: "{% if value_json.wWActivated == 'off' %} off {% else %} auto {% endif %}"

  temperature_command_topic: "ems-esp/boiler_cmd_wwtemp"
  mode_command_topic: "ems-esp/boiler_cmd_wwactivated"
  

```

proddy, 2020-09-24:
yes, its because the data hasn't come over in the MQTT yet. Two options: set the MQTT Retain to true to persist the data, or switch the MQTT Format over to HA. Then you don't need ever to touch any .yaml files.

glitter-ball, 2020-09-25:
OK - yep, keep the dog happy.

I do need to get to grips at some point with switching to HA format. Not 100% sure what that entails doing in HA. I seemed to end up hand-crafting a lot of .yaml files to get mine workind and then use the UI to tart it up on screen.

As long as I take a full snapshot before I start, I guess I can always roll back.

Keep up the good work and sense of humour - I'm not sure I'd always have your patience with the volume of stuff to get through!

proddy, 2020-09-26:
the climate card in HA needs the current temperature, the setpoint temperature and mode to render. This data comes in on different telegrams at different times so this could cause the issues. You could leave MQTTExplorer on for a while, select the thermostat_data topic and look at the history pane to see which fields are populated and when.

in MQTT Discovery (MQTT Format=Home Assistant) you shouldn't be getting these errors any more. 

I would wait until #288 is done and then move over from Single to HA in one go. "If it ain't broke, don't touch it". And knowing me it will take a few attempts to get it right.

glitter-ball, 2020-09-26:
Thanks. I like the idea of getting #288 sorted first. As you say, it all works for now, pretty much.

From my own experiments, I know the thermostat can take up to 30+ seconds to transmit after an ems-esp restart and the boiler gets in there much quicker every time. With it all set to publish on change (as mine is) you can easily get 8-10 boiler telegrams before the first thermostat one. I just measured a 50-second wait for the the first thermostat telegram.

proddy, 2020-09-26:
> I know the thermostat can take up to 30+ seconds to transmit after an ems-esp restart and the boiler

that's not normal behaviour. As soon as any EMS device is detected it requests it's data and then subsequently every 10 seconds. When I boot my EMS-ESP I see the thermostat almost instantaneously, either via a `show` command in the Console or in the Web dashboard. If your not then we have another bug that needs squishing.

glitter-ball, 2020-09-26:
So... the instance I quote above was not associated with a restart - that was ‘business as usual’.

I wonder whether it’s linked to me setting publish in change rather than the fixed 10 seconds?

glitter-ball, 2020-09-26:
[image]

Here's an MQTT Explorer screenshot. In the LH half of the plots, MQTT was set to publish on change. In the RH half, all topics were set to publish every 10 seconds. The thermostat plot shows the biggest change. With publish on demand, it was updating every 60 seconds (default thermostat update interval?). With fixed publish interval, it's updating every 9-10 seconds or so.

proddy, 2020-09-30:
@glitter-ball , a bit lost now. Is it better or is something not working how it should?

glitter-ball, 2020-09-30:
Hi, I think some of this resolved with the work on #288 and a move to HA MQTT layout.

The only other issue is the trace above, which I'm trying to get my head around. With MQTT set to publish on change for the thermostat, I get thermostat updates once a minute (driven by the stat itself?) With MQTT set to publish every 10 seconds, it publishes every 10 seconds. This is probably doing 'what it says on the tin' (!) but it leaves me with two questions:

1) Set to publish on change, what triggers the refresh of thermostat data? Is it the thermostat itself broadcasting an unsolicited  telegram once a minute or is EMS-ESP polling it once a minute to produce that telegram?
2) Set to publish every 10 seconds, where does the thermostat data come from? Is MQTT publishing the same telegram 6 times over until the next 60-second update from the stat, or does EMS-ESP then poll the stat every 10 seconds and obtain fresh data from the thermostat for it to publish each time?

Thanks!

proddy, 2020-09-30:
(1) yes, the thermostat is sending out the data itself after every 60 seconds
(2) yes, it's the same message every 10 seconds. which I admit sounds a bit noddy.

Thinking about this perhaps the normal defacto behaviour is to fetch fresh data every n seconds and then always send when something has changed.

MichaelDvP, 2020-10-13:
@glitter-ball a bit more detailed:
- ems-esp fetches some values that are not periodically broadcasted every minute, i.e. settings from thermostat
- the thermostat broadcasts some telegrams periodically every minute, regardless of changes.
- the thermostat broadcasts telegrams immediately if a value has changed
- thermostat sends messages to other devices like boiler if necessary, ems-esp catches these messages
- ems-esp checks all values and publishes if one value has changed. 
- once a minute the thermostat clock changes, that's why you always get a mqtt message every minute with on-change setting
- publish every 10 sec will publish regardless there is a message from thermostat, it will not query the thermostat every 10 sec.

glitter-ball, 2020-10-13:
@MichaelDvP - thanks. That’s useful information.

proddy, 2020-10-16:
@MichaelDvP that is useful. Think we should add it to the documentation somewhere.

proddy, 2020-10-26:
closing this one.

closed 2020-10-26 by proddy