emsesp/EMS-ESP #518: add circulation pump mode commands for boiler and thermostat

MichaelDvP, 2020-09-25:
As discussed on gitter with @inter-faze there are some settings of the ww circulation pump not implemented as command.
I'd like to add commands to boiler for the values that are already fetched:
`wwcircpump` on|off
`wwcircmode` 1..7

and new to thermostat:
fetch value for `wwCircMode_` off|on|auto
command `wwcircmode` off|on|auto

This value and also `wwmode_` is located in type_id 0x37. This message is registered for read only for RC30/35, 
but the `wwmode` command is registered to all thermostats, that is not consistent.

The question is, which thermostats works with type_id 0x37? 
RC30/35 is clear, for RC20 i checked the manual, there isn't such a setting. EMS+ may have another type_id, but i find nothing in Norberts list? 
What about Junkers and other thermostats?

Can everyone please check in terminal with `read 10 37`, or `read 17 37` and give me the output and the thermostat type.

cyniu88, 2020-09-25:
in my case the wwcirculation command does not work on ems-esp but on RC300 it works fine.

[wwcirculation_from_RC300.txt](https://github.com/proddy/EMS-ESP/files/5282017/wwcirculation_from_RC300.txt)
[wwcirculation_from_terminal.txt](https://github.com/proddy/EMS-ESP/files/5282018/wwcirculation_from_terminal.txt)

proddy, 2020-09-25:
not supported on RC20 `[emsesp] Thermostat(0x17) -> (0x0B), (0x37), data: <empty>`

MichaelDvP, 2020-09-25:
@proddy Thanks, as expected, only the boiler commands will work.

@cyniu88  Thanks, also as expected the type-id 0x37 give:
```
iDom-client/buderus/ems-esp:/# read 10 37
001+14:38:55.688 N 1: [emsesp] Thermostat(0x10) -> (0x0B), (0x37), data: <empty>
```
for RC300 there are some ems+ messages. In RC300 have you set the configuration of circulation pump to off? 
It switches to `Warm Water circulation pump available: off` that is the boiler command `001+14:39:50.574 D 22: [emsesp] Received UBAParameterWW`, this is what i'll add to boiler-commands.  
But RC300 have in menu settings for circulation off/on/program as ww/own program, this setting i'm searching for. 
Also the ww has a setting in RC300 for mode (Betriebsart) off/on/program as hc/own program. 
Can you set `watch on 10` and change these mode settings (ww and circulation mode) on the RC300. I think it's one of this messages:
```
001+14:39:51.967 D 25: [emsesp] No telegram type handler found for ID 0x309 (src 0x10)
001+14:39:52.159 D 26: [emsesp] No telegram type handler found for ID 0x2F5 (src 0x10)
001+14:39:52.580 D 27: [emsesp] Received RC300WWmode
```

lmdecroos, 2020-09-25:
My master thermostat is a RC35 and i have a RC20RF for my second HC
EMS-ESP version 2.0.1 

ems-esp:/# read 10 37
009+23:50:01.249 N 6: [emsesp] Thermostat(0x10) -> (0x0B), (0x37), data: FF 00 02 02 00 01 01 00 3C FF
ems-esp:/# read 17 37
009+23:50:24.123 N 7: [emsesp] Thermostat(0x10) -> (0x0B), (0x37), data: FF 00 02 02 00 01 01 00 3C FF
009+23:50:27.625 E 8: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
ems-esp:/# 
ems-esp:/# read 10 37
009+23:50:45.873 N 9: [emsesp] Thermostat(0x10) -> (0x0B), (0x37), data: FF 00 02 02 00 01 01 00 3C FF
ems-esp:/# read 17 37
009+23:51:05.383 E 10: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
009+23:51:23.997 N 11: [emsesp] Thermostat(0x10) -> (0x0B), (0x37), data: FF 00 02 02 00 01 01 00 3C FF

the read 17 37 returns always a error, but returns also the data from RC35 master thermostat 

ems-esp:/# show devices
These EMS devices are currently active:

Boiler: Topline/GB162 (DeviceID:0x08, ProductID:115, Version:05.05)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE9 

Thermostat: RFM20 Remote (DeviceID:0x18, ProductID:94, Version:02.01)
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 

Thermostat: RC20RF (DeviceID:0x19, ProductID:93, Version:02.00)
 This Thermostat will respond to telegram type IDs: 0xAF 

Thermostat: RFM20 Remote (DeviceID:0x1A, ProductID:94, Version:02.01)

Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:21.08) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x3E 0x3D 0x48 0x47 0x52 0x51 0x5C 0x5B 0xA5 0x37 

Mixing Module: MM10 (DeviceID:0x21, ProductID:69, Version:02.01)
 This Mixing Module will respond to telegram type IDs: 0xAA 0xAB 0xAC 

Switching Module: WM10 (DeviceID:0x11, ProductID:71, Version:02.00)

Controller: BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)

glitter-ball, 2020-09-25:
I have a Sense II

```
ems-esp:/# read 10 37
001+01:57:47.972 N 1: [emsesp] Thermostat(0x10) -> (0x0B), (0x37), data: <empty>
ems-esp:/# read 17 37
001+01:58:02.186 E 2: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
ems-esp:/# read 10 37
001+01:58:52.275 N 3: [emsesp] Thermostat(0x10) -> (0x0B), (0x37), data: <empty>
ems-esp:/# read 17 37
001+01:58:58.309 E 4: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
ems-esp:/# show devices
These EMS devices are currently active:

Boiler: Worcester Logamax Plus/GB192/Condens GC9000 (DeviceID:0x08 ProductID:208, Version:01.01)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA

Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:33.03) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x2A5 0x2B9 0x2A6 0x2BA 0x2A7 0x2BB 0x2A8 0x2BC 0x31D 0x31E

Controller: ErP (DeviceID:0x09, ProductID:209, Version:01.03)

ems-esp:/#

```

Extracts from Sense II manual:
The controller has time programs:
– Heating systems: For each heating circuit, 2 time
programs with 6 switching times per day can be set
(exception: for one constant heating circuit there is
only 1 time program).
– Domestic hot water: A time program for DHW heating
and a time program for the DHW circulation pump for
each hot water circuit, each with 6 switching times per
day.

[image]
[image]
[image]

cyniu88, 2020-09-25:
@MichaelDvP  I can collect logs for you, but please describe all the steps I need to take

MichaelDvP, 2020-09-26:
@cyniu88 
The command i'm lookin for is in the picture of @glitter-ball this one:
[image]
and there is a similar command for DHW (off/on/as_hc/own_program).
In terminal type `watch on 10` to get all messages from the thermostat,
than go to your thermostat and change the settings for DHW mode and circulation mode. 
If your standard setting for circulation mode is "as_DHW_system", change to off, wait a few seconds, change to on, wait a few seconds, change back to "as_DHW_system".  Than change the settings for DHW mode in the same manner. 
(the own_program setting will ask to enter a program if you don't have it, you can skip the setting).
Than post the telnet log and tell me from which mode you started to change. Hopefully we can identify the type-id and position of the setting for the RC300.

@lmdecroos 
Thanks, i have also RC35 and use the simulated roomcontrol for hc2 (set as RC20 on device-id 19). If i test on my system it should also work on yours.

@glitter-ball 
The Sense II works with the same ems+ as RC300, With the log from @cyniu88 it should work. The settings for circulation installed and circulation frequency are boiler settings mentioned in the OP, they are read already  and i add the commands.

glitter-ball, 2020-09-26:
> @glitter-ball
> The Sense II works with the same ems+ as RC300, With the log from @cyniu88 it should work. The settings for circulation >installed and circulation frequency are boiler settings mentioned in the OP, they are read already and i add the commands.

OK. My boiler doesn't support a secondary circulation pump (it's only the thermostat that mentions it) although it has connections on the controller marked "unused".

lsw271, 2020-09-29:
@MichaelDvP 
I havent got circulation pump in my setup but I checked telegrams.

I can't read by command `read 10 37`
[image]

I tried to check by `watch on `

turn on circulation pump on thermostat and after few seconds turn off
[image]

turn on circulation pump on boiler:
[image]

`wwmode `what it's used for? its a ECO / HOT mode of warm water? If not, what should I do to show you telegrams to this command?

lsw271, 2020-10-01:
Ok, I found wwmode in thermostat.

Setting to OFF
[image]
[image]

setting to lower
[image]

setting to high
[image]

setting to "like hc1 schedule"
[image]

setting to "own schedule"
[image]

my own schedule:
[image]

glitter-ball, 2020-10-02:
@lsw271 - thanks for doing this. I have a Sense II which looks like a CW400 so I have these same settings for hot water.

MichaelDvP, 2020-10-02:
@lsw271: perfect catch again. The mode in thermostat is set in the undocumented messagetype 0x02F5 (offset 2) with `off/on/auto/own` =`1/2/3/4`. The new temperature is send to the boiler in known flags-messsage 0x35 (offset 3). 

I'll add this to the thermostat, but command-numbering for ems-esp will to be consistent with other thermostats and on/off commands (bool rendering): `0/1/2/3` or `off/on/auto/own` or `false/true/auto/own`.

can you please also do a `read 10 2F5` on the terminal to see the full message.

lsw271, 2020-10-02:
@MichaelDvP 
I've got five modes
1. Off 
2. Lower
3. High
4. Like hc1 schedule [own?]
5. Own schedule [own?] 
four and five its the same?

I tried to change modes on thermostat but in telnet was only one telegram. 
ems-esp:/# read 10 2F5
000+08:30:50.384 N 1: [emsesp] Thermostat(0x10) -> (0x0B), (0x2F5), data: 01 FF 04 00 00 00 08 01 00 08 02 00 00 00 00 00 00 00 00 00

MichaelDvP, 2020-10-03:
Good, with "0=off" it's a lot easier. 4 and 5 can be the same program, but also can be different if you choose a schedule for heating and a different one for water. 
In this telegram are a lot of zeros, I think we stay with the mode on pos 2 and analyse the other parameters on demand.
If you want to change the mode you can test with `send telegram "0B 10 FF 02 01 F5 01"` to set to low,  `send telegram "0B 10 FF 02 01 F5 02"` for high, `send telegram "0B 10 FF 02 01 F5 03"` for auto(hc1). 
I've added the command to ems-esp, but, because of some other changes, send it proddy as pull request.

proddy, 2020-10-03:
@MichaelDvP - thanks. working on another big chance that's going to introduce a lot more bugs I think. I'll merge your changes in manually and upload shortly.

lsw271, 2020-10-03:
> Good, with "0=off" it's a lot easier. 4 and 5 can be the same program, but also can be different if you choose a schedule for heating and a different one for water.
> In this telegram are a lot of [...]

Ok, everything works fine. I tried also `send telegram "0B 10 FF 02 01 F5 00"` (off) and  `send telegram "0B 10 FF 02 01 F5 03"`(own - fith on my list)

[image]

@MichaelDvP  could you also add command for off and own ?

hetii, 2020-10-12:
@lsw271 Hi, If I may ask where did you connect your circular pump?
I'm not sure if this is right direction but my plan is to add some triacs to esp-ems and control it by esp itself.

inter-faze, 2020-10-13:
> @lsw271 Hi, If I may ask where did you connect your circular pump?
> I'm not sure if this is right direction but my plan is to add some triacs to esp-ems and control it by esp itself.

This had been my thought as well, but @MichaelDvP posted a good configuration and way how to just start the circulationPump via ems.esp. I cannot find it in the Gitter chat. On my configuration this works really well.

MichaelDvP, 2020-10-13:
@hetii it's directly connected to the MC10:
[image]

But you are free to make your own control. On ems-esp you can control some gpios via mqtt, but some of the cheap wemos-relais have much to little insulation-distance. Be carefull.

hetii, 2020-10-13:
@MichaelDvP  Thx for information.
I don't have any extra devices beside my junkers, so so far I desing such simple board.
Still in work in progress, but at the end should handle 3 additional devices.
Also as I'm not sure how much current can draw from ems bus or service port I add there own psu.
[image]

MichaelDvP, 2020-10-15:
@cyniu88 you wrote circulation setting works from your thermostat, but not from ems-esp. If i should add the setting to ems-esp you have to provide me the commands (as lsw271 for ww, but he have no circulation pump): 
In terminal use `watch on 10` to see the messages from thermostat and than change the circulation modes on the thermostat.

MichaelDvP, 2020-10-16:
Ok, boiler and RC35 is fully added, RC300 ww modes and temperatures also. RC300 circulation and commands for other thermostats are not requested, i close this.

closed 2020-10-16 by MichaelDvP