emsesp/EMS-ESP #519: Create a derived sensor value in HA

glitter-ball, 2020-09-26:
**Question**
Apologies - this is a bit off-topic, more a HA question than an EMS-ESP one. But it might be useful to other people too.

How can I create a new sensor value in HA from two existing values? Specifically, if I want to create a new value...

`differential = flow_temp - return_temp`

... how would I code that in the sensor.yaml file? Existing entries are:

```
- platform: mqtt
  state_topic: 'ems-esp/sensor_28-BE2A-7791-0402-65'
  name: 'Return temp'
  unit_of_measurement: '°C'
  value_template: '{{ value_json.temp | round(1) }}'

- platform: mqtt
  state_topic: 'ems-esp/boiler_data'
  name: 'Flow temperature'
  unit_of_measurement: '°C'
  value_template: '{{ value_json.curFlowTemp }}'

```

Thanks!

proddy, 2020-09-26:
from the top of my head you would do something like

```yaml
- platform: template
  sensors:
    differential:
      value_template: "{{ (states('sensor.flow_temperature') | float) - (states('sensor.return_temp') | float) }}"
```

You can try the formula directly in HA by going to the Developer Tools->Template and copying in the {{ ... }} bit

glitter-ball, 2020-09-26:
Thanks - this seems to work in the Developer Tools preview. I want to round to one decimal place, so tried inserting a `round(1)` filter in several places  (with extra brackets as necessary) to achieve this but without luck. Any idea how to crack, that?

glitter-ball, 2020-09-26:
The issue is something to do with the subtraction operator. If I just put each reading with the `float() `filter into the template, they both report back values in the form `xx.x`. This was a bit of a suprise as the return_temp has a `round(1)` in it because the Dallas sensor values aren't necessarily neat numbers. Anyway, on their own they're OK.

Putting the subtraction operator in produces ridiculous numbers of decimal places. Adding and multiplying also. Maybe I'm missing something with floating point maths?

glitter-ball, 2020-09-26:
Great - cracked it! Thanks for your help.

[image]

proddy, 2020-09-26:
Fancy pants!

glitter-ball, 2020-09-26:
Ha! Yep - more ‘saddo numbers‘ to look at and for the rest of the family to take the p*ss out of!

But... my son is learning Python now at school and getting bored with drawing houses with the turtle, so he might have to play cool and learn some HA stuff too. *8()

glitter-ball, 2020-09-26:
For the curious, this is the complete config.yaml block:

```
- platform: template
  sensors:
    differential:
      friendly_name: "Flow-Ret diff"
      unit_of_measurement: '°C'
      icon_template: 'mdi:format-align-middle'
      value_template: "{{ (states('sensor.flow_temperature') | float - states('sensor.return_temp') | float) | round(1) }}"
```

proddy, 2020-09-30:
I've added to the doc. thanks

closed 2020-09-30 by proddy

mupsje, 2020-10-03:
Cool,
Delta T........

Just what I was missing

glitter-ball, 2020-10-03:
Yep. Had been on my to-do list for a while *8) Loads of fun...
[image]