emsesp/EMS-ESP #520: Temperature offset adjustment for additional sensors

Domoberry, 2020-09-28:
**Is your feature request related to a problem? Please describe.**
I have added three temperature sensors and noted a relatively large difference in temperature measured while all sensors measure the same temperature. In my case, the relative delta was 1.0 degrees.

**Describe the solution you'd like**
Having the ability to statically adjust the value measured with a positive or negative number would allow me to make better comparing measurements by aligning the relative values. If a calibrated measurement would be available, I could also correct for absolute values. To keep it simple, such adjustment can be independent of the temperature.

**Describe alternatives you've considered**
Use better specified  or selected temperature sensors

**Additional context**
Practical when you want to measure temperature differences. 
The UI could for example be added to the Dashboard screen as a non breaking change when the creation defaults to 0.

Thanks for EMS-ESP !

bbqkees, 2020-09-28:
It could be a useful enhancement.
Those DS18B20 sensors are often fake and have wild offsets up to 2 degrees.

proddy, 2020-09-28:
yes, good idea. It's been on my wish-list for a while. It would have to be one single value for all sensors though.

@MichaelDvP  ?

MichaelDvP, 2020-09-28:
For a proper adjustment it should have per sensor an offset (at 0 °C) and a slope. I think it's better to do this in the home automation system, where you know what the sensor is for.  And there you can also do a nonlinear calibration. The ds-datasheet shows a nonlinear error over temperature.
But also possible in ems-esp. The values have to match a sensor by sensor id and store values for >10 sensors. If a sensor is disconnected remember the value. If a new sensor is connected  add it until memory count is full, than overwrite values from  a old disconnected sensor.

Domoberry, 2020-09-28:
Agree that the offset and accuracy for a DS18B20 are depending on the temperature. My logic was that in a typical use case one would use the sensor around a limited range and that getting  comparable values for starters in that range would be nice. I had in this practical case:
Sensor 1 / 20.0 / 0 (taking this one as '0' is not based on facts)
Sensor 2 / 20.7 / +0.7
Sensor 3 / 19.7 / -0.3
Are these fake sensors? Could be as the source country has a reputation :-)
According to the specification of the device, offset and (in)accuracy are notable, yet OK given their price.
Agree that for (real) fine-tuning depending on its intended use, the HA system is probably best

glitter-ball, 2020-09-28:
If you were prepared to tolerate doing the corrections in HA, you could presumably impelment this right now with a template definition much like @proddy helped me knock up over this last weekend for flow-return differential calculation? That doesn't help non-HA users and leaves the dodgy values in the EMS-ESP GUI so there's still an argument for doing it there, albeit requiring more work,

Taking it to the extreme, you could have a calibration routine where you designate one of a pair of sensors the 'standard' and then heat both up over the temperature range of interest over a few minutes. Compare the output from both and get EMS-ESP to calculate the slope and offset necessary to get the second one to best-fit match the standard one. Alternatively, get some square paper and a pencil... *8)

MichaelDvP, 2020-09-29:
> Are these fake sensors? 

I don't think so, from the maxim datasheet:
```
Thermometer Error t ERR
-10°C to +85°C 	     ±0.5°C 
-30°C to +100°C      ±1°C
-55°C to +125°C      ±2°C
```

Domoberry, 2020-09-29:
Application note 208 from Maxim [https://www.maximintegrated.com/en/design/technical-documents/app-notes/2/208.html](url) talks about such compensation, using indeed a 2nd order approach based on an offset and a correction component based on the temperature at hand. Yet, looks like overkill to me, assuming the typical limited temperature range in which we use these sensors, just an average offset correction for that range is already great! Also, if such 'deluxe' correction is available, a typical user will probably find it difficult to determine the actual 2nd order parameters. In any case, I appreciate the discussion!  
ps: seems that "HA" refers to a specific home automation system. I'm using Domoticz, where one can also use scripts to optimise incoming data.

closed 2020-10-09 by proddy