emsesp/EMS-ESP #522: "Tx Write operation failed" when triggering "wwonetime"

toblum, 2020-09-28:
Dear ems-esp team,

I hope you can help me with this issue:

**Question**
I have ems-esp v2, now v2.0.1 working in read mode for some weeks without issues, but now I'm trying to use TX mode. I want to trigger the warm water load by sending {"cmd":"wwonetime","data":"on"} to topic "ems-esp/boiler_cmd" via MQTT. Nothing happens, but I can see the message "Last Tx Write operation failed after 3 retries" like in the logs below.

Tx mode is set to 1. I'm using the "EMS BUS (PI HAT) interface board v1.0", directly connected to an ESP8266 D1 mini (GPIO 13/15, D7/D8, RX/TX). Boiler is Buderus as in screenshot below, connected via service plug.

Let me know if I missed some relevant informations. Surely I did.

**Screenshots**
[image]
[image]

**Device information**
This is what I see in log debug:
ems-esp:/$ [K012+23:16:45.029 D 152: [emsesp] Received UBAMonitorWW
ems-esp:/$ [K012+23:16:45.336 D 153: [emsesp] Received UBAMonitorFast
ems-esp:/$ [K012+23:16:48.117 D 154: [emsesp] No telegram type handler found for ID 0xA3 (src 0x10)
ems-esp:/$ [K012+23:16:55.068 D 155: [emsesp] Received UBAMonitorSlow
ems-esp:/$ [K012+23:16:55.325 D 156: [emsesp] Received UBAMaintenanceStatus
ems-esp:/$ [K012+23:16:55.530 D 157: [emsesp] Received UBAMonitorWW
ems-esp:/$ [K012+23:16:56.061 D 158: [emsesp] Received UBAMonitorFast
ems-esp:/$ [K012+23:16:57.440 I 159: [boiler] Setting boiler warm water OneTime loading on
ems-esp:/$ [K012+23:16:57.572 D 160: [telegram] Sending write Tx [#242], telegram: 0B FF 35 00 22 F1
ems-esp:/$ [K012+23:16:58.020 D 161: [telegram] Sending write Tx [#243], telegram: 0B FF 35 00 22 F1
ems-esp:/$ [K012+23:16:58.444 D 162: [telegram] Sending write Tx [#244], telegram: 0B FF 35 00 22 F1
ems-esp:/$ [K012+23:16:59.140 D 163: [telegram] Sending write Tx [#245], telegram: 0B FF 35 00 22 F1
**ems-esp:/$ [K012+23:16:59.388 E 164: [telegram] Last Tx Write operation failed after 3 retries. Ignoring request.**
ems-esp:/$ [K012+23:17:04.964 D 165: [emsesp] Received UBAMonitorWW
ems-esp:/$ [K012+23:17:05.245 D 166: [emsesp] Received UBAMonitorFast
ems-esp:/$ [K012+23:17:12.227 D 167: [emsesp] No telegram type handler found for ID 0x06 (src 0x10)
ems-esp:/$ [K012+23:17:14.600 D 168: [mqtt] Publishing topic ems-esp/boiler_data (#24889, attempt #1, pid 1)
ems-esp:/$ [K012+23:17:14.917 D 169: [emsesp] Received UBAMonitorWW
ems-esp:/$ [K012+23:17:15.151 D 170: [emsesp] Received UBAMonitorFast

This is what I can see in watch mode:
ems-esp:/$ [K012+23:11:21.266 N 141: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 FF 37 FB 00 19 FF 02 46 00 FF
ems-esp:/$ [K012+23:11:21.486 N 142: [emsesp] Boiler(0x08) -> (0x0B), (0x14), data: 58 C1 69
ems-esp:/$ [K012+23:11:21.769 N 143: [emsesp] Boiler(0x08) -> (0x0B), (0x16), data: FF 5A 3C 00 06 FA 0A 01 2D 5B 1E 01
ems-esp:/$ [K012+23:11:22.129 N 144: [emsesp] Thermostat(0x10) -> (0x0B), (0x3E), data: 80 02 30 7D 00 00 00 00 00 00 00 00 64 00 31 00
ems-esp:/$ [K012+23:11:22.532 N 145: [emsesp] Thermostat(0x10) -> (0x0B), (0x3D), data: 01 28 30 20 00 28 00 01 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 15 05 05 01 02
ems-esp:/$ [K012+23:11:22.907 N 146: [emsesp] Thermostat(0x10) -> (0x0B), (0xA5), data: 00 00 00 00 FF F6 01 06 00 01 00 00 00 FF FF 00 00 05 1E 05 28 FF 00 00 00
ems-esp:/$ [K012+23:11:23.368 N 147: [emsesp] Thermostat(0x10) -> (0x0B), (0x37), data: FF 00 02 02 FF 05 09 00 3C FF
**ems-esp:/$ [K012+23:11:25.157 E 148: [telegram] Last Tx Write operation failed after 3 retries. Ignoring request.**
ems-esp:/$ [K012+23:11:26.377 N 149: [emsesp] Boiler(0x08) -> (0x00), (0x34), data: 0A 01 AA 01 AA 00 00 00 03 00 00 90 C3 00 0A DE 00
ems-esp:/$ [K012+23:11:26.585 N 150: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 31 01 BC 3C 00 01 01 20 62 80 00 01 AA 01 AA 00 00 12 30 59 00 CC 00 00 00
ems-esp:/$ [K012+23:11:29.376 N 151: [emsesp] Thermostat(0x10) -> (0x00), (0x3E), data: 80 02 30 7D 00 00 00 00 00 00 00 00 64 00 31 00

**Additional context**

lsw271, 2020-09-28:
Hi, maybe try like that: 
{"cmd":"wwonetime" ,"data":1}
1=on
2=off

MichaelDvP, 2020-09-29:
Yes there is something wrong, the command is addressed to 0xFF instead of the boiler 0x08. 0xFF is not a valid device, there is no acknowledge, so it's count as tx-fail. The tx-mode setting is ok.

Can you please test with the latest beta, maybe it's fixed already due to the command changes.

proddy, 2020-09-29:
Also in console do a `show devices`

toblum, 2020-09-29:
Hi guys,

thank you for your quick resposnse.

"show devices" returns:
```
ems-esp:/$ show devices
These EMS devices are currently active:

Boiler: Topline/GB162 (DeviceID:0x08, ProductID:115, Version:03.06)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE9

Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:01.08) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x3E 0x3D 0x48 0x47 0x52 0x51 0x5C 0x5B 0xA5 0x37

Controller: BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)
```

I then updated using "EMS-ESP-2_1_0b2-esp8266.bin". 

"show devices" returns now (note the additional 0xEA):
```
These EMS devices are currently active:

Boiler: Topline/GB162 (DeviceID:0x08, ProductID:115, Version:03.06)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA

Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:01.08) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x3E 0x3D 0x48 0x47 0x52 0x51 0x5C 0x5B 0xA5 0x37

Controller: BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)
```

I tried again MQTT: {"cmd":"wwonetime","data":"on"}, but now to ems-esp/boiler
And it seems that it works now as expected. Warm water heats up. Also setting of temperatures is working. That also didn't work before.

So I'm happy guys. Waiting for the final 2.1.
Thank you and keep up with the good work!

Best regards
Tobias

closed 2020-09-29 by toblum