emsesp/EMS-ESP #523: setting wwtemp by telnet not working (wrong mqtt topic?) CW400, Junkers Cerapur

lsw271, 2020-09-28:
**Bug description**
I tried to change wwtemp by telnet, I saw in the console "Setting boiler warm water temperature to 51 C" but temperature returns to previous value. I also saw in mqtt explorer (sending by telnet) topic home/boiler/boiler_wwtemp (see screenshot below) but I think it should be ems-esp/boiler(...)

**Steps to reproduce**
call command by the console "call wwtemp 51"  
comfort mode was on ECO, previous value was 54

**Expected behavior**
posibility to change wwtemp by telnet later by HA :) 

**Screenshots**
[image]

i tried few times with different temp, temp always return to 54 (red circles)
[image]

[image]

**Device information**
Boiler: Junkers GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.02)
mine is Junkers CerapurAcu Smart
Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:33.04) Clock: 21:25:08 25/09/2020 and CW400
EMS-ESP version 2.0.1

**Additional context**
Its normal that the wwtemp change temperature in wWSetTemp (warm water temp set on boiler) and wWSelTemp(warm water temp set on thermostat)? I thought this would change one of them, but which have priority?

MichaelDvP, 2020-09-29:
@lsw271 `wWSetTemp` and `wWSelTemp`  are both on the boiler, `wWSelTemp` is the setting in UBAParametersWW which can be written, `wWSetTemp` is the setpoint the boiler reported back in UBAMonitorWW-message and can not be written by us.
The log reports, that the value is written and readback, but after the publish there is another UBAParameterWW message not from us. Something sets the value back. Can you do the same with `watch on 33` to see the full message.

In my system (GB125, MC10, RC35) the boiler poti control on MC10 has to be set to auto to accept wwtemp settings, If this control is set to fixed value, any setting is rewritten by the controller 0x09. Maybe there is something similar on your system.

mglatz, 2020-09-29:
I also have Junkers (Cerapur comfort in my case) and my findings so far are 
`wwSelTemp` is read only and reads the temerature that is set on boiler with the adjustment wheel
`wwSetTemp` shows the same value (in my case, the temp is set to 50). I can change `wwSetTemp `with payload `{"cmd":"wwtemp" ,"data":40}`, but the value (data) must be lower than `wwSelTemp `is. So I can set 40, but I cant set 55. After some time (10 minutes or so), `wwSetTemp `will return to default value (equal to `wwSelTemp`). I have yet to explore if changing this has any effect on the boiler behavior whatsoever.

lsw271, 2020-09-29:
@MichaelDvP 
@mglatz 

ok, I forgot i've got another temperature for warm water on CW400, so it will be in my setup `wWSelTemp` 
i've got two options in CW400 
1. Warm Water temperature 
2. lower temperature at night to reduce costs of gas (i'dont know how to tell it in english) 

it is for the hot water timetable (hours/ days) which i've got in CW400

I will check it one more time tonight

lsw271, 2020-09-29:
@MichaelDvP @mglatz 
I am after some testing

Values changed right, maybe dev version 2.1.0b2 fixed it

my changes by "call boiler wwtemp X"
49 > 47 --telnet
47 > 43 --telnet
43 > 45 --thermostat 
45 > 44 --thermostat

and telegrams
[image]

In mqtt explorer I saw right values, **but it was wrong in thermostat**

Trying to catch telegram when i changed warm water temp on thermostat (firstly 47 degrees secondly 49)
[image]

Trying to catch telegram when i changed warm water temp (lower temperature at night) // (firstly 37 degrees secondly 33)
[image]

I think i caught by chance telegram which shoud means my thermostat change wwtemp (wWSetTemp) to lower temperature at night - changed by timetable. I've got for example between 5 am and 1 am 50 degrees on warm water, but between 1.01am and 4:59 am i've got 38 degrees
[image]

MichaelDvP, 2020-10-01:
Let's see:
First picture: 
N2 Boiler confirms setting selTemp to 0x2F (47 °C), 
but in N3 Controller rewrites to 0x38 (56°C). 
It's like something in hardware is set to 56°C, like my MC10 if the poti isn't set to auto (=60°C).

Second picture:
Thermostat send setTemp to boiler in UBAFlags (N1) and publishes the temp in 0x31B (N2), this is a undocumented message type.
Boiler confirms setTemp in message 0x34 (N4) to 47°C., selTemp (N9) stays on 56°C.

Third picture:
Thermostat publishes lower wwTemp in message 0x31B (offset 1) (N4), it's not send to the boiler.

Last picture, same as first, nothing changed for selTemp.

What to do:
`call boiler wwtemp` sets selTemp, which is overwritten and UBAFlags like the thermostat, that should work for the boiler, but not for thermostat setting. (I think wwtemp is temporary set until next switchtime of the thermostat?)
To set the thermostat setting we need to send a command to the thermostat, we have to check if write to 0x31B works, with wwtemp in offset 0 and wwtemplow in offset 1.

@lsw271 can you check the commands
`send telegram "0B 10 FF 00 02 1B 2D"` to set wwtemp to 45°C in thermostat and
`send telegram "0B 10 FF 01 02 1B 1E"` to set the lower wwtemp to 30°C
If this works i can add the commands to thermostat. 
The CW400 should also have a setting to set wwmode to always high, always lower and auto (timer), this is one of the settings i ask for in #518.

mglatz, 2020-10-01:
I dont have a thermostat at all but as mentioned earlier, boiler would overwrite `wwSetTemp` I sent in about 4 minutes time. I configured my home automation to send the desired `wwSetTemp` value every 3 minutes. This setup seems to work and boiler will heat the water to `wwSetTemp` temperature.

MichaelDvP, 2020-10-01:
@mglatz Is it [this boiler](https://www.manualslib.de/manual/247776/Junkers-Cerapur-Comfort.html?page=32#manual)? Try to set the rotary control to max. If Bosch uses the same logic as in MC10, than only in this position a ems-control is possible.

mglatz, 2020-10-01:
probably, the panel looks the same. Thanks for the tip, I'll try it. I would anyway but first I wanted to test for some time that ems-esp can reliably set and maintain lower temperature than `wwSelTemp` else my budget would not appreciate it :).

mglatz, 2020-10-01:
@MichaelDvP I don't think it works as you described.  
Tested with the heating wheel (lefto one): switched to max: `heatingTemp` and `selFlowTemp` went to 88. I then called `/api?device=boiler&cmd=flowtemp&data=39` which set `selFlowTemp` to 39, but after 4 minutes boiler set it automatically back to 88

lsw271, 2020-10-01:
@MichaelDvP 

> 
> 
> Let's see:
> First picture:
> N2 Boiler confirms setting selTemp to 0x2F (47 °C),
> but in N3 Controller rewrites to 0x38 (56°C).
> It's like something in hardware is set to 56°C, like my MC10 if the poti isn't set to auto (=60°C).

Ok, setted it to 60°C by rotary control on the boiler
> Second picture:
> Thermostat send setTemp to boiler in UBAFlags (N1) and publishes the temp in 0x31B (N2), this is a undocumented message type.
> Boiler confirms setTemp in message 0x34 (N4) to 47°C., [...]
I think it will be send to the boiler when lower temp starts as set in schedule. 

> Last picture, same as first, nothing changed for selTemp.
>
> What to do:
> `call boiler wwtemp` sets selTemp, which is overwritten and UBAFlags like the thermostat, that should work for the boiler, but [...]

I tried, temperature back to previous after few seconds, i done this after setting temperature on boiler to max 

> To set the thermostat setting we need to send a command to the thermostat, we have to check if write to 0x31B works, with wwtemp in offset 0 and wwtemplow in offset 1.
>
> @lsw271 can you check the [...]

Yes! it's working! :) 

> The CW400 should also have a setting to set wwmode to always high, always lower and auto (timer), this is one of the settings i ask for in #518.

I found this, tested and added a comment ;)

lsw271, 2020-10-06:
@MichaelDvP thanks for adding wwmode / wwtemp / wwtemplow.
Values setting in putty are correct in thermostat but I didn't saw wwtemp and wwtemplow in mqtt explorer and putty, only wwmode. Could you add this?
[image]

proddy, 2020-10-07:
@MichaelDvP I'll add this. I'm planning to make some changes to the thermostat to show all the data in the web UI too, like I did with the Boiler.

proddy, 2020-10-07:
@lsw271 do you mean the commands? not sure what's missing

MichaelDvP, 2020-10-07:
I know, I can add it with summermode correction after you finished your changes. 
It's in `export_values_main(`:
```
    // Warm water temp
    if (Helpers::hasValue(wwTemp_)) {
        rootThermostat["wwtemp"] = wwTemp_;
    }

    // Warm water temp low
    if (Helpers::hasValue(wwTempLow_)) {
        rootThermostat["wwtemplow"] = wwTempLow_;
    }
```

proddy, 2020-10-07:
ah, ok.

MichaelDvP, 2020-10-08:
@lsw271 
Please check if it is now working as expected, also for summermode.

proddy, 2020-10-09:
@lsw271 if this works I'll move the 2.1.0b out of dev into the next official stable release, so let us know if this now works soon please !

lsw271, 2020-10-09:
Sorry for my late, i will give you response tonight

lsw271, 2020-10-09:
@MichaelDvP @proddy commands are correct but in mqtt explorer I saw wwmode / wwtemp /wwtemplow only after called command, I think it should always be in payload, not only after changes

after sending wwtemp 
[image]
[image]

MichaelDvP, 2020-10-10:
These telegrams are only broadcasted by the thermostat on change, to get the values we have to fetch them. Should work now.

lsw271, 2020-10-10:
> 
> 
> These telegrams are only broadcasted by the thermostat on change, to get the values we have to fetch them. Should work now.

yes, its working, thanks!

commit 10fcd207 (2020-10-10): fetch thermostat wwmode, wwtemp, fix #523
closed 2020-10-10 by lsw271