emsesp/EMS-ESP #524: thermostat emulation

mglatz, 2020-09-29:
How difficult would it be to have EMS-ESP to emulate real thermostat?
Why: I dont have any thermostat at all. I have a smart home where are radiator valves are electronic and controlled independently based on current temperature in each room. So I have one circuit and boiler will only heat the heating water if at least one of the vales is open. But it seems that without a thermostat, very few "things" can be directly controlled on the boiler (jukers cerapur).
So... could technically EMS-ESP pretend to be a thermostat to make the boiler more controllable?

proddy, 2020-09-29:
quick answer: difficult! I wanted originally to build a smart thermostat but quickly realized that commercial thermostats (even the cheap ones) are quite clever. The know the heating curves of the boiler and adapt to ambient room temperatures. Trying to simulate this is quite complex. In your case it could be possible by using some home automation software to turn on and off the boiler.

mglatz, 2020-09-29:
interesting. I'd expected them to be quite dumb, and thought that they only measure temperature and memorize the week plan that one can set up.
I'm planning to integrate ems-esp more deeply in my home automation, so far I have not really been very lucky to understand how to control the boiler with ems-esp. I'll switch on the heating in a week or so and then I'll experiment with the settings some more. Feel free to close this issue :)

OT, but I'd really appreciate if there'd be more detailed wiki, perhaps for various boiler types and devices. For example I found out that my Junkers will send wwSetTemp and wwSelTemp the value that is set on the boielr with the control wheel. I can set wwSetTemp to a value, but only lower than wwSelTemp is and it restores to its original value in 10 minutes or so. There should be a page describing this behavior for newcomers :)

proddy, 2020-09-29:
known issue. The boiler resets the values. Michael has been battling with this for the last weeks.

Thermostats are clever little things. Google for pictures of the circuit boards and you'll see there's a lot going on. @bbqkees knows this area better than I do.

bbqkees, 2020-09-30:
Technically EMS-ESP can send the correct commands to the boiler to turn on the heating.
But there is an algorithm in the EMS thermostats that will control the burner very finely. So its not just on/off but depending on a heating curve, the setpoint and the current room temperature.
If for instance the room temperature is near the setpoint, the burner will not have to work 100% but will be modulated to like 15%. This increases comfort and energy efficiency.
So its more a design decision to have EMS-ESP just change the setpoint of the thermostat, and let the thermostat do all the hard work of actually controlling the burner.

The wiki could indeed have more details on boiler types but there are over 150 supported boilers and most of them have a specific behavior for each parameter so that would be a massive task to create a detailed wiki for that.

mglatz, 2020-09-30:
yeah, I don't expect you guys to have all of it prepared at once for us, but you could create a wiki where we could gradually fill out our findings about various devices for future reference and others to try.

mglatz, 2020-10-05:
shall we close this?

bbqkees, 2020-10-05:
Yes this can be closed I guess.

Also keep in mind that EMS-ESP is not meant to substitute a critical component like an EMS thermostat.
Its intended to augment an EMS system.
Its open source, everything is reverse engineered so it might brick up at some point and you don't want that to happen to a thermostat emulator if its -10 degrees Celsius outside and you are on a holiday for two weeks.

jorritsmit, 2020-10-16:
I am also interested in this, for the following reasons:

1. Use temperature from other sensors in your smart home
2. Base the heating curve on outside temperature as well (also needed for point 3 i think)
3. Control the heating curve based on the desired room temperature at desired time. e.g slow warming up when possible (know desired temp at time far ahead of time) for extra comfort (no overshoot) and energy efficiency or quick warming up when you want the house now.  
4. Fun. I guess the current thermostat knows the boiler and the current room temperature but that's about it. I can make the thermostat aware of outside temperature and learn the heat capacity of the room (how quickly it warms up), which would be different for every house of course. This would basically require making a feedback and maybe feedforward controller, and for that i would need to do system identification and decide if that has to be done for every house or only every boiler. In summary the project can start simply and be expanded indefinitely in complexity. 

To start this all however i need to know two things:
1. Can i change the device type of the gateway, so i can get the same control as a thermostat (what happens when two thermostats are present in the same network?)
2. Is the boiler current power the only thing that is controlled by the thermostat (i can probably find that out by myself)

any other resources would be greatly appreciated

proddy, 2020-10-16:
@jorritsmit that would indeed be a nice project. If you have a active thermostat you can use EMS-ESP to learn its behavior by watching the telegram commands, and then build a simulation. I would start there and figure out which commands its sending to the boiler, which is not just boiler power. EMS-ESP can emulate any device-type so that's not a problem but not co-exist as we haven't build in a response to the version telegram yet. There is always some handshaking happening when a thermostat is attached to the bus master (boiler on 0x08). Some non-Bosch branded thermostats have solved this (like tado and google nest) so it is possible. I guess they bought the protocols from Bosch.  

Like I said, it would be great if you get this working but remember the goal of EMS-ESP is to be a programmable bridge that understands EMS, but we'll support any side projects.

jorritsmit, 2020-10-16:
I understand the goal of the project and think the function of the gateway as is, is best in 99% of causes and therefore should remain. This will probably take some time to develop, since I won't be working on it for more then a day a week. But I'll fork and report back when I have something useful.

Thanks for helping me on my way and congrats on the project. It looks like there has been put a lot of effort in and I love it so far.

MichaelDvP, 2020-10-16:
@jorritsmit 
>   Can i change the device type of the gateway, so i can get the same control as a thermostat 

yes, you can also use additional device types. I've added a roomcontroller which recognized as remote control for the master thermostat for every heatingcircuit. 
 
> (what happens when two thermostats are present in the same network?)

There is only one masterthermostat allowed. Other thermostats can only play the role of remote controllers. Some thermostats for single heatingcircuit can be combined to a master, where every thermostat controlls one circuit.

>  Is the boiler current power the only thing that is controlled by the thermostat (i can probably find that out by myself)

It's quite simple, most devices have own logic. the boiler gets only the desired flow temperature and switches on, off and modulate by himself. The thermostat tells only "heating on" and "setpoint"  (0x1A). The mixer have also a PI-control integrated and gets only the setpoint from thermostat (0xAC). 

But thermostat is the HMI for all devices and collects all data and show them on a display. 
And it calculates the flowtemp by heatingcurve, outdoortemperature, roomtemperature, damping of the building, and some other parameters stored inside the thermostat. It can also make optimization cycles. The calculations are not difficult and mainly described in the thermostat manuals. But it's mainly what you like to do.

> any other resources would be greatly appreciated

For the ems messages [https://emswiki.thefischer.net/doku.php](https://emswiki.thefischer.net/doku.php)
For ems+ and HT3: [https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf](https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf)
Manuals from Buderus (german site, but some docs are multi language [https://www.buderus.de/de/technische-dokumentation](https://www.buderus.de/de/technische-dokumentation)

ems-esp can change the parameters the thermostat uses to calculate, we can always manipulate what the thermostat do.

@proddy version response is in roomcontrol for device-id 0x18-0x1B as fixed RC20, product 113, version 2.01.

proddy, 2020-10-17:
@MichaelDvP ah yes, I forgot you added version response. I need to keep up!

fiskn, 2020-10-17:
I'm currently embarking on building something similar to these requirements, here is what I'm doing:

1. This is the most important bit. Don't modify the existing legacy controls. Otherwise you can pretty much guarantee something will go wrong with your "smart" system when you least want it to. Simply set any legacy control panel to off, you can always turn it back on again if needed in an emergancy.
2. To achieve the above, use the existing "dumb" call for heat wire from the zone valves to still turn the boiler on/off. Wire arduino/esp8266 controlled relays in parallel with whatever was controlling the zone valves
3. Using a combination of homeassistant & node-red, build some logic to turn on/off zone values and thus boiler.
4. Deploy temperature sensors around house feeding into homeassistant
5. Using more logic in node-red, take value(s) from temperature sensors and feed into a PID node in node-red, map the output of the PID 0-1 to the flow temp (40-70C???) and send via MQTT to EMS-ESP

The above should give you a fairly good "smart" thermostat, but also allow you to add in additional automation, like basing heat requirements on calendar entries or if nobody is home...etc

Oderik, 2020-11-16:
What is discussed here is more or less the reason why I started using EMS-ESP. First I'd like to describe my situation:

I'm living in rented appartment. There is a boiler installed in the bathroom for heating and warm water. It is controlled by a thermostat at the other end of the appartment. The room wit hthe thermostat was obviously supposed to be the living room, but unfortunately we use that room as our bedroom. So the thermostat measures temperatures of a room that barely needs heating. My home office room is connected to the same heat circuit but it is on a different floor of the house, forming a separate appartment.

I configured a heating curve with an eco temperature of 16°C for most of the time and 20°C in the morning to heat up the bath room before anybody gets up. When I do get up, I put he thermostat to manual heating mode (if I want it to heat of course) and put it back to auto when I go to bed again. I often forget those manual interventions. When it gets cold in my office, I need to go all the way to the bedroom to activate the boiler. And when I forget to deactivate heating, it runs all night.

Now that I have started to play around with smart home stuff (Home Assistant), I thought there should be a more convenient way to deal with boiler automation. One added convenience is of course the ability to control the thermostat from HA manually. It is even more convenient that I can replace my manual intervention with some automation triggered by presence, alarm a.s.o.

What I am still missing is the ability of how to make the boiler run only if any room is below it's desired temperature (and then automating that desired temperature). I guess there is a way to do it already, but I find it a bit confusing to see what would be the "best" way to achieve that. I know how to monitor temperatures I am interested in so I know when the boiler should be heating or not. So one approach would be to set the bedroom's setpoint below or above the bedroom's room temperature to indirectly control the boiler. But that doesn't seem to be very elegant.

I already learned about the possibility to control the boiler directly, mainly by setting it's flow temperature. But when I set the bedroom thermostat to eco (as I don't want the bedroom to be heated), it cancels my custom heating request after several minutes. I could just disconnect the thermostat, but I don't want to for several reasons (safety, temperature input, manual control).

So, @MichaelDvP and @norberts1 mentioned some interesting stuff related to all this: adding additional device types, remote controler for master thermostat, combining thermostats, changing thermostat calculation parameters, roomcontrol, any module controlling the system. I did not understand everything in detail. Can I let my single thermostat integrate more temperature sources (outside and other room temperatures) by providing it with readings over the ems bus? Do I have to create virtual remote thermostat controllers for that (ems-esp boradcasting with multiple device ids)? Can that be easily achieved without changing the source code of ems esp? Or would I rather create a new virtual master thermostat using ems esp and let the hardware thermostat act as a remote control? Or woul I just directly control the boiler from HA and override unwanted changes by the original master thermostat as soon as HA observes them?

I know that are a lot of questions. Maybe some of you guys are willing to have a voice chat about these topics soon?

Oderik, 2020-11-16:
> Or woul I just directly control the boiler from HA and override unwanted changes by the original master thermostat as soon as HA observes them?

I implemented that approach for now. In HA, I started by defining generic thermostat devices:
```yaml
- platform: generic_thermostat
  name: Wohnzimmer
  heater: input_boolean.heizung_wohnzimmer
  target_sensor: sensor.28_0517601008ff_temperature
  min_temp: 15
  max_temp: 23
  away_temp: 16
  target_temp: 17
- platform: generic_thermostat
  name: Badezimmer
  heater: input_boolean.heizung_badezimmer
  target_sensor: sensor.dallas_sensor_1
  min_temp: 15
  max_temp: 23
  away_temp: 16
  target_temp: 17
- platform: generic_thermostat
  name: Büro
  heater: input_boolean.heizung_buro
  target_sensor: sensor.diy1_ds18b20_temperature
  min_temp: 15
  max_temp: 23
  away_temp: 16
  target_temp: 17
```
The input_booleans are of course defined helpers. I put them together in a "or" group:
```yaml
heating:
  name: Heizung
  all: false
  entities:
    - input_boolean.heizung_wohnzimmer
    - input_boolean.heizung_badezimmer
    - input_boolean.heizung_buro
```
An automation monitors the group's state as well as the boiler's selected flow temperature. If the group is on and the flow temperature is not equal to the heating temperature setting, it will be set to it.
```yaml
- alias: Heizung
  description: ''
  trigger:
  - type: value
    platform: device
    device_id: 341f22ae81c32def5096f8e26a8530d6
    entity_id: sensor.boiler_selected_flow_temperature
    domain: sensor
    below: 15
  - platform: state
    entity_id: group.heating
    to: 'on'
  condition:
  - condition: not
    conditions:
    - condition: template
      value_template: '{{ is_state(''sensor.boiler_selected_flow_temperature'', states(''sensor.boiler_heating_temperature_setting''))}}'
  - condition: state
    entity_id: group.heating
    state: 'on'
  action:
  - service: mqtt.publish
    data:
      topic: ems-esp/boiler
      payload_template: '{ "cmd": "flowtemp", "data": {{ states("sensor.boiler_heating_temperature_setting")
        }} }'
  mode: single
```

Deactivation of heating is handled by the original thermostat. I don't force the boiler off for not overriding the thermostat's requirement to heat.

I will later add intelligent radiator thermostats to the system.

This seems to work more or less but yet it doesn't feel 100% satisfying. I would really like to play nicer together with the thermostat.

levran, 2020-11-28:
I also have a Junkers system boiler and the command **heatingactivated**  does not work. 
To operate the boiler as before. As standard, the HA thermostat gave the command to warm / not warm to relay. The mode is clear and quite convenient.

I only have a Junkers Cerapur Compact ZWB 24 boiler and an interface EMS board. 
How can I help with data and experiments to fix command **heatingactivated** function?

Oderik, 2020-11-28:
For me the the most reliable command for activating heating is `flowtemp`. Whenever I want to start heating, I set the `flowtemp` to the boiler's heating temperature setting (the temperature you set using the corresponding dial on the boiler).

As mentioned above, I leave the resetting of the flow temp to 0 to the hardware thermostat. It fires flow temp settings every now and then.

proddy, 2020-11-28:
@levran can you open a new issue for "heatingactivated does not work" so we can fix that

MichaelDvP, 2020-11-28:
The last "heatingactivated does not work" was #620 and was fixed only 6 days ago. @levran can you check this and try with the actual dev-software if it is the same issue.

levran, 2020-11-28:
Very interesting, now on version v2.1.1b6 the heating is turned off immediately after the start.
An increase in temperature and a command (call boiler heatingactivated on) to turn on the heating does not give a reaction.
The most interesting thing is that you can now change the temperature on the boiler itself, but the heating remains inactive.