emsesp/EMS-ESP #525: Command syntax / API documentation

deqw, 2020-09-29:
Since my commands to the thermostat are not working (v 2.0.1) I like to have a confirmation I'm using the correct syntax:

the topic I use: `ems-esp/thermostat_cmd`
the payload: 

```
{
  "cmd": "mode",
  "data": "auto"
}
```

When I check thermostat_data after publishing this with MQTT Explorer the mode setting remains "off". What am I missing here? Tx mode is set to 1.

In the API documentation some text is missing at the end of this piece of text: "via MQTT in the payload with {"cmd":<command> ,"data":<data>, "id":<id>}. The MQTT topic is the ."  So what must be filled in here? 

The 'Enable WEB API' option in the Settings configuration cannot be found in my v2.0.1. Does this have something to do with my commands not having any effect?

[image]

proddy, 2020-09-29:
The API was changed in 2.1 and the doc updated to reflect it. I haven't had time to merge 2.1 back into the mainline. So grab 2.1 and the MQTT command syntax is topic `ems-esp/thermostat` and the payload as you described. To check this you can also first try this in the Console using the `call` command.

deqw, 2020-09-29:
Thanks. Compliments for the new software and GUI, looking good.

I downloaded the EMS-ESP-2_1_0b2-esp8266 binary and tried to upload the firmware, however to no avail. Tried Safari, Chrome and Firefox, all latest versions on macOS. In all three browsers the bin file can be dragged into the drop file area, after that nothing happens, no message, nothing. Only in Chrome the bin can be selected for upload (but again nothing happens), in the other browsers the file is not selectable at all. Guess I have to wait wait for an OTA update.

proddy, 2020-09-29:
that's strange. I'll test myself tomorrow. Just to be sure, you selected the ESP8266 build and not the ESP32? Also not the compressed .gz.bin file?

deqw, 2020-09-29:
I took EMS-ESP-2_1_0b2-esp8266.bin that was compiled about an hour ago.

FireWizard52, 2020-09-29:
Upgraded to 2.1.0b2 using Waterfox (Firefox fork) under Linux (PCLinuxOS). I did not use drag/drop, but I selected the correct file.
It uploaded directly without any issues.

bbqkees, 2020-09-30:
I had someone else yesterday that could not update from 2.0.1 via the web interface. Only via OTA.

proddy, 2020-09-30:
Works fine on all browsers using Windows. I tested 2.0.1->2.1.0. But there is a problem with MacOSX because its blocking the file access. Probably a security thing. Need to check.

proddy, 2020-09-30:
fixed the upload for non-Windows

closed 2020-09-30 by proddy