emsesp/EMS-ESP #526: Update Domoticz plugin for EMS-ESP v2

proddy, 2020-09-29:
The previous Domoticz plugin for EMS-ESP v1.9.5 is no longer compatible with all the changes in v2.0.

@Gert05 has been working on new version. Would you mind sharing it so I can include the link in the EMS-ESP Documentation. And we can keep this GitHub issue open to collect feedback.

thanks

bbqkees, 2020-09-30:
The Domoticz plugin is still functional for the most part except for the thermostat topics.
https://github.com/bbqkees/ems-esp-domoticz-plugin.
We should not create two separate plugins.

Gert05, 2020-09-30:
Attached the updated plugin.py.

It provides a basic interface in Domoticz: read room temperature & system pressure, change thermostat temperature.

Plugin.py goes into a “nefit” directory in /home/pi/domoticz/plugins (on a raspberry pi).

It also needs an mqtt.py; this is unchanged from the previous version.

Tested on Domoticz 2020.2 (12125) & EMS-ESP v2.0.1b4.

proddy, 2020-09-30:
missing attachment !

Gert05, 2020-09-30:
Noticed that. Struggling with GitHub…

proddy, 2020-09-30:
Could you include a small README on how to install/use. Then I'll also add it to the dev branch and include a link in the Documentation. Also make sure it works with 2.1 where the topic names have changed (e.g. from thermostat_cmd to thermostat).

closed 2020-09-30 by proddy