emsesp/EMS-ESP #527: new unit with firmware V2.0.1b4 is not working

florin-laurentiu, 2020-09-29:
dear ALL,
 just receive a new unit with firmware V 2.0.1b4 which is not running.
it is mounted in  a Buderus GB162, 35kW with RC 35 installed as also solar module MM10 .
1. first try is to connect new interface with supplied cable jack  to the service port in front of Buderus and is appear on RC 35  display: CALL SERVICE and Buderus unit not working ! Power down and restart with jack cable out and Buderus working again.
2. second try on EMS BUS connector, connect existing interface and power up, Buderus start working and interface blinking fast; at this step i was able to do configuration according with wiki instructiond and for checking, i switch power off again and after up.  ALL parameters that was setted before is missing as never it was good set; try again and some result: each time when restart interface or power up interface or Buderus, need to start again to set up interface.
More of this, if keep alive without any restart or power off, after few minutes interface is begin to blinking fast and loose connection with EMS bus as home network also.
3 step is to connect power on USB, with no connection on EMS of Buderus : same result, means each time when restart or power off , interface is not configurated and if not switch power off, after few minutes is begin blinking fast and loose connection.
4 step it was to download firmware 2.0 from Github and try to install in OTA mode; APARENTLY EVERYTHING LOOKS OK, but after restart interface not working; after power up, for one second is light blue led from inside and nothing after.

I can t do nothing more , because i don t know what to do and interface is not working/doing nothing.
Do you think that can help me to do interface running or is broken?
thank you, 
Laurentiu Pitran ; laurentiu.pitran@gmail.com, GSM :+40 724 209 647

bbqkees, 2020-09-29:
On some boilers the jack cable does not fit perfectly because the hole is too tight. So try to unplug it and reinsert it firmly.
Indeed I shipped some Gateways in the past two weeks with 2.0.1b4, which appears to have the issue that it did not store the values properly.
If you think you bricked the device please try out the NodeMCU method to flash the firmware: https://bbqkees-electronics.nl/wiki/gateway/firmware-update-to-v2.html

You can try out the latest beta; https://github.com/proddy/EMS-ESP/blob/firmware/firmware/EMS-ESP-2_1_0b2-esp8266.bin

florin-laurentiu, 2020-09-29:
hello,
received your answer.
check 2 time the jack cable with connector as also how is fit inside hole - everything is OK , no mistake.
i will try the link that you post on the answer; do you think that i can do this action properly as i don t have any previous experience with this kind of hardware?

thank you!

florin-laurentiu, 2020-09-29:
Hello,

received your answer.
proceed as in the link that you send;
unfortunately, it not possible to put/download inside WEMOS the .bin
package.
as soon as i try to flash with NodeMCU, flashing is not possible as the
WEMOS is apparently broken .

laurentiu

proddy, 2020-09-29:
@florin-laurentiu just remove the Wemos from the gateway, use a USB cable to laptop/PC and follow the guidelines to upload the latest firmware manually. Then test without attached to your Buderus, using the Wifi and Console (see Documentation).  BBQKees physically tests all gateways with EMS-ESP before shipping so I'm 100% it works plus we have other users with the same hardware (GB162) so presume it works.

bbqkees, 2020-09-30:
@florin-laurentiu did you select the correct COM port?
Are you using Windows 10?

Its unlikely the Wemos is broken.
Do you get an error when trying to flash?

glitter-ball, 2020-09-30:
You'll also need a Windows 10 driver for the EMS-ESP's serial port. I had all sorts of fun trying to find one on the web to download, but got there in the end. It would be great if the wiki could include that info or a link.

Wonder whether any of the original issues are due to bus mode setting? It took me a while to realise mine was EMS+ and therefore needed to be on Mode 3.

proddy, 2020-09-30:
Think @bbqkees needs to add the driver link to his wiki as it depends on which serial/usb chip is in the Wemos he installs. They do differ per batch. Most people coming to this GitHub page are usually technically apt and don't mind tinkering & building firmware. Afterall it is free open-source code, and not a commercial support site :-)

bbqkees, 2020-09-30:
It could be years ago I installed a driver for such a board. Did not think
of it again so indeed it could be new users stumble upon this issue.

All Lolin ESP8266 D1 boards use the CH340 chip and most clones do as well.
The ESP32 boards usually have a SiLabs chip but this can vary a lot.

proddy, 2020-09-30:
right, and the driver is on their website: https://docs.wemos.cc/en/latest/ch340_driver.html

glitter-ball, 2020-09-30:
I think I found it by getting the hardware ID out of Windows 10 USB properties and then Googling it. I ended up on a strange .cn website but the driver worked OK.

florin-laurentiu, 2020-09-30:
come back in picture with some news, but have in mind same idea:  when i buy something from a shop, i consider that i can use this good as is it; in my case, to buy an interface with a version of software that is not official and more of that is not work properly, this is a same for me as person, that i was buying in this condition . 
more or less, even the vendor can tell me if not in advance, in writing in a paper in that parcell  that interface is running a software that is with problem, and all setting that i need to do is for nothing because the interface will not work as was build......
anyway, i was buying a new WEMOS same version, i was able to install the last firmware from GITHUB, and now all is running.
at this step i have some ""precaution "" from my part regarding percent of TX errors which is more that 700 hundreds after few hours; don t know if is OK or NOT....
for DALLAS temperature sensor, is not clear for me if i can use only one piece or more, and if is more , around which number...
if the firmware support more DALLAS sensor, i can put a name for each and send data to INFLUX and GRAFANA in such way to can record and use individually values from each sensor?
i will come back witn news or questions according with next  events regarding this interface for Buderus.

closed 2020-09-30 by proddy