emsesp/EMS-ESP #528: can't set wwtemp in V2.1.0b2

mglatz, 2020-09-30:
*Before creating a new issue please check that you have:*

**Bug description**
sending `[hostname]/boiler_cmd {"cmd":"wwtemp" ,"data":40}` will correctly update `wWSetTemp` in v 2.01 but does nothing in V2.0.1b4

**Steps to reproduce**
send the MQTT message in 2.01 (works); update to V2.1.0b2 (does not work); downgrade to 2.01 (works again) 

**Device information**
Junkers Cerapur Comfort identified in 2.1.0.b2 as Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3
device: Gateway Premium II

**Additional context**
In V2.1.0b2 setting wwtemp with REST `[hostname]/api?device=boiler&cmd=wwtemp&data=40` works
adding  "id" to meessage does not help

proddy, 2020-09-30:
does it work in the Telnet console? See https://emsesp.github.io/docs/#/Troubleshooting

mglatz, 2020-09-30:
telnet command works
```
su
call boiler wwtemp 43
```

but pushing the above message via mqtt will show in log
`No MQTT handler found for topic kotel/boiler_cmd and payload {"cmd":"wwtemp" ,"data":40}`

proddy, 2020-09-30:
the topic's have been renamed. It's now `boiler` and not `boiler_cmd` in 2.1.

See https://github.com/proddy/EMS-ESP/blob/dev/CHANGELOG.md#ems-esp-changelog
and https://emsesp.github.io/docs/#/MQTT

closed 2020-09-30 by proddy