emsesp/EMS-ESP #531: incomplete dashboard values

mglatz, 2020-10-02:
**Question**
Why the dashboard in web UI does not show all the available values we can see in console with `show`. I see 46d different params for boiler in console, but dashboard only displays 11 of them.

**Device information**
Junkers Carapur, ems-esp v2.1.0b2

proddy, 2020-10-02:
memory!
I can change one line in the code and all 62 will be shown, but I'm afraid it'll run out of available memory. I need to see if I can chunk into sections and re-build in the webUI.

mglatz, 2020-10-02:
I see :(

proddy, 2020-10-03:
I'll work on this next.

MichaelDvP, 2020-10-03:
@mglatz you can use [http://ems-esp.local/api?device=boiler&cmd=info](http://ems-esp.local/api?device=boiler&cmd=info) to see all boiler values.

mglatz, 2020-10-03:
I know. I was curious why the dashboard lacks those values.
reading json is not that great (although is probably imporoved in latest dev, not tested it yet) and the shortcuts are rather diffucult to understand (Current flow temperature is better than curFlowTemp).
In the meantime, you could maybe add a link to dashboard that would open up the json api

proddy, 2020-10-03:
@mglatz not sure what you're suggesting. Do you have a use case or scenario on how you want to work with the data from EMS-ESP. Most of us use MQTT or Domoticz to record and create actions on the data. The tiny web interface was really built to configure EMS-ESP, not to replace Gafana/Home Assistant/Domoticz etc. But happy to make improvements and understand how people plan to use the software

mglatz, 2020-10-03:
usually I'd read the values I'm interested in with home automation but I dont pull all the values the boiler offers. Time to time I want to view some "unusual" ones. Of course, they are visible in mqtt or api as suggested by @MichaelDvP but none of them are very user friendly. Since the ems-esp ui is nice, it would also be nice to have all the values.
And since there is currently no way to see all the values in the UI, adding a link to /api?device=boiler&cmd=info

something like this could work befofre (and if) you'd integreate all the values directly to dashboard
[image]

proddy, 2020-10-03:
ok, I see. I'm working on adding all the values to the web. I could make a new command which will expose the json with human-readable names as you suggested.

proddy, 2020-10-04:
@mglatz I also noticed in your screenshot above that your service code string is empty and the code is 0. Has this always been the case? We're using the value of the code to determine if the heating and hot water is on/off.

mglatz, 2020-10-04:
Yes. I was thinking of manually crating some error (such as gas cutoff) to see how these two behaves but did not go that way yet. Nevertheless I would expect they should be empty/0 under normal circumstances, shouldn't they?
in MQTT:
heating_active always shows 1 even though I don't heat yet.
wWactivated always shows 1
wwActive always shows 0

proddy, 2020-10-04:
are you on the latest 2.1 dev build?

mglatz, 2020-10-04:
v2.1.0b2

proddy, 2020-10-06:
added all the values in 2.1.0b5. If you're seeing empty/0 for some of the values please log a new bug.

closed 2020-10-06 by proddy