emsesp/EMS-ESP #533: EMS-ESP does not keep settings when rebooting

digdagdigedag, 2020-10-03:
Whenever I reboot my new EMS-ESP (v2.0.1b4) , all settings for Wifi, NTP and MQTT are lost !!!
Is there any special setting needed to ensure that settings are kept ?
Thank you for your support

EMS : Buderus Plus GB 152-24

proddy, 2020-10-04:
you're the 2nd person now to report this with 2.0.1b4, assuming its an ESP8266. Unfortunately, I'm not sure what is causing it. I would suggest taking the latest dev build from https://github.com/proddy/EMS-ESP/tree/firmware/firmware and using that. If it still happens then we'll need to wipe the ESP chip and reflash it. https://emsesp.github.io/docs/#/Troubleshooting

MichaelDvP, 2020-10-04:
@proddy: You have fixed it long time ago in #492 (unformated littlefs).
@digdagdigedag: update to an actual version (release or beta) and it is fixed.

closed 2020-10-04 by proddy