emsesp/EMS-ESP #536: HA error -

majdzik84, 2020-10-05:
Proddy, thank you for your work. I ran into errors with autodiscovery in HA.

I migrated from ver.1.9.5 to the newest 2.1.0b4 and I'm getting errors in the logs.
I can see that the detected thermostat does not show me manual modes (in 1.9.5 I had day / night - I have the RC35 driver).
[image] 
Below I am sending logs from HA.

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:358
First occurred: 10:19:09 (6 occurrences)
Last logged: 10:20:30

Error parsing value: 'dict object' has no attribute 'hc2' (value: {"hc1":{"seltemp":0,"targetflowtemp":5,"mode":"auto","modetype":"summer"}}, template: {{value_json.hc2.currtemp}})
Error parsing value: 'dict object' has no attribute 'hc2' (value: {"hc1":{"seltemp":0,"targetflowtemp":5,"mode":"auto","modetype":"summer"}}, template: {{value_json.hc2.seltemp}})
Error parsing value: 'dict object' has no attribute 'hc2' (value: {"hc1":{"seltemp":0,"targetflowtemp":5,"mode":"auto","modetype":"summer"}}, template: {{value_json.hc2.mode}})

```
```
Logger: homeassistant.components.mqtt.climate
Source: components/mqtt/climate.py:467
Integration: MQTT (documentation, issues)
First occurred: 10:19:09 (2 occurrences)
Last logged: 10:20:30

Invalid modes mode: {"hc1":{"seltemp":0,"targetflowtemp":5,"mode":"auto","modetype":"summer"}}
```
```
Logger: homeassistant.util.logging
Source: util/logging.py:108
First occurred: 10:16:30 (96 occurrences)
Last logged: 10:20:41

Exception in async_discover_sensor when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'Summer temperature', 'unique_id': 'thermostat_hc2_summertemp', 'unit_of_measurement': '°C', 'state_topic': 'ems-esp/thermostat_data', 'value_template': None, 'icon': None, 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 83, in async_discover_sensor config = PLATFORM_SCHEMA(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: template value is None for dictionary value @ data['value_template']
Exception in async_discover_sensor when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'Nofrost temperature', 'unique_id': 'thermostat_hc2_nofrosttemp', 'unit_of_measurement': '°C', 'state_topic': 'ems-esp/thermostat_data', 'value_template': None, 'icon': None, 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 83, in async_discover_sensor config = PLATFORM_SCHEMA(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: template value is None for dictionary value @ data['value_template']
Exception in async_discover_sensor when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'Current flow temperature', 'unique_id': 'mixing_flowTemp', 'unit_of_measurement': '°C', 'state_topic': 'ems-esp/mixing_data', 'value_template': None, 'device': {}, 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 83, in async_discover_sensor config = PLATFORM_SCHEMA(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: template value is None for dictionary value @ data['value_template']
Exception in async_discover_sensor when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'Setpoint flow temperature', 'unique_id': 'mixing_flowSetTemp', 'unit_of_measurement': '°C', 'state_topic': 'ems-esp/mixing_data', 'value_template': None, 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 83, in async_discover_sensor config = PLATFORM_SCHEMA(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: template value is None for dictionary value @ data['value_template']
Exception in async_discover_sensor when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'Valve status', 'unique_id': 'mixing_valveStatus', 'state_topic': 'ems-esp/mixing_data', 'value_template': '{{value_json.valveStatus}}', 'device': {}, 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 83, in async_discover_sensor config = PLATFORM_SCHEMA(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: Device must have at least one identifying value in 'identifiers' and/or 'connections' for dictionary value @ data['device']
```
I was also not detected (probably due to errors) of all sensors
[image]
[image]
[image]
[image]
[image]

proddy, 2020-10-05:
You don't see an HA device called 'EMS-ESP Thermostat' ? Can you send me the MQTT payload for topics `thermostat_data` and `homeassistant/climate/ems-esp/*`

majdzik84, 2020-10-05:
I see the thermostat as well, but only in the auto mode, the photo is not stuck.
[image]

jtonk, 2020-10-06:
I'm getting the same issue. the HA autodiscovery has the wrong modes. When I manually update the MQTT message from `"modes": ["night","day","auto"]` to `"modes": ["off","heat","auto"]` it works again, however it's not persistent on a reboot.

jtonk, 2020-10-06:
In the Thermostat.cpp I see this code line 574:
`    if (flags == EMSdevice::EMS_DEVICE_FLAG_RC20_2) {
        modes.add("night");
        modes.add("day");`

I have the RC20 thermostat, maybe it is a simple as changing it to `off` and `heat` but I can't compile and test.

proddy, 2020-10-06:
@jtonk thanks for hunting that bug down. I pushed a fix.

jtonk, 2020-10-06:
tnx, it's working again!

proddy, 2020-10-06:
@majdzik84 can you test again?

majdzik84, 2020-10-06:
I confirm, it is much better :) Thank you
[image]

closed 2020-10-07 by proddy