emsesp/EMS-ESP #537: Junkers FR120 thermostat

zanerv, 2020-10-05:
Hi

**Is your feature request related to a problem? Please describe.**
I can't set temperature setpoint to Junkers FR120 thermostat 

**Describe the solution you'd like**
I'll like to be able to set the thermostat temperature from the ems-esp gateway but in compatibility list it's marked as read only
I tried to set the temp/mode from homeassistant and that didn't work

**Additional context**
[image]

proddy, 2020-10-05:
The FR120 should support sending write commands. See https://emsesp.github.io/docs/#/Supported-EMS-Devices

Try out https://emsesp.github.io/docs/#/Troubleshooting?id=some-commands-via-mqtt-are-not-working and send the console output so we can trace what is happening.

zanerv, 2020-10-06:
thanks for looking into this, here is the log 
```

ems-esp:/# show devices
These EMS devices are currently active:

Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:15.07)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE9 

Thermostat: FR120 (DeviceID:0x10, ProductID:191, Version:55.01) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x16F 0x165 0x170 0x166 0x171 0x167 0x172 0x168 

Controller: HT3 (DeviceID:0x09, ProductID:95, Version:15.07)

000+00:01:01.745 E 3: [telegram] Rx: FF FE FF 10 (CRC 10 != CD)
ems-esp:/# su
Password: 
000+00:01:12.199 N 4: [shell] Admin session opened on console
000+00:01:14.260 E 5: [telegram] Rx: 90 00 06 00 14 0A 08 06 1A 01 02 00 0D (CRC 0D != 18)
000+00:01:20.369 E 6: [telegram] Rx: 88 00 18 00 39 01 54 64 00 01 23 20 C0 00 F9 80 00 80 00 FF FF FF 00 00 00 00 00 00 39 (CRC 39 != 4B)
000+00:01:24.705 E 7: [telegram] Rx: FF FC FE FF 0C (CRC 0C != 76)
ems-esp:/# log debug
Log level = debug
000+00:01:25.741 E 8: [telegram] Rx: FE FE FE FF 10 (CRC 10 != 76)
000+00:01:26.612 D 9: [emsesp] Received UBAMonitorFast
000+00:01:26.775 D 10: [mqtt] Publishing topic ems-esp/tapwater_active (#26, attempt #1, pid 1)
000+00:01:26.976 D 11: [mqtt] Publishing topic ems-esp/heating_active (#27, attempt #1, pid 1)
000+00:01:29.950 D 12: [emsesp] Received UBAMonitorFast
ems-esp:/# thermostat
000+00:01:30.189 E 13: [telegram] Rx: FF FE FF 88 00 34 00 31 00 F8 80 00 81 00 00 01 00 00 F2 5A 00 DB C9 00 D8 (CRC D8 != 4C)
000+00:01:30.194 D 14: [mqtt] Publishing topic ems-esp/boiler_data (#28, attempt #1, pid 1)
000+00:01:30.395 D 15: [mqtt] Publishing topic homeassistant/climate/ems-esp/hc1/state (#29, attempt #1, pid 1)
000+00:01:31.382 E 16: [telegram] Rx: FF FC FE FF 10 (CRC 10 != 76)
000+00:01:32.228 D 17: [emsesp] Received UBAMonitorFast
000+00:01:32.441 E 18: [telegram] Rx: FE C7 9E FF FE FF 20 (CRC 20 != 4B)
000+00:01:33.440 E 19: [telegram] Rx: FF FC FE FF 40 (CRC 40 != 76)
000+00:01:34.317 D 20: [emsesp] Received UBAMonitorFast
000+00:01:34.533 E 21: [telegram] Rx: FF FC FE FE FF 68 (CRC 68 != 11)
000+00:01:35.346 D 22: [emsesp] Received UBAMonitorFast
000+00:01:35.561 E 23: [telegram] Rx: FE C7 9E FF FF 0F (CRC 0F != 5B)
000+00:01:36.560 E 24: [telegram] Rx: FF FE FF 10 (CRC 10 != CD)
000+00:01:37.405 D 25: [emsesp] Received UBAMonitorFast
000+00:01:37.621 E 26: [telegram] Rx: FE C7 9E FF FE FF 22 (CRC 22 != 4B)
ems-esp:/thermostat# call thermostat temp 15
000+00:01:39.807 D 27: [emsesp] Received UBAMonitorFast
000+00:01:40.047 E 28: [telegram] Rx: FF FF 88 00 34 00 31 00 F8 80 00 81 00 00 01 00 00 F2 5A 00 DB C9 00 D8 (CRC D8 != 21)
000+00:01:40.048 D 29: [mqtt] Publishing topic ems-esp/boiler_data (#30, attempt #1, pid 1)
000+00:01:40.249 D 30: [mqtt] Publishing topic homeassistant/climate/ems-esp/hc1/state (#31, attempt #1, pid 1)
000+00:01:42.022 E 31: [telegram] Rx: 98 88 00 18 00 39 01 F3 64 20 09 23 25 C0 00 F8 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 10 (CRC 10 != CC)
000+00:01:45.581 D 32: [emsesp] Received UBAMonitorFast
000+00:01:48.074 D 33: [emsesp] Received UBAMonitorFast
000+00:01:50.039 D 34: [emsesp] Received UBAMonitorFast
000+00:01:50.103 D 35: [mqtt] Publishing topic ems-esp/boiler_data (#32, attempt #1, pid 1)
000+00:01:50.280 D 36: [emsesp] Received UBAMonitorWW
000+00:01:50.304 D 37: [mqtt] Publishing topic homeassistant/climate/ems-esp/hc1/state (#33, attempt #1, pid 1)
000+00:01:50.572 D 38: [emsesp] Received UBAMonitorSlow
000+00:01:51.101 D 39: [emsesp] Received UBAMonitorFast
000+00:01:54.251 D 40: [emsesp] Received UBAMonitorFast
000+00:01:59.775 D 41: [emsesp] Received UBAMonitorFast
000+00:02:00.002 D 42: [emsesp] Fetching values for device ID 0x08
000+00:02:00.002 D 43: [telegram] Tx read request to device 0x08 for type ID 0x19
000+00:02:00.002 D 44: [telegram] Tx read request to device 0x08 for type ID 0x33
000+00:02:00.002 D 45: [telegram] Tx read request to device 0x08 for type ID 0x14
000+00:02:00.002 D 46: [telegram] Tx read request to device 0x08 for type ID 0x16
000+00:02:00.002 D 47: [emsesp] Fetching values for device ID 0x09
000+00:02:00.002 D 48: [emsesp] Fetching values for device ID 0x10
000+00:02:00.002 D 49: [telegram] Tx read request to device 0x10 for type ID 0x16F
000+00:02:00.005 D 50: [emsesp] Received UBAMonitorWW
000+00:02:00.161 D 51: [mqtt] Publishing topic ems-esp/boiler_data (#34, attempt #1, pid 1)
000+00:02:00.269 D 52: [telegram] Sending read Tx [#22], telegram: 8B 88 19 00 20 18
[... 34 lines cut ...]
000+00:02:21.018 D 87: [emsesp] Received UBAMonitorFast
000+00:02:21.084 D 88: [mqtt] Publishing topic ems-esp/tapwater_active (#41, attempt #1, pid 1)
000+00:02:21.285 D 89: [mqtt] Publishing topic ems-esp/heating_active (#42, attempt #1, pid 1)
000+00:02:24.262 D 90: [emsesp] Received UBAMonitorFast
000+00:02:25.260 D 91: [emsesp] Received UBAMonitorFast
000+00:02:26.290 D 92: [emsesp] Received UBAMonitorFast
000+00:02:27.319 D 93: [emsesp] Received UBAMonitorFast
000+00:02:28.320 D 94: [emsesp] Received UBAMonitorFast
000+00:02:29.722 D 95: [emsesp] Received UBAMonitorFast
000+00:02:29.962 D 96: [emsesp] Received UBAMonitorWW
000+00:02:30.136 D 97: [mqtt] Publishing topic ems-esp/boiler_data (#43, attempt #1, pid 1)
000+00:02:30.337 D 98: [mqtt] Publishing topic homeassistant/climate/ems-esp/hc1/state (#44, attempt #1, pid 1)
[... cut ...]
000+00:02:39.705 D 105: [emsesp] Received UBAMonitorFast
000+00:02:39.944 D 106: [emsesp] Received UBAMonitorWW
000+00:02:40.138 E 107: [telegram] Rx: FF FC FE FE FF 13 (CRC 13 != 11)
000+00:02:40.189 D 108: [mqtt] Publishing topic ems-esp/boiler_data (#45, attempt #1, pid 1)
000+00:02:40.390 D 109: [mqtt] Publishing topic homeassistant/climate/ems-esp/hc1/state (#46, attempt #1, pid 1)
```
not sure if it matters but I have the gateway connected to the thermostat terminals and I tried setting Tx Mode: 3

proddy, 2020-10-06:
from the logs I see the command is not being executed. I tested in the standalone version and it works there, so not sure why on your system. Can you do a `show` from the root of the Console to list the thermostat details. This is what `call thermostat temp 18` should look like in the console (notice the yellow line):

[image]

zanerv, 2020-10-06:
updated to version 2.1.0b4
```
ems-esp:/# show devices
These EMS devices are currently active:

Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:15.07)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 

Thermostat: FR120 (DeviceID:0x10, ProductID:191, Version:55.01) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x16F 0x165 0x170 0x166 0x171 0x167 0x172 0x168 

Controller: HT3 (DeviceID:0x09, ProductID:95, Version:15.07)

ems-esp:/# log debug
Log level = debug
000+00:13:50.170 D 7: [mqtt] Publishing topic ems-esp/boiler_data (#264, attempt #1, pid 1)
000+00:13:50.271 D 8: [mqtt] Publishing topic ems-esp/thermostat_data (#265, attempt #1, pid 1)
000+00:13:50.395 E 9: [telegram] Rx: 3E 87 F3 15 (CRC 15 != 1C)
000+00:13:50.765 D 10: [emsesp] Received UBAMonitorWW
ems-esp:/# call thermostat temp 15
000+00:13:57.408 I 11: [thermostat] Setting thermostat temperature to 15.0 for heating circuit 1, mode auto
000+00:13:57.508 D 12: [telegram] Sending write Tx [#205], telegram: 8B 10 FF 0F 00 65 1E 10
000+00:13:57.533 D 13: [emsesp] Last Tx write successful
000+00:13:57.533 D 14: [telegram] Sending post validate read, type ID 0x16F to dest 0x10
000+00:13:57.633 D 15: [telegram] Sending read Tx [#206], telegram: 8B 90 FF 00 20 00 6F DA
000+00:13:58.040 D 16: [telegram] Sending read Tx [#207], telegram: 8B 90 FF 00 20 00 6F DA
000+00:13:58.476 D 17: [telegram] Sending read Tx [#208], telegram: 8B 90 FF 00 20 00 6F DA
000+00:13:58.851 D 18: [telegram] Sending read Tx [#209], telegram: 8B 90 FF 00 20 00 6F DA
000+00:13:59.131 E 19: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
000+00:14:00.018 D 20: [emsesp] Fetching values for device ID 0x08
000+00:14:00.018 D 21: [telegram] Tx read request to device 0x08 for type ID 0x14
000+00:14:00.018 D 22: [telegram] Tx read request to device 0x08 for type ID 0x16
000+00:14:00.018 D 23: [telegram] Tx read request to device 0x08 for type ID 0x19
000+00:14:00.018 D 24: [telegram] Tx read request to device 0x08 for type ID 0x33
000+00:14:00.018 D 25: [emsesp] Fetching values for device ID 0x09
000+00:14:00.018 D 26: [emsesp] Fetching values for device ID 0x10
000+00:14:00.018 D 27: [telegram] Tx read request to device 0x10 for type ID 0x16F
000+00:14:00.188 D 28: [mqtt] Publishing topic ems-esp/boiler_data (#266, attempt #1, pid 1)
000+00:14:00.289 D 29: [mqtt] Publishing topic ems-esp/thermostat_data (#267, attempt #1, pid 1)
000+00:14:00.320 D 30: [emsesp] Received UBAMonitorFast
000+00:14:00.622 D 31: [emsesp] Received UBAMonitorWW
000+00:14:00.847 D 32: [telegram] Sending read Tx [#210], telegram: 8B 88 14 00 20 2C
000+00:14:01.098 D 33: [mqtt] Publishing topic ems-esp/heartbeat (#268, attempt #1, pid 1)
000+00:14:01.253 D 34: [telegram] Sending read Tx [#215], telegram: 8B 88 14 00 20 2C
000+00:14:01.627 D 35: [telegram] Sending read Tx [#216], telegram: 8B 88 14 00 20 2C
000+00:14:02.063 D 36: [telegram] Sending read Tx [#217], telegram: 8B 88 14 00 20 2C
000+00:14:02.376 E 37: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
000+00:14:02.469 D 38: [telegram] Sending read Tx [#211], telegram: 8B 88 16 00 20 24
000+00:14:02.843 D 39: [telegram] Sending read Tx [#218], telegram: 8B 88 16 00 20 24
000+00:14:03.249 D 40: [telegram] Sending read Tx [#219], telegram: 8B 88 16 00 20 24
000+00:14:03.623 D 41: [telegram] Sending read Tx [#220], telegram: 8B 88 16 00 20 24
000+00:14:03.967 E 42: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
000+00:14:04.029 D 43: [telegram] Sending read Tx [#212], telegram: 8B 88 19 00 20 18
000+00:14:04.409 D 44: [telegram] Sending read Tx [#221], telegram: 8B 88 19 00 20 18
000+00:14:04.809 D 45: [telegram] Sending read Tx [#222], telegram: 8B 88 19 00 20 18
000+00:14:05.214 D 46: [telegram] Sending read Tx [#223], telegram: 8B 88 19 00 20 18
000+00:14:05.526 E 47: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
000+00:14:05.588 D 48: [telegram] Sending read Tx [#213], telegram: 8B 88 33 00 20 B0
000+00:14:05.994 D 49: [telegram] Sending read Tx [#224], telegram: 8B 88 33 00 20 B0
000+00:14:06.368 D 50: [telegram] Sending read Tx [#225], telegram: 8B 88 33 00 20 B0
000+00:14:06.774 D 51: [telegram] Sending read Tx [#226], telegram: 8B 88 33 00 20 B0
000+00:14:07.086 E 52: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
000+00:14:07.211 D 53: [telegram] Sending read Tx [#214], telegram: 8B 90 FF 00 20 00 6F DA
000+00:14:07.561 D 54: [emsesp] Received JunkersMonitor
000+00:14:07.582 D 55: [mqtt] Publishing topic ems-esp/thermostat_data (#269, attempt #1, pid 1)
000+00:14:07.835 D 56: [telegram] Sending read Tx [#227], telegram: 8B 90 FF 00 20 00 6F DA
000+00:14:08.241 D 57: [telegram] Sending read Tx [#228], telegram: 8B 90 FF 00 20 00 6F DA
000+00:14:08.615 D 58: [telegram] Sending read Tx [#229], telegram: 8B 90 FF 00 20 00 6F DA
000+00:14:08.901 E 59: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
000+00:14:10.212 D 60: [mqtt] Publishing topic ems-esp/boiler_data (#270, attempt #1, pid 1)
000+00:14:10.313 D 61: [mqtt] Publishing topic ems-esp/thermostat_data (#271, attempt #1, pid 1)
000+00:14:10.521 D 62: [emsesp] Received UBAMonitorFast
[... cut ...]
000+00:14:41.526 E 76: [telegram] Rx: 88 00 19 00 80 00 80 00 80 00 FF FF 00 64 04 52 C5 05 EB F8 00 00 00 04 F9 9C 03 76 FB 14 (CRC 14 != CD)
000+00:14:49.890 D 77: [emsesp] Received UBAMonitorFast
000+00:14:50.177 D 78: [mqtt] Publishing topic ems-esp/boiler_data (#278, attempt #1, pid 1)
000+00:14:50.278 D 79: [mqtt] Publishing topic ems-esp/thermostat_data (#279, attempt #1, pid 1)
000+00:14:50.412 D 80: [emsesp] Received UBAMonitorWW
000+00:14:50.604 E 81: [telegram] Rx: FE 87 3E FC FC FC FE FF 0F (CRC 0F != 67)
```

proddy, 2020-10-06:
did it work?

zanerv, 2020-10-06:
no, it didn't
```
ems-esp:/# show
EMS-ESP version 2.1.0b4

Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:15.07)
  Heating Active: on
  Warm water/DHW Active: off
  Service Code: 
  Service Code Number: 0
  Warm water selected temperature: 49°C
  Warm water set temperature: 49°C
  Warm water disinfection temperature: 70°C
  Selected flow temperature: 57°C
  Burner selected max power: 100%
  Burner current power: 21%
  Pump modulation: 100%
  Warm water type: flow
  Warm Water charging type: 3-way valve
  Warm water circulation pump available: off
  Warm water circulation pump freq: 2x3min
  Warm water circulation active: off
  Warm water current temperature (intern): 26.1°C
  Warm water current tap water flow: 0.0l/min
  Current flow temperature: 61.5°C
  Warm water storage temperature (intern): 26.1°C
  Warm water activated: on
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: off
  Warm water recharge: off
  Warm water temperature ok: on
  Warm water active: off
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Warm water heating: off
  Heating activated: on
  Heating temperature setting on the boiler: 57°C
  Boiler circuit pump modulation max power: 100%
  Boiler circuit pump modulation min power: 10%
  Boiler circuit pump delay time: 3min
  Boiler burner min period: 3min
  Boiler burner min power: 0%
  Boiler burner max power: 100%
  Boiler temp hysteresis on: -5°C
  Boiler temp hysteresis off: 0°C
  Warm water # starts: 56269
  Warm water active time: 62048
  Burner # starts: 283423
  Warm Water active time: 43 days 2 hours 8 minutes
  Total burner operating time: 269 days 13 hours 10 minutes
  Total heat operating time: 226 days 11 hours 1 minutes
  Total UBA working time: 2712 days 17 hours 52 minutes

Thermostat: FR120 (DeviceID:0x10, ProductID:191, Version:55.01)
  Time: 21:32:25 06/10/2020
Heating Circuit 1:
  Setpoint room temperature: 23.0°C
  Current room temperature: 22.1°C
  Mode: auto
  Mode type: eco
```

proddy, 2020-10-06:
strange. I see the command being sent `8B 10 FF 0F 00 65 1E 10` which looks correct. Also a few Tx errors. What does `show ems` say?

zanerv, 2020-10-06:
I notices so far is that if I set Tx mode to 3 I get an tx write error and for 1&2 i get Last Tx write successful but the thermostat setpoint remains the same
```
ems-esp:/# show ems
EMS Bus is connected.

EMS Bus info:
  Tx mode: 2
  Bus protocol: HT3
  #telegrams received: 355
  #read requests sent: 13
  #write requests sent: 2
  #incomplete telegrams: 40 (11%)
  #tx fails (after 3 retries): 0

Rx Queue is empty

Tx Queue is empty
```

zanerv, 2020-10-07:
also when switching modes from thermostat I get 
`000+00:06:12.292 D 81: [emsesp] No telegram type handler found for ID 0x179 (src 0x10)`

proddy, 2020-10-07:
@zanerv bingo. That pointed me to the problem. The FR120 was configured to behave like the newer junkers FW* models and not the FR's which use different telegram types. I've made the change and it'll go into the next push today, 2.1.0b6

zanerv, 2020-10-07:
updated to the latest firmware(v2.1.0b6) and I still get
```
000+00:08:32.628 D 690: [emsesp] No telegram type handler found for ID 0x1D3 (src 0x10)
000+00:08:33.212 D 691: [emsesp] No telegram type handler found for ID 0x179 (src 0x10)
```

proddy, 2020-10-07:
`show devices` gives you? Hopefully you should see 0x179 shown in the line

```
Thermostat: FR120 (DeviceID:0x10, ProductID:191, Version:55.01) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x16F 0x165 0x170 0x166 0x171 0x167 0x172 0x168 
```

zanerv, 2020-10-07:
@proddy here is what I got
```
┌──────────────────────────────────────────┐
│ EMS-ESP version 2.1.0b6                  │
│ https://github.com/proddy/EMS-ESP        │
│                                          │
│ type help to show available commands     │
└──────────────────────────────────────────┘

ems-esp:/$ show devices
These EMS devices are currently active:

Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:15.07)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 

Thermostat: FR120 (DeviceID:0x10, ProductID:191, Version:55.01) ** master device **
 This Thermostat will respond to telegram type IDs: 0x16F 0x165 0x170 0x166 0x171 0x167 0x172 0x168 

Controller: HT3 (DeviceID:0x09, ProductID:95, Version:15.07)
```

proddy, 2020-10-08:
Found the bug. Working on a fix.

zanerv, 2020-10-08:
Nice! you're on to something, I got the new firmware and now I get
```
ems-esp:/# call thermostat temp 7
000+00:03:13.439 W 217: [thermostat] Set temperature: Invalid value
```
```
ems-esp:/# show devices
These EMS devices are currently active:

Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:15.07)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA

Thermostat: FR120 (DeviceID:0x10, ProductID:191, Version:55.01) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x16F 0x179 0x170 0x17A 0x171 0x17B 0x172 0x17C

Controller: HT3 (DeviceID:0x09, ProductID:95, Version:15.07)

```

proddy, 2020-10-08:
argggh!

zanerv, 2020-10-08:
bring out the the champagne, it's working!
I can set the temperature and the mode to auto/manual as for 'off' I haven't figured out what it does or if it's working (tested from homeassistant/telnet)
@proddy if you want to test something else let me know otherwise I'll close this
Thank you!

proddy, 2020-10-08:
great! You helped find and fix a bug so thanks

closed 2020-10-08 by proddy

juggler2019, 2020-12-29:
Hello

I do automation of turning on the heating on Junkers Cerapur and Fr120 thermostat. The service only forces heating without changing the temperature. 
........................................................................
service: climate.set_temperature
data:
  entity_id: climate.thermostat_hc1
  temperature: 24
  hvac_mode: heat

proddy, 2020-12-29:
@juggler2019 can you try the command from the console, with logging on so we can see what is actually happening? https://emsesp.github.io/docs/#/Troubleshooting?id=commands-via-mqtt-are-not-working

juggler2019, 2020-12-29:
000+01:43:36.186 N 38: [shell] su session opened on console
000+01:43:37.211 E 39: [telegram] Rx: 7C 0F E3 11 (CRC 11 != 14)
000+01:43:37.460 E 40: [telegram] Rx: FF C3 9F FE FF 0B (CRC 0B != 6D)
000+01:43:57.796 E 41: [telegram] Rx: 7C 0F E3 0B (CRC 0B != 14)
000+01:44:03.285 E 42: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
000+01:44:06.874 E 43: [telegram] Rx: D8 88 00 18 00 00 00 FE 46 00 00 12 00 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 38 (CRC 38 != 32)
ems-esp:/# log debug
Log level = debug
000+01:44:20.866 D 44: [mqtt] Publishing topic ems-esp/boiler_data (#1993, retain=0, try#1, size 519, pid 1)
000+01:44:21.067 D 45: [mqtt] Publishing topic ems-esp/boiler_data_ww (#1994, retain=0, try#1, size 409, pid 1)
000+01:44:21.268 D 46: [mqtt] Publishing topic ems-esp/thermostat_data (#1995, retain=0, try#1, size 157, pid 1)
000+01:44:27.113 E 47: [telegram] Rx: 7C 0F E3 53 (CRC 53 != 14)
000+01:44:27.394 E 48: [telegram] Rx: FE 87 3E FE FE FF 10 (CRC 10 != 56)
000+01:44:29.673 D 49: [emsesp] No telegram type handler found for ID 0xBE (src 0x10)
000+01:44:29.984 D 50: [emsesp] No telegram type handler found for ID 0x1D3 (src 0x10)
000+01:44:30.923 D 51: [mqtt] Publishing topic ems-esp/boiler_data (#1996, retain=0, try#1, size 519, pid 1)
000+01:44:31.124 D 52: [mqtt] Publishing topic ems-esp/boiler_data_ww (#1997, retain=0, try#1, size 409, pid 1)
000+01:44:31.325 D 53: [mqtt] Publishing topic ems-esp/thermostat_data (#1998, retain=0, try#1, size 157, pid 1)
000+01:44:37.344 E 54: [telegram] Rx: FF F8 FE FE 4F (CRC 4F != 67)
000+01:44:40.980 D 55: [mqtt] Publishing topic ems-esp/boiler_data (#1999, retain=0, try#1, size 519, pid 1)
000+01:44:41.182 D 56: [mqtt] Publishing topic ems-esp/boiler_data_ww (#2000, retain=0, try#1, size 409, pid 1)
000+01:44:41.383 D 57: [mqtt] Publishing topic ems-esp/thermostat_data (#2001, retain=0, try#1, size 157, pid 1)
ems-esp:/# call thermostat temp 15
000+01:44:42.489 I 58: [thermostat] Setting thermostat temperature to 15.0 for heating circuit 1, mode auto
000+01:44:42.585 E 59: [telegram] Rx: 90 08 1A 1F F1 FF 98 0C (CRC 0C != A7)
000+01:44:42.615 D 60: [telegram] Sending write Tx [#92], telegram: 8B 10 FF 07 00 79 1E 68
000+01:44:42.645 D 61: [emsesp] Last Tx write successful
000+01:44:42.645 D 62: [telegram] Sending post validate read, type ID 0x16F to dest 0x10
000+01:44:43.301 D 63: [telegram] Sending read Tx [#93], telegram: 8B 90 FF 00 20 00 6F DA
000+01:44:43.359 D 64: [emsesp] Last Tx read successful
000+01:44:43.396 D 65: [mqtt] Publishing topic ems-esp/thermostat_data (#2002, retain=0, try#1, size 157, pid 1)
000+01:44:43.645 E 66: [telegram] Rx: FE FC FE FE 10 (CRC 10 != 7F)
000+01:44:47.075 E 67: [telegram] Rx: 70 3F 83 FB 11 (CRC 11 != B3)
000+01:44:50.836 D 68: [mqtt] Publishing topic ems-esp/boiler_data (#2003, retain=0, try#1, size 519, pid 1)
000+01:44:51.040 D 69: [mqtt] Publishing topic ems-esp/boiler_data_ww (#2004, retain=0, try#1, size 409, pid 1)
000+01:44:51.241 D 70: [mqtt] Publishing topic ems-esp/thermostat_data (#2005, retain=0, try#1, size 157, pid 1)
000+01:45:00.123 D 71: [emsesp] Fetching values for device ID 0x08
000+01:45:00.123 D 72: [telegram] Tx read request to device 0x08 for type ID 0x14
000+01:45:00.123 D 73: [telegram] Tx read request to device 0x08 for type ID 0x16
000+01:45:00.123 D 74: [telegram] Tx read request to device 0x08 for type ID 0x19
000+01:45:00.123 D 75: [telegram] Tx read request to device 0x08 for type ID 0x33
000+01:45:00.123 D 76: [emsesp] Fetching values for device ID 0x09
000+01:45:00.123 D 77: [emsesp] Fetching values for device ID 0x10
000+01:45:00.123 D 78: [telegram] Tx read request to device 0x10 for type ID 0x16F
000+01:45:00.123 D 79: [telegram] Tx read request to device 0x10 for type ID 0x179
000+01:45:00.174 D 80: [telegram] Sending read Tx [#94], telegram: 8B 88 14 00 20 2C
000+01:45:00.830 D 81: [telegram] Sending read Tx [#100], telegram: 8B 88 14 00 20 2C
000+01:45:00.868 D 82: [emsesp] Last Tx read successful
000+01:45:00.897 D 83: [mqtt] Publishing topic ems-esp/boiler_data (#2006, retain=0, try#1, size 519, pid 1)
000+01:45:01.048 D 84: [telegram] Sending read Tx [#95], telegram: 8B 88 16 00 20 24
000+01:45:01.098 D 85: [emsesp] Last Tx read successful
000+01:45:01.098 D 86: [mqtt] Publishing topic ems-esp/boiler_data_ww (#2007, retain=0, try#1, size 409, pid 1)
000+01:45:01.299 D 87: [mqtt] Publishing topic ems-esp/thermostat_data (#2008, retain=0, try#1, size 157, pid 1)
000+01:45:01.459 D 88: [telegram] Sending read Tx [#96], telegram: 8B 88 19 00 20 18
000+01:45:01.501 D 89: [mqtt] Publishing topic ems-esp/heartbeat (#2009, retain=0, try#1, size 145, pid 1)
000+01:45:01.827 D 90: [telegram] Sending read Tx [#101], telegram: 8B 88 19 00 20 18
000+01:45:02.420 D 91: [emsesp] Last Tx read successful
000+01:45:02.420 E 92: [telegram] Rx: 88 0B 19 00 80 00 80 00 80 00 FF FF 00 00 00 45 3F 06 F2 D0 00 00 00 06 0B 8A 00 40 7B 10 (CRC 10 != 5D)
000+01:45:02.794 D 93: [telegram] Sending read Tx [#97], telegram: 8B 88 33 00 20 B0
000+01:45:02.844 D 94: [emsesp] Last Tx read successful
000+01:45:03.206 D 95: [telegram] Sending read Tx [#98], telegram: 8B 90 FF 00 20 00 6F DA
000+01:45:03.265 D 96: [emsesp] Last Tx read successful
000+01:45:03.636 D 97: [telegram] Sending read Tx [#99], telegram: 8B 90 FF 00 20 00 79 CC
000+01:45:03.699 D 98: [emsesp] Last Tx read successful
[... cut ...]
000+01:45:46.466 E 112: [telegram] Rx: FF FF 88 13 05 22 00 80 (CRC 80 != AB)
000+01:45:47.396 E 113: [telegram] Rx: 38 9F C3 F6 53 (CRC 53 != FE)
000+01:45:48.020 E 114: [telegram] Rx: 7C 0F E3 10 (CRC 10 != 14)
000+01:45:51.011 D 115: [mqtt] Publishing topic ems-esp/boiler_data (#2022, retain=0, try#1, size 519, pid 1)
000+01:45:51.212 D 116: [mqtt] Publishing topic ems-esp/boiler_data_ww (#2023, retain=0, try#1, size 409, pid 1)
000+01:45:51.413 D 117: [mqtt] Publishing topic ems-esp/thermostat_data (#2024, retain=0, try#1, size 157, pid 1)

juggler2019, 2020-12-29:
Via telnet works fine but by calling a service in HA it does not change temperature only the heat mode

proddy, 2020-12-29:
things to try
1) with console on and `log info` change the temperature in HA and see if something pops up like "`[thermostat] Setting thermostat temperature to 15.0 for heating circuit 1, mode auto`"
2) using MQTTExplorer see which payloads and topics are being published
3) make sure you have Home Assistant checked as MQTT format