emsesp/EMS-ESP #541: Connection issue

czykwin, 2020-10-08:
@bbqkees I've just connected brand new gateway for the first time and it's not working.
No EMS devices found. 129 (Rx) Incomplete telegrams and growing (0 others).
Connected by service jack to the Buderus GB072 (with Logamatic webKM200 and Logamatic RC300).
I've no idea what is wrong, please help.

czykwin, 2020-10-08:
v2.0.1
[image]
[image]

bbqkees, 2020-10-08:
Hi,
Can you open the Gateway, remove the white plastic side of the jack connector and reinsert the plug?
Some of the cables I got recently have a very slightly thicker plug and do not fit properly in the recessed hole in front of the jack causing the plug to misalign and not make proper contact.

czykwin, 2020-10-08:
The housing was not twisted at all, I enlarged the jack hole and now it's ok :)

closed 2020-10-08 by czykwin