emsesp/EMS-ESP #542: remove MQTT individual topics heating_active and tapwater_active, as they're included in boiler_data

proddy, 2020-10-09:
Originally I used individual MQTT topics for signaling whether the heating and/or hot water was on or off. This was to allow Boolean sensors in Home Assistant. Now with the latest HA MQTT Discovery implemented this data is already captured in the `boiler_data` topic and picked up as a separate entity. 

Ok to remove it? Anyone using it?

@MichaelDvP @bbqkees

bbqkees, 2020-10-09:
On one hand it does not make sense to have them in two places.
On the other hand its not a direct boiler parameter but one EMS-ESP makes. 

I had one guy who only wanted a Gateway so he could trigger on heating_active to switch on his floor heating circulation pump.
But it's only 3 additional lines of code to look into boiler_data instead. But also for just looking at heating_active you don't need the json library so that makes the program simpler....

If it doesn't really hurt memory wise you could leave it as it is now.
We always have to look beyond other uses than HA as well.

MichaelDvP, 2020-10-09:
The difference is: the individual topics are always publish on change, the large boiler_data mostly scheduled. Setting boiler to publish on change gives a lot of messages.
I'm not using it, but there are some discussions last time showing that others using it. Better do not remove.

@bbqkees 
> On the other hand its not a direct boiler parameter but one EMS-ESP makes.

No, it's now a boiler parameter, we've found that the boiler publishes these states directly as bits in message 0x18, byte 5.

bbqkees, 2020-10-09:
So all the reasons to keep it in then.

proddy, 2020-10-09:
ok, they stay! thanks.

closed 2020-10-09 by proddy