emsesp/EMS-ESP #545: measure temperature after update fw to v2.1.0b6 to not work

Bartekn86, 2020-10-09:
Hi, 
after update fw to b6 temperature measuring ftom boiler do not want to refresh. showing static temperature gb072v2
Please help, 
thank you

proddy, 2020-10-09:
from which version did you upgrade from? Which version of the domoticz plugin are you using? Do you see the values changing in the console, web and MQTT? Please be more specific so we can help

Bartekn86, 2020-10-09:
upodated from b3  domoticz plugin latest dev-2, value changing in web, (dont know how to check console or mqtt).

MichaelDvP, 2020-10-09:
Ah, seems the same rename problem i get a few days ago in my scripts: `wWCurTmp` is now corrected to `wWCurTemp`. 
@bbqkees: can you adapt this in [your plugin](https://github.com/bbqkees/ems-esp-domoticz-plugin/blob/master/plugin.py)

Bartekn86, 2020-10-09:
MichaelDvP thank you for your time and help :-)

bbqkees, 2020-10-09:
@MichaelDvP Yes I got a few comments on that one already. Had it changed locally for the next update but I'll change it right away in the repository.

closed 2020-10-09 by Bartekn86