emsesp/EMS-ESP #546: how to change boilhystoff? (Junkers Cerapur / CW400)

lsw271, 2020-10-09:
**Question**
How can I change boilhystoff? In my boiler I have only one hysteresis in mqtt payload its a boilHystOn - when i changed values in boiler this sensor had changed. In mqtt explorer in boilHystOff I've got 6 degress, but from where it is? :) My boiler really work with boilHystOff (i've got template sensor in HA which shows me that) but I don't know where I can change this :) 
I tried to change this value from mqtt explorer / telnet but nothing happens. 

**Screenshots**
[image]

**Device information**
boilhystoff
[image]

boilhyston (its working)
[image]

**Additional context**
My Junkers installer dont know why setting hysteresis on boiler doesnt work (I always change boilhyston but boiler working with bolihystoff), but ems-esp shows that i have one more hysteresis :D

MichaelDvP, 2020-10-10:
I think the boiler or the thermostat sets the values back: After the change you receive two `UBAParameters` messages, (last picture D5, D7) before validating by ems-esp (D10, D12). You can use `watch on 16` to see what device overwrites your setting.
I expext D5 is the acknowledge from the boiler to your change and D7 is the setback.

lsw271, 2020-10-10:
@MichaelDvP  I hope you will find something there :) 
[image]

MichaelDvP, 2020-10-11:
Sorry, looks like `boilhystoff` is not writable for your boiler (byte 4 is always 06). With my boiler it works. 
But `boilhyston` works on your boiler, byte 5 changes from F8 (-8) to F7 (-9).

lsw271, 2020-10-11:
> 
> 
> Sorry, looks like `boilhystoff` is not writable for your boiler (byte 4 is always 06). With my boiler it works.
> But `boilhyston` works on your boiler, byte 5 changes from F8 (-8) to F7 (-9).

It's too bad. I really need to change this parameter :/

closed 2020-10-11 by lsw271

lsw271, 2020-11-04:
@MichaelDvP 
I tried again with latest stable released version 2.1. Now i tried to change boilhystoff to lower value than defualt 6 degrees. Maybe you will find someting interesting - new value saved correctly. But when I was trying to change value to higher than 6 degrees nothing happened as earlier

[image]