emsesp/EMS-ESP #547: outdoorTemp not reported anymore in 2.0

frankhommers, 2020-10-10:
I don't see outdoorTemp anywhere anymore in MQTT since v2.0.

Through syslog I see that EMS-ESP receives it:
```
Time: Oct 10 15:09:09
Host: 1
Facility: logalert
Priority: debug
Message: [emsesp] Received RCOutdoorTemp
```

But that doesn't end up in a MQTT message anywhere.

[image]

[image]

proddy, 2020-10-10:
Can you try 2.1. Now all values are both in web and MQTT

frankhommers, 2020-10-10:
Do you mean 2.0.1? This is what I was using... Or else: When can I get the 2.1 binary? :-P

proddy, 2020-10-10:
It's in https://github.com/proddy/EMS-ESP/tree/firmware

I'm planning to release 2.1 soon, also with the update documentation

closed 2020-10-11 by proddy