emsesp/EMS-ESP #548: Support for Bosch Condens GC2300W.

hetii, 2020-10-10:
Hello dear. I'm owner of Bosch Condens GC2300W.

Is this model supported by ems-esp?
What kind of functionality I can have when connecting it to ems-esp?

Best Regards.

MichaelDvP, 2020-10-13:
You can check the tested devices on [bbqkees page](https://bbqkees-electronics.nl/ems-device-compatibility/) 
But also if the device is not on the list, it may work if ems compatible (or can added to ems-esp with your help).
For the functionality consult the wikis here and from bbqkees and the documentation of your favorit home automation system. 
Also if you miss a setting or a value you can ask here to integate it.

bbqkees, 2020-10-13:
@hetii If I check the control panel of your boiler it looks exactly the same as of the Condens 5000i. So it will likely work.
The functionality depends a bit on the specific boiler and the thermostat but usually you can read out 30+ parameters of the boiler and f.i. change the thermostat setpoint or change the warm water mode of the boiler.

hetii, 2020-10-13:
@bbqkees ok but something is not clear for me regarding esp-ems. 
So I have my Condens 2300w, do I need then any additional hardware beside esp-ems to control it like cw400 or can I do it entirely by esp-ems and Home Assistant way?

bbqkees, 2020-10-13:
No its not encrypted.
But it is encoded into specific telegrams of which there is no public datasheet.
But most of those telegrams that have now been reverse engineered. And that's why we know how it works.

F.i. the TADO thermostat is a third party thermostat that can communicate via the EMS protocol.

You cannot use EMS-ESP as a thermostat replacement.
If you have an EMS thermostat EMS-ESP can change the setpoint of the thermostat but EMS-ESP cannot directly modulate the burner. That's still the job of the thermostat.

glitter-ball, 2020-10-13:
The thermostat decides the flow temperature required from the boiler based on : inside temperature, outside temperature, heating characteristic curve, building type etc.

The thermostat does these calculations continually. So, while you can set the flow temperature with EMS-ESP, something has to do the calculations.

I see no reason why you couldn’t do it with another system (with the necessary sensors and algorithm) and then pass a calculated flow temp to EMS-ESP. But it could be a lot of work to implement time programmes, optimum start (i.e. fire earlier in cold weather to meet occupancy time) etc.

hetii, 2020-10-13:
Hmm so if I understand correctly, esp-ems has knowledge how to start ignition and general all commands to act as thermostat but has no thermostat logic? In that case my plan was to use home assistance and has all such logic there if possible as this will be my center place that will gather all sensors data in house.

glitter-ball, 2020-10-13:
I’m not sure how you’d get HA to calculate flow temp, do time programmes etc. Also not sure if EMS-ESP can start the boiler or only change settings once something else has started it.

Maybe a good test is to see if you can start and stop the boiler by sending a high and low flow temperature. But... there are then pumps and valves to manage too.

It would be a lot easier with a thermostat - even a basic one - and probably more reliable.

bbqkees, 2020-10-13:
The commands to turn on the burner are known from observing the bus.
However, it's the same as everybody knows how a car looks like but almost nobody knows and can put the effort into actually building one.
It's just a much better option to let a robust thermostat deal with that stuff.

glitter-ball, 2020-10-13:
@bbqkees  - that is good advice.

I've been involved in a few custom heating control systems over the years and, yes, it's great being able to code and build your own system to do clever things. But... when it all goes wrong and a few hundred people have no heating or hot water it's not such a great day!! Now, my family would kill me if I broke heating and hot water at home.

Having a thermostat reliably do the essential stuff with cool things like  EMS-ESP / HA to tweak it is brilliant. You can do extra stuff if you want, but if you have a bad week and it all breaks, you still have a habitable home,

hetii, 2020-10-14:
Ok I understand the risk about using experimental project like this one to control boilers. But In [fork](https://github.com/shreknel/EMS-ESP-Boiler) of this repository author claim that his original intention was to build his own smart thermostat.
Now you claim it's not the case anymore that's why it's not clear for me for what this project is beside exposing existing theromstat over wifi. Ok this gives some benefits when someone forget to setup vacation mode and will turn it on over internet but still I thought that the main idea was to have replacement of existing product with extra capabilities.
That's why I ask with part of the system is responsible for what as the border is blurry.

proddy, 2020-10-16:
@hetii I'm the original author. EMS-ESP (previously called Boiler back in v1) was intended to be a bridge to decode EMS messages running on an ESP8266. Hence the name. I built it so I could control my existing dumb thermostat manually using my mobile phone and also via triggers in Home Assistant. Then I started add features to set boiler settings. It was never intended to be an actual thermostat. If I had solved that problem I'd be a rich man now (and certainly would not share my code with anyone else!).