emsesp/EMS-ESP #549: HA -  incorrect value of the sensor.nofrost_temperature

majdzik84, 2020-10-10:
**In home assistant, sensor.nofrost_temperature is entered as half the correct value**

[image]

**Let me write on the same topic. I hope that this is not a problem.**

I also noticed that the HA detected the EMS-ESP Mixing entities, however the status is unknown

[image]

proddy, 2020-10-10:
thanks for reporting, we'll look into it.

proddy, 2020-10-11:
@MichaelDvP the code assumes the `nofrosttemp` value is *2, except in the RC35 it takes *1.

https://github.com/proddy/EMS-ESP/blob/ab3e4731a4a2a2a2eb98640fff98f0f0e23360b5/src/devices/thermostat.cpp#L1232

proddy, 2020-10-11:
@majdzik84 for the mixing entities can you send what the MQTT payloads look like (use MQTTExplorer). In our tests it works. Also in HA the mixing values are sent to a separate homeassistant device so check the values there too (HA->Configuration->Devices->EMS-ESP Mixing)

majdzik84, 2020-10-11:
@proddy I am uploading screenshots. I see values ​​in mqtt explorer but not in HA. I tried to remove the device and the topic was still in the same form.

[image]

[image]

proddy, 2020-10-11:
thanks. I think I know where the problem is now.

MichaelDvP, 2020-10-11:
>  the code assumes the nofrosttemp value is *2, except in the RC35 it takes *1.

Right, Junkers is *2, RC35 is *1, settings are all ok, we need only change `export_values_hc` to:
```
            // Nofrost temperature
            if (Helpers::hasValue(hc->nofrosttemp)) {
                if (flags == EMSdevice::EMS_DEVICE_FLAG_JUNKERS) {
                    dataThermostat["nofrosttemp"] = (float)hc->nofrosttemp / 2;
                } else {
                    dataThermostat["nofrosttemp"] = hc->nofrosttemp;
                }
            }
```

The mixing is due to publish topic. The values are published as `mixing_data2`, but the HA-config sets `mixing_data`  in stat_t:
```
{"name":"Setpoint flow temperature","uniq_id":"mixing_flowSetTemp","unit_of_meas":"°C","stat_t":"ems/mixing_data","val_tpl":"{{value_json.flowSetTemp}}","ic":"mdi:coolant-temperature","dev":{"ids":["ems-esp-mixing"]}}
```
But change the register_HA is a lot of work, maybe it's easier to make a nested mixing_data with combined devices as in api-info.

BTW: But mixing is the only HA-config that uses hostname (here `ems`), all other configs are fixed to `ems-esp`, this could also make problems.

proddy, 2020-10-11:
> But change the register_HA is a lot of work, maybe it's easier to make a nested mixing_data with combined devices as in api-info.

Nested is going to hard too because each unit is a separate device (instance class in EMS-ESP). Which makes me think there could be multiple mixings devices in HA. I need to think about this.

> BTW: But mixing is the only HA-config that uses hostname (here ems), all other configs are fixed to ems-esp, this could also make problems.

good point. Forgot about that. I'll make a change.

MichaelDvP, 2020-10-11:
> Nested is going to hard too because each unit is a separate device 

I think it's easy, since it is in for api-info. The json for nested mixers has to be declared and published in emsesp- publish_device_values and call export_values to fill the nests. For single format call publish_values() for each device instead.

proddy, 2020-10-11:
> I think it's easy, since it is in for api-info. The json for nested mixers has to be declared and published in emsesp- publish_device_values and call export_values to fill the nests. For single format call publish_values() for each device instead.

yes, that would work.  If the format is nested moved the publishing out of mixing.cpp to emsesp.cpp. You'll need to do some clever logic in process_telegram() to keep it clean.

MichaelDvP, 2020-10-11:
I've pushed a first test my repo, compiles but not all tested yet. Take a look.

proddy, 2020-10-11:
@MichaelDvP looks good. I tested and works fine as well. I would have implemented it the same way. The `JsonObject dummy` is not pretty but I couldn't find a nicer way either. So good to go.

proddy, 2020-10-11:
now in 2.1.0b7. Thanks again Michael.

proddy, 2020-10-11:
@MichaelDvP just did another test and I'm not seeing the nested values. But I'm using the standalone version to test (no ESP8266) so I could be my setup. One thing to note is that we need to make sure each Mixing device is independently created in Home Assistant. Now they will be overwritten as they are both have the same unique_ids (mixing).

majdzik84, 2020-10-11:
EMS-ESP-2_1_0b7

[image]
[image]

proddy, 2020-10-11:
@majdzik84 in MQTTExplorer can you show the value of the topics homeassistant/sensor/ems-esp/mixing*

majdzik84, 2020-10-11:
[image]

[image]

[image]

MichaelDvP, 2020-10-11:
I've made some changes and pushed to my repo, but i'm not very familiar with HA. Does this look right?
[image]

proddy, 2020-10-11:
yes. the val_tpl needs to match the nested format in mixing_data. Before mixing_data was using hc1 and the HA config topics were using wwc2

MichaelDvP, 2020-10-13:
@majdzik84 I think nofrost is ok and mixing mqtt is also ok now. Have you checked the HA config? If it works, close this issue.

majdzik84, 2020-10-13:
Yes, the values ​​from my mixer showed up. I did not pay attention to them because more entities appeared. Well done. Thank you.

[image]

closed 2020-10-13 by majdzik84