emsesp/EMS-ESP #550: Design temperature value for HC2 is incorrect

majdzik84, 2020-10-10:
EMS-ESP version 2.1.0b6 

I noticed that the value for HC2 Design temperature for HC2 is incorrect.
The temperature does not change with thermostat control (RC35).

[image]
[image]

With HC1, this is not a problem. The temperature is indicated correct and reacts to changes.

proddy, 2020-10-10:
@MichaelDvP any idea to why this is happening?

MichaelDvP, 2020-10-11:
I'm not sure. I have also hc2 and it works, I suspect that the difference is the heating type, i use radiators and this low temperature seems to be floor heating. The RC35 has two fields for desigtemp, possible that the second is for floorheating. 
@majdzik84 Can you go to terminal and check:
`read 10 47` and `send telegram "0B 90 47 1B 20"` (i will add offset_optional to read command next time).

majdzik84, 2020-10-11:
@MichaelDvP  For me HC1 is radiators, HC2 floor heating.
```
ems-esp:/# read 10 47
000+01:50:10.014 N 1: [emsesp] Thermostat(0x10) -> (0x0B), (0x47), data: 03 2A 2A 26 08 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 0D 04 05 02 02

ems-esp:/# send telegram "0B 90 47 1B 20"
000+01:51:35.236 N 2: [emsesp] Thermostat(0x10) -> (0x0B), (0x47), data: 00 01 28 19 0A 03 00 FF 2B 25 00 F1 05 05 02 (offset 27)

```

MichaelDvP, 2020-10-11:
That fits: Pos 17 is 0x4B (75°C) and pos 36 is 0x25 (37°C). We have to pick up pos 36 if floor heating. I'll make a change.

majdzik84, 2020-10-11:
2.1.0b7 - Design temperature missing

[image]

MichaelDvP, 2020-10-11:
Oh, the problem of high offsets, as workaround try from terminal `read 10 47 1B` or change the value, than it will appear.

majdzik84, 2020-10-11:
MichaelDvP  changing the value helped. now I can see and change. Thank you very much.

majdzik84, 2020-10-12:
@MichaelDvP design temperature disappears after each device restart and appears only when I change it. Can you help it?

MichaelDvP, 2020-10-12:
@majdzik84 Yes, i know, it was a workaround until i find a solution that works for every value in the higher telegram regions. 
I think i found this solution now. Update and check, please.

majdzik84, 2020-10-12:
Temperature now shows up after restart! good job. Thank you!

closed 2020-10-12 by majdzik84