emsesp/EMS-ESP #553: HA - binary_sensor.heating_active not working

majdzik84, 2020-10-12:
I'd like to report that binary_sensor.heating_active  unfortunately it does not change the state in HA. Always shows off
[image]

[image]

proddy, 2020-10-12:
I noticed that too and fixed it during the weekend. At least I think I did. Are you on the latest dev build, b7?

majdzik84, 2020-10-12:
Yes, I have the latest version 2.1.0b7

MichaelDvP, 2020-10-13:
I'm not sure and have no HA to test. But the mqtt is `heating_active = on` and the configuration is `"payload_on": "on"` with quotation marks. I didn't find exact information about configuration of binary sensors, but i read that binary sensors will react on on/off, true/false and 1/0, What happen if you skip the `payload_` from config and leave it to HA to interprete the value?

MichaelDvP, 2020-10-13:
Since these are the only binary sensors and not working i tried to remove the "payload"-setting from config. 
@majdzik84 please give feedback if it's working or give errors, than i revert this.

majdzik84, 2020-10-13:
I must apologize to you very much. The corrected version does not work. The previous one works for it! (SORRY) so you should undo the change. HomeAssistant created a second sensors (binary_sensor.heating_active_2) that I had not noticed before.

[image]

closed 2020-10-13 by majdzik84

MichaelDvP, 2020-10-13:
Ok, change undone.