emsesp/EMS-ESP #554: Question: additional error messages?

giovanne123, 2020-10-12:
Does EMS-ESP also fetch telegrams with additional error messages, e.g. UBAErrorMessages1, RCErrorMessages: https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#uebersicht

I ask because currently my MixingModul MM10 does make some trouble (the relay is clicking, maybe elko end of life https://www.heizungsforum.de/threads/mischermodul-mm10-reparatur.9061/page-2 ;-)

**But the main question, the following error message is displayed on the BC10 display:
"A32" - "816"** (https://www.heizungfehlermeldung.de/fehler.php?merk=9&ketel=375&storing=10147&subcode=3573)
**But this is not reported within the "UBAMonitorFast" (ServiceCode,ServiceCodeNumber) telegram (I don't know why this isn't)
and I was wondering what I can monitor to fetch such error messages.**

Someone ideas how to catch this error? (with EMS-ESP to have it in HA for monitoring such faults?)**  

My system - Buderus GB132 ->
These 5 were recognized by EMS-ESP:
 Boiler: Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID: 0x08, ProductID: xx, Version: 02.07)
 Controller: BC10/RFM20 Receiver (DeviceID: 0x09, ProductID: xx, Version: 02.00)
 Thermostat: RC30 (DeviceID: 0x10, ProductID: xx, Version: 02.08)
 Switching Module: WM10 Switch Module (DeviceID: 0x11, ProductID: xx, Version: 02.00)
 Mixing Module: MM10 Mixing Module (DeviceID: 0x21, ProductID: xx, Version: 02.00)

I'm still on 1.9.5

MichaelDvP, 2020-10-13:
I've tested these messages, 10 and 11 are from boiler, 12 comes from thermostat and communication errors are only recognized by the thermostat. 
[image]

These error-logs are not fetched from ems-esp, but in v2.1 you can check them via mqtt: For the first dataset (12 bytes):
"system": `{"cmd":"send","data":"0B 90 12 00 0C"}`
"response": `{"src":"10","dest":"0B","type":"12","offset":"00","data":"32 32 03 30 94 0A 0E 0C 2A 00 01 19"}`
This first error is A22/816 on 2020-10-12 from device 19 (i forced this error to check)

> I'm still on 1.9.5

There i see no chance to check it.

MichaelDvP, 2020-10-13:
Since the message type 12 is a log i tried to find the actual message by generating the error. Found it in 0xA2:
No error: `Thermostat(0x10) -> (0x00), (0xA2), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 02`  
Error: `Thermostat(0x10) -> (0x00), (0xA2), data: 41 32 32 03 30 00 02 00 00 00 00 00 00 02` 
It's tree bytes ascii-char, then uint16 errorcode. If the error is cleared, the 0-message is send immediatly. 
Here is the result:
[image]

giovanne123, 2020-10-13:
Ok thanks, first I will have to update to 2.x ;-)
Then I will see...

bbqkees, 2020-10-13:
@giovanne123 Just follow the [guide](https://bbqkees-electronics.nl/wiki/gateway/firmware-update-to-v2.html) to update.

MichaelDvP, 2020-10-13:
Ok, i've pushed to the latest beta [here](https://github.com/proddy/EMS-ESP/tree/firmware/firmware)
@giovanne123: I tried to add the HA autoconfiguration for this too, Can you check (i don't have HA).

giovanne123, 2020-10-14:
Ok and thanks for the findings. I will report after I have updated to 2.x . Need some timeslot for updating my "productive" environment ;-)

giovanne123, 2020-10-15:
Tested in 2.1.0b7:
- Updated from 1.9.5 to 2.1.0b7 with automated migrated settings :-)
[image]

- ErrorCode "A32(816)" in MQTT topic:
[image]

- ErrorCode Reset "(0)" in MQTT topic:
[image]

- ErrorCode within HA (sensor manually set up because I'm currently not using AutoDiscover):
I was to slow to set up the sensor to get the A32(816) within HA, therefor currently the (0) (Reset) is shown:
[image]

Now I have to check all the details after firmware update (or mayb have a look to auto discover) ;-)   

Thanks again for the findings !!!!

closed 2020-10-20 by proddy