emsesp/EMS-ESP #556: Upgrade from v2.0.0b7 to v2.1.0b7

suysh, 2020-10-15:
Hello,

The ems-esp is working for a long time and today I updated again my interface to the latest build.
Badly, now the mqtt is not working :-(
I get status "disconnected" and the reason "Client ID rejected".
Don't know what this means, or how to fix.

Sadly I can't find the old build to go back to v2.0.0b7.bin

Kind regards, Hans

bbqkees, 2020-10-15:
What MQTT broker are you using and what if you delete the Client ID in the MQTT settings?

suysh, 2020-10-15:
I'm using mosquitto on a raspberry (openhab2.5).
The Client ID is empty. If I'm correct this setting was not present in my previous version.

bbqkees, 2020-10-15:
And what if you set it do 'ems-esp'?
That's now the default.

suysh, 2020-10-15:
I cleared the Client ID, then
Listening to ems-esp/status          I receive : online
Listening to ems-esp/heartbeat    I receive : offline
The others seems not receiving, even not when I do publish in telnet

MichaelDvP, 2020-10-15:
There are a lot of changes from 2.0 to 2.1, maybe the broker expects the old structure. Have you tried to restart the broker before connecting?
Also you can check `Clean session`.

suysh, 2020-10-15:
Forgot to mention that I overwrited my module via USB, before I used the update from the upload firmware section.
Now the MQTT status is connected.
Clean session is checked by default.
I also restarted my broker.
Can I check what is published on which topics? The 'show mqtt' queue is empty.

suysh, 2020-10-15:
I receive : `000+00:03:51.254 E 22: [mqtt] Failed to publish to ems-esp/boiler_data after 3 attempts`

Status is connected, but get publishing errors for topic ems-esp/boiler_data
Topic : ems-esp/solar_data is publishing
Topic : ems-esp/thermostat_data is publishing

suysh, 2020-10-15:
The topics `ems-esp/mixing_data1` and `ems-esp/mixing_data2` are also not publishing data (but not showing an error message in Telnet console like `ems-esp/boiler_data`.

suysh, 2020-10-15:
Sometimes the `ems-esp/boiler_data` is publishing correctly. Can't figure out why, or reproduce the behaviour. Sometimes after a restart, but after some time it stops working.
If you want I can share my screen live to show/test something with MS Teams.

suysh, 2020-10-16:
While trying to modify/update my code to the new mqtt json-keys I see in the documentation 2 versions of topics. I suppose the one at the bottom is the new structure?
I have mixing modules and compared to the structure before, I'm missing the pumpMod value?

MichaelDvP, 2020-10-16:
`boiler_data` is a very long message. Do you have a `message_size_limit` in your `mosquitto.conf`?

There was a long discussion and research for mixing-values. The old values were wrong and pumps are only switched in mixers, not modulated. It's renamed to pumpstatus.

suysh, 2020-10-16:
New names are much better :-)
I don't have a limit like `message_size_limit` set. Default is none.
I see in my mosquitto log:
```
1602825109: New connection from ::1 on port 1883.
1602825109: New client connected from ::1 as mosqsub/11918-openHABia (c1, k60).
1602825132: New connection from 10.0.0.23 on port 1883.
1602825132: Client ems-esp already connected, closing old connection.
1602825132: Client ems-esp disconnected.
1602825132: New client connected from 10.0.0.23 as ems-esp (c1, k60).
```

suysh, 2020-10-16:
[mosquitto.log](https://github.com/proddy/EMS-ESP/files/5390648/mosquitto.log)

MichaelDvP, 2020-10-16:
Is your boiler detected by ems-esp? What does the web dashboard show? 
Output of [http://10.0.0.23/api?device=boiler&cmd=info](http://10.0.0.23/api?device=boiler&cmd=info)
What have you set as boiler publish interval?

suysh, 2020-10-16:
My boiler is detected perfectly, dashboard show all actual values as they are in my boiler.
The link shows indeed the text that I want to receive from the topic `ems-esp/boiler_data`
The interval is default 10 seconds.
The menu MQTT show the publish errors counting up.

suysh, 2020-10-16:
Maybe something : "wWChargeType":"3-way valve" is printed with an enter/end-of-line between 3-way and valve.

proddy, 2020-10-16:
@suysh try with QoS 0. It could be my implementation of QoS 1.

suysh, 2020-10-16:
It was default QoS 0. I tried with 0...1...2 no difference.
Can you check why a special character (enter/return) appears in "3-way valve"?

proddy, 2020-10-16:
I don't see any special chars in the `wWChargeType` key on my setup. Also looking at the code I can't see how this can happen.

What is the error you're seeing in mosquitto ? just a TCP disconnect?

suysh, 2020-10-16:
[image]
I'm looking at the API link receipt from MichaelDvP.

proddy, 2020-10-16:
wow, never noticed that before. I used Chrome and the "JSON formatted text" plugin which nicely renders the JSON. But using firefox/edge I see the same thing:

[image]

now need to find out why.

suysh, 2020-10-16:
Haha, funny :-)
Don't know if this is the reason why my Mosquitto rejects some topics, and others are always working perfect. It worked before the update.
[image]
I'm not an mqtt expert. I have no clue where to search for the reason.

MichaelDvP, 2020-10-16:
@proddy It's simply the only white space where firefox can automatically set a new line. The same happens on a minus. Use a json-formater addon.

@suysh Have you tried to restart your raspberry?

suysh, 2020-10-16:
Oops. Indeed !
I rebooted the raspberry multiple times.

proddy, 2020-10-16:
what's helped before is running a local copy of mosquitto on my windows dev machine from https://mosquitto.org/download/ and running it with verbose logging using `"C:\Program Files\mosquitto\mosquitto" -v` from a dos prompt

suysh, 2020-10-16:
```
[17:12:37] openhabian@openHABianPi:~$ mosquitto_sub -t ems-esp/# -v
ems-esp/status online
ems-esp/start {"version":"1.9.6b0","IP":"10.0.0.23","boottime":"2020-06-16T21:04:51"}
ems-esp/shower_data offline
ems-esp/mixing_data offline
ems-esp/heartbeat offline
ems-esp/sm_data offline
ems-esp/thermostat_data {"time":"17:11:50 16/10/2020","errorcode":"(0)","wwmode":"high","wwtemp":70,"wwtemplow":30,"hc1"                                                                                                      :{"seltemp":14,"currtemp":20.9,"comforttemp":14,"ecotemp":10,"manualtemp":14,"targetflowtemp":0,"summertemp":25,"summerm                                                                                                      ode":"auto","mode":"manual","modetype":"comfort"},"hc2":{"seltemp":20,"currtemp":20.9,"comforttemp":24,"ecotemp":15,"man                                                                                                      ualtemp":20,"targetflowtemp":0,"summertemp":17,"summermode":"auto","mode":"manual","modetype":"comfort"}}
ems-esp/solar_data {"collectorTemp":29.4,"tankBottomTemp":39.9,"solarPumpModulation":0,"cylinderPumpModulation":0,"solar                                                                                                      Pump":"off","valveStatus":"off","tankHeated":"off","collectorShutdown":"off","energyLastHour":9.8,"energyToday":571,"ene                                                                                                      rgyTotal":333.8}
ems-esp/mixing_data {"hc1":{"flowSetTemp":0,"pumpStatus":"off","valveStatus":0},"hc2":{"flowTemp":26.4,"flowSetTemp":0,"                                                                                                      pumpStatus":"off","valveStatus":0}}
ems-esp/thermostat_data {"time":"17:11:50 16/10/2020","errorcode":"(0)","wwmode":"high","wwtemp":70,"wwtemplow":30,"hc1":{"seltemp":14,"currtemp":20.9,"comforttemp":14,"ecotem                                               p":10,"manualtemp":14,"targetflowtemp":0,"summertemp":25,"summermode":"auto","mode":"manual","modetype":"comfort"},"hc2":{"seltemp":20,"currtemp":20.9,"comforttemp":24,"ecotem                                               p":15,"manualtemp":20,"targetflowtemp":0,"summertemp":17,"summermode":"auto","mode":"manual","modetype":"comfort"}}
ems-esp/solar_data {"collectorTemp":29.4,"tankBottomTemp":39.9,"solarPumpModulation":0,"cylinderPumpModulation":0,"solarPump":"off","valveStatus":"off","tankHeated":"off","col                                               lectorShutdown":"off","energyLastHour":9.8,"energyToday":571,"energyTotal":333.8}
ems-esp/mixing_data {"hc1":{"flowSetTemp":0,"pumpStatus":"off","valveStatus":0},"hc2":{"flowTemp":26.4,"flowSetTemp":0,"pumpStatus":"off","valveStatus":0}}
ems-esp/thermostat_data {"time":"17:11:50 16/10/2020","errorcode":"(0)","wwmode":"high","wwtemp":70,"wwtemplow":30,"hc1":{"seltemp":14,"currtemp":20.9,"comforttemp":14,"ecotem                                               p":10,"manualtemp":14,"targetflowtemp":0,"summertemp":25,"summermode":"auto","mode":"manual","modetype":"comfort"},"hc2":{"seltemp":20,"currtemp":20.9,"comforttemp":24,"ecotem                                               p":15,"manualtemp":20,"targetflowtemp":0,"summertemp":17,"summermode":"auto","mode":"manual","modetype":"comfort"}}
ems-esp/solar_data {"collectorTemp":29.4,"tankBottomTemp":39.9,"solarPumpModulation":0,"cylinderPumpModulation":0,"solarPump":"off","valveStatus":"off","tankHeated":"off","col                                               lectorShutdown":"off","energyLastHour":9.8,"energyToday":571,"energyTotal":333.8}
ems-esp/mixing_data {"hc1":{"flowSetTemp":0,"pumpStatus":"off","valveStatus":0},"hc2":{"flowTemp":26.4,"flowSetTemp":0,"pumpStatus":"off","valveStatus":0}}
ems-esp/thermostat_data {"time":"17:11:50 16/10/2020","errorcode":"(0)","wwmode":"high","wwtemp":70,"wwtemplow":30,"hc1":{"seltemp":14,"currtemp":20.9,"comforttemp":14,"ecotem                                               p":10,"manualtemp":14,"targetflowtemp":0,"summertemp":25,"summermode":"auto","mode":"manual","modetype":"comfort"},"hc2":{"seltemp":20,"currtemp":20.9,"comforttemp":24,"ecotem                                               p":15,"manualtemp":20,"targetflowtemp":0,"summertemp":17,"summermode":"auto","mode":"manual","modetype":"comfort"}}
ems-esp/solar_data {"collectorTemp":29.2,"tankBottomTemp":39.9,"solarPumpModulation":0,"cylinderPumpModulation":0,"solarPump":"off","valveStatus":"off","tankHeated":"off","col                                               lectorShutdown":"off","energyLastHour":9.8,"energyToday":571,"energyTotal":333.8}
ems-esp/mixing_data {"hc1":{"flowSetTemp":0,"pumpStatus":"off","valveStatus":0},"hc2":{"flowTemp":26.4,"flowSetTemp":0,"pumpStatus":"off","valveStatus":0}}
ems-esp/heartbeat {"status":"connected","rssid":100,"uptime":"000+09:28:01.670","uptime_sec":34081,"mqttpublishfails":9,"txfails":0,"rxfails":0,"freemem":54,"fragmentation":22}
ems-esp/thermostat_data {"time":"17:11:50 16/10/2020","errorcode":"(0)","wwmode":"high","wwtemp":70,"wwtemplow":30,"hc1":{"seltemp":14,"currtemp":20.9,"comforttemp":14,"ecotemp":10,"manualtemp":14,"targetflowtemp":0,"summertemp":25,"summermode":"auto","mode":"manual","modetype":"comfort"},"hc2":{"seltemp":20,"currtemp":20.9,"comforttemp":24,"ecotemp":15,"manualtemp":20,"targetflowtemp":0,"summertemp":17,"summermode":"auto","mode":"manual","modetype":"comfort"}}
[... cut ...] temp":0,"summertemp":17,"summermode":"auto","mode":"manual","modetype":"comfort"}}
ems-esp/solar_data {"collectorTemp":29.1,"tankBottomTemp":39.9,"solarPumpModulation":0,"cylinderPumpModulation":0,"solarPump":"off","valveStatus":"off","tankHeated":"off","collectorShutdown":"off","energyLastHour":9.8,"energyToday":571,"energyTotal":333.8}
ems-esp/mixing_data {"hc1":{"flowSetTemp":0,"pumpStatus":"off","valveStatus":0},"hc2":{"flowTemp":26.4,"flowSetTemp":0,"pumpStatus":"off","valveStatus":0}}

```

suysh, 2020-10-16:
I'm only missing the ems-esp/boiler_data

proddy, 2020-10-16:
run mosquitto in verbose to see if it is rejecting the boiler_data topic. If its not coming through then check EMS-ESP from the console with debug logging and after a `publish` command.

MichaelDvP, 2020-10-16:
You can also try to use a different hostname, there are a lot of confusion with data from the old version: 
`ems-esp/start {"version":"1.9.6b0","IP":"10.0.0.23","boottime":"2020-06-16T21:04:51"}`

suysh, 2020-10-16:
I stopped mosquitto and started it again in verbose mode.
There is a Receipt for a subscribe from ems-esp for `ems-esp/boiler` on line 16
But no further publish commands for boiler
```
[17:35:16] openhabian@openHABianPi:~$ mosquitto -v
1602862521: mosquitto version 1.4.10 (build date Wed, 17 Oct 2018 19:03:03 +0200) starting
1602862521: Using default config.
1602862521: Opening ipv4 listen socket on port 1883.
1602862521: Opening ipv6 listen socket on port 1883.
1602862522: New connection from 10.0.0.23 on port 1883.
1602862522: New client connected from 10.0.0.23 as ems-esp (c0, k60).
1602862522: Sending CONNACK to ems-esp (0, 0)
1602862522: Received PUBLISH from ems-esp (d0, q0, r0, m0, 'ems-esp/info', ... (54 bytes))
1602862522: Received PUBLISH from ems-esp (d0, q0, r1, m0, 'ems-esp/status', ... (6 bytes))
1602862522: Received SUBSCRIBE from ems-esp
1602862522:     ems-esp/system (QoS 0)
1602862522: ems-esp 0 ems-esp/system
1602862522: Sending SUBACK to ems-esp
1602862522: Received SUBSCRIBE from ems-esp
1602862522:     ems-esp/boiler (QoS 0)
1602862522: ems-esp 0 ems-esp/boiler
1602862522: Sending SUBACK to ems-esp
1602862522: Received SUBSCRIBE from ems-esp
1602862522:     ems-esp/thermostat (QoS 0)
1602862523: ems-esp 0 ems-esp/thermostat
1602862523: Sending SUBACK to ems-esp
1602862523: New connection from 10.0.0.21 on port 1883.
1602862523: New client connected from 10.0.0.21 as S1_764 (c0, k60, u'openhab').
1602862523: Sending CONNACK to S1_764 (0, 0)
1602862523: New connection from 10.0.0.21 on port 1883.
1602862523: New client connected from 10.0.0.21 as P1_7311 (c0, k60, u'openhab').
1602862523: Sending CONNACK to P1_7311 (0, 0)
1602862523: Received PUBLISH from ems-esp (d0, q0, r0, m0, 'ems-esp/thermostat_data', ... (441 bytes))
1602862523: Received PUBLISH from ems-esp (d0, q0, r0, m0, 'ems-esp/solar_data', ... (237 bytes))
1602862523: Received PUBLISH from ems-esp (d0, q0, r0, m0, 'ems-esp/mixing_data', ... (135 bytes))
1602862532: Received PUBLISH from ems-esp (d0, q0, r0, m0, 'ems-esp/thermostat_data', ... (441 bytes))
1602862533: Received PUBLISH from ems-esp (d0, q0, r0, m0, 'ems-esp/solar_data', ... (237 bytes))
1602862533: Received PUBLISH from ems-esp (d0, q0, r0, m0, 'ems-esp/mixing_data', ... (135 bytes))
1602862534: Received PUBLISH from P1_7311 (d0, q0, r1, m0, 'Zehnder/ComfoAirQ450/120', ... (3 bytes))
1602862542: Received PUBLISH from ems-esp (d0, q0, r0, m0, 'ems-esp/thermostat_data', ... (441 bytes))
1602862543: Received PUBLISH from ems-esp (d0, q0, r0, m0, 'ems-esp/solar_data', ... (237 bytes))
1602862543: Received PUBLISH from ems-esp (d0, q0, r0, m0, 'ems-esp/mixing_data', ... (135 bytes))
1602862543: Received PUBLISH from ems-esp (d0, q0, r0, m0, 'ems-esp/heartbeat', ... (158 bytes))

```

proddy, 2020-10-16:
any errors in console when you do a `publish` command?

suysh, 2020-10-16:
Aha, we're are close. I copied the string from the API as message
```
[17:48:01] openhabian@openHABianPi:~$ mosquitto_pub -t ems-esp/boiler_data -m "{"heatingActive":"off","tapwaterActive":"off","wWComfort":"Eco","wWSelTemp":50,"wWSetTemp":53,"wWDisinfectionTemp":70,"selFlowTemp":5,"selBurnPow":0,"curBurnPow":0,"pumpMod":0,"wWType":"layered buffer","wWChargeType":"3-way valve","wWCircPump":"off","wWCircPumpMode":"2x3min","wWCirc":"off","outdoorTemp":12.1,"wWCurTemp":33.6,"wWCurTemp2":46,"wWCurFlow":0,"curFlowTemp":26.5,"retTemp":23.9,"sysPress":1.3,"wwStorageTemp1":33.6,"wwStorageTemp2":46,"wWActivated":"on","wWOneTime":"off","wWDisinfecting":"off","wWCharging":"off","wWRecharging":"off","wWTempOK":"on","wWActive":"off","burnGas":"off","flameCurr":0,"heatPump":"off","fanWork":"off","ignWork":"off","wWHeat":"on","heatingActivated":"on","heatingTemp":82,"pumpModMax":100,"pumpModMin":10,"pumpDelay":3,"burnMinPeriod":10,"burnMinPower":0,"burnMaxPower":100,"boilHystOn":-6,"boilHystOff":6,"wwMixTemperature":249,"wwBufferBoilerTemperature":403,"wWStarts":724,"wWWorkM":7373,"UBAuptime":761559,"burnStarts":2528,"burnWorkMin":61934,"heatWorkMin":54561,"serviceCode":"","serviceCodeNumber":203}"
Error: Unknown option 'buffer,wWChargeType:3-way'.

Use 'mosquitto_pub --help' to see usage.

```

suysh, 2020-10-16:
Can you replace the space by an underscore "_" ?

proddy, 2020-10-16:
yes, lets try that

proddy, 2020-10-18:
> Aha, we're are close. I copied the string from the API as message
>
> ```
> [17:48:01] openhabian@openHABianPi:~$ mosquitto_pub -t ems-esp/boiler_data -m [...]

you have to escape the special characters if you're using the command line. So \` and \"

I changed the 3-way to 3way in b8 but it didn't make a difference. I think the problem is somewhere else. From the mosquitto logs your topic boiler_data isn't being sent. Did you see errors in the EMS-ESP console?

MichaelDvP, 2020-10-18:
I thought it's not the dash. but the space that causes the problem. Try `3-way-valve` or `3way_valve`

suysh, 2020-10-18:
I tested the b8 version and now there are everywhere spaces, also in the solar_data telegram and this is still receiving well.
What is the difference?
So, the problem is somewhere else. Maybe the boiler_data is too long, but it's not displayed as rejected as should be in verbose mode.

proddy, 2020-10-18:
nothing has changed in the MQTT or solar logic. As I said earlier can you go to the Console, `log debug`, `su`, then `publish`. Watch out for messages like "000+01:44:06.042 D 47: [mqtt] Publishing topic ems-esp/boiler_data (#2010, attempt #1, pid 1)"

suysh, 2020-10-18:
When I reset the ems-esp, after the booting I receive 1x boiler_data.
Indeed publishing is failing:
```
000+00:05:10.086 D 33: [mqtt] Publishing topic ems-esp/boiler_data (#131, attempt #1, pid 0)
000+00:05:10.086 D 34: [mqtt] Failed to publish to ems-esp/boiler_data. Trying again, #1
000+00:05:10.187 D 35: [mqtt] Publishing topic ems-esp/boiler_data (#131, attempt #2, pid 0)
000+00:05:10.187 D 36: [mqtt] Failed to publish to ems-esp/boiler_data. Trying again, #2
000+00:05:10.288 D 37: [mqtt] Publishing topic ems-esp/boiler_data (#131, attempt #3, pid 0)
000+00:05:10.288 E 38: [mqtt] Failed to publish to ems-esp/boiler_data after 3 attempts
000+00:05:10.389 D 39: [mqtt] Publishing topic ems-esp/thermostat_data (#132, attempt #1, pid 1)
000+00:05:10.490 D 40: [mqtt] Publishing topic ems-esp/solar_data (#133, attempt #1, pid 1)
```
I have many boiler values, indeed maybe buffer problem. Then right after booting not all values are published and it fits into memory.
```
{
  "heatingActive": "off",
  "tapwaterActive": "off",
  "wWComfort": "Eco",
  "wWSelTemp": 50,
  "wWSetTemp": 53,
  "wWDisinfectionTemp": 70,
  "selFlowTemp": 5,
  "selBurnPow": 0,
  "curBurnPow": 0,
  "pumpMod": 0,
  "wWType": "layered buffer",
  "wWChargeType": "3-way valve",
  "wWCircPump": "off",
  "wWCircPumpMode": "2x3min",
  "wWCirc": "off",
  "outdoorTemp": 14.5,
  "wWCurTemp": 38.2,
  "wWCurTemp2": 50.2,
  "wWCurFlow": 0,
  "curFlowTemp": 42.2,
  "retTemp": 27,
  "sysPress": 1.4,
  "wwStorageTemp1": 38.2,
  "wwStorageTemp2": 50.2,
  "wWActivated": "on",
  "wWOneTime": "off",
  "wWDisinfecting": "off",
  "wWCharging": "off",
  "wWRecharging": "off",
  "wWTempOK": "on",
  "wWActive": "off",
  "burnGas": "off",
  "flameCurr": 0,
  "heatPump": "off",
  "fanWork": "off",
  "ignWork": "off",
  "wWHeat": "on",
  "heatingActivated": "on",
  "heatingTemp": 82,
  "pumpModMax": 100,
  "pumpModMin": 10,
  "pumpDelay": 3,
  "burnMinPeriod": 10,
  "burnMinPower": 0,
  "burnMaxPower": 100,
  "boilHystOn": -6,
  "boilHystOff": 6,
  "wwMixTemperature": 314,
  "wwBufferBoilerTemperature": 436,
  "wWStarts": 729,
  "wWWorkM": 7424,
  "UBAuptime": 764321,
  "burnStarts": 2543,
  "burnWorkMin": 62094,
  "heatWorkMin": 54670,
  "serviceCode": "",
  "serviceCodeNumber": 203
}
```

suysh, 2020-10-18:
In the MQTT settings, `Max Topic Length` does it make difference? My `boiler_data` is more than 1400 characters long.

proddy, 2020-10-18:
no, the topic length is ok (max 128). the payload could be an issue. I'll make a change in the code that displays it and do some further testing

proddy, 2020-10-18:
found the error I think. The max was 768. I'll push the fix along with some other changes later.

suysh, 2020-10-18:
Problem not solved :-(
```
000+00:02:27.747 D 222: [mqtt] Publishing topic ems-esp/boiler_data (#49, attempt #1, pid 0)
000+00:02:27.747 D 223: [mqtt] Failed to publish to ems-esp/boiler_data. Trying again, #1
000+00:02:27.848 D 224: [mqtt] Publishing topic ems-esp/boiler_data (#49, attempt #2, pid 0)
000+00:02:27.848 D 225: [mqtt] Failed to publish to ems-esp/boiler_data. Trying again, #2
000+00:02:27.949 D 226: [mqtt] Publishing topic ems-esp/boiler_data (#49, attempt #3, pid 0)
000+00:02:27.949 E 227: [mqtt] Failed to publish to ems-esp/boiler_data after 3 attempts
000+00:02:28.051 D 228: [mqtt] Publishing topic ems-esp/thermostat_data (#50, attempt #1, pid 1)
000+00:02:28.152 D 229: [mqtt] Publishing topic ems-esp/solar_data (#51, attempt #1, pid 1)
```

proddy, 2020-10-18:
right, I need to try and simulate it here. Can you send me the output of http://ems-esp/api?device=boiler&cmd=info

suysh, 2020-10-19:
OK, Thanks for checking this.
You can try my outputs from the previous posts as well, which are also taken from this link.
```
{
  "heatingActive": "off",
  "tapwaterActive": "off",
  "wWComfort": "Eco",
  "wWSelTemp": 50,
  "wWSetTemp": 53,
  "wWDisinfectionTemp": 70,
  "selFlowTemp": 5,
  "selBurnPow": 0,
  "curBurnPow": 0,
  "pumpMod": 0,
  "wWType": "layered buffer",
  "wWChargeType": "3-way valve",
  "wWCircPump": "off",
  "wWCircPumpMode": "2x3min",
  "wWCirc": "off",
  "outdoorTemp": 9.4,
  "wWCurTemp": 34.3,
  "wWCurTemp2": 47.3,
  "wWCurFlow": 0,
  "curFlowTemp": 29.3,
  "retTemp": 24.2,
  "sysPress": 1.3,
  "wwStorageTemp1": 34.3,
  "wwStorageTemp2": 47.3,
  "wWActivated": "on",
  "wWOneTime": "off",
  "wWDisinfecting": "off",
  "wWCharging": "off",
  "wWRecharging": "off",
  "wWTempOK": "on",
  "wWActive": "off",
  "burnGas": "off",
  "flameCurr": 0,
  "heatPump": "off",
  "fanWork": "off",
  "ignWork": "off",
  "wWHeat": "on",
  "heatingActivated": "on",
  "heatingTemp": 82,
  "pumpModMax": 100,
  "pumpModMin": 10,
  "pumpDelay": 3,
  "burnMinPeriod": 10,
  "burnMinPower": 0,
  "burnMaxPower": 100,
  "boilHystOn": -6,
  "boilHystOff": 6,
  "wwMixTemperature": 251,
  "wwBufferBoilerTemperature": 414,
  "wWStarts": 731,
  "wWWorkM": 7443,
  "UBAuptime": 765243,
  "burnStarts": 2545,
  "burnWorkMin": 62113,
  "heatWorkMin": 54670,
  "serviceCode": "",
  "serviceCodeNumber": 203
}
```

proddy, 2020-10-19:
I'm able to reproduce it and found the error, but not the cause. there's a limit to 1072 bytes buffer in the asyncTCP, only on the ESP8266.

suysh, 2020-10-19:
Hmm, that's bad.
Or maybe split the long boiler_data topic into sub topic's ?

proddy, 2020-10-19:
that could be an option. I'm playing around now with using a higher bandwidth lwip. If that fails I'll looking into chunking the TCP packets during send

suysh, 2020-10-19:
There are a lot of spaces in the string. If you remove all of them we are just below the limit.

suysh, 2020-10-19:
When removing all spaces, enter/returns I have 1050 bytes left.

proddy, 2020-10-19:
that's just the way its rendered on the screen. There are no spaces when in the real MQTT payload. You can see this using MQTTExplorer for example.

So lwip2 (-DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH) gives me a max TCP buffer of 2090 bytes, which fixes this problem but breaks the web server. I think I need to chunk it...

suysh, 2020-10-19:
OK. If I can help or test something... let me know.

proddy, 2020-10-19:
I split the payload into 2 now. called boiler_data_main and boiler_data_ww. in b9

suysh, 2020-10-20:
After adapting my code it's working again. Thanks !

proddy, 2020-10-20:
that's good. I spent quite some time last night writing some chunking/buffering code for the ESP8266 in the AsyncMqtt libraries but it was just too much effort to get it working correctly and then I was thinking that having an MQTT payload with more than 2000 characters in it is just plain silly. It's so much better to split the data into separate payloads. Having `boiler_data` replaced with `boiler_data_main` and `boiler_data_ww` will most definitely annoy a lot of people.

commit 3bb236ee (2020-10-18): increase json buffer for mqtt (fixes #556)
closed 2020-10-20 by proddy