emsesp/EMS-ESP #559: (v2) Build error for esp8266 - Build Env, Platform IO

giovanne123, 2020-10-16:
Need some help, 
tried to compile 2.0.1 or 2.1.0b7 with Platform IO 
but for esp8266 I get following error (esp32 is compiling fine):

```
Compiling .pio\build\esp8266\libc57\framework\NTPStatus.cpp.o
lib\framework\NTPSettingsService.cpp: In member function 'void NTPSettingsService::configureNTP()':
lib\framework\NTPSettingsService.cpp:68:34: error: 'setTZ' was not declared in this scope
     setTZ(_state.tzFormat.c_str());
                                  ^
Compiling .pio\build\esp8266\libc57\framework\OTASettingsService.cpp.o
*** [.pio\build\esp8266\libc57\framework\NTPSettingsService.cpp.o] Error 1
```

Already seen same error in https://github.com/proddy/EMS-ESP/issues/421#issuecomment-655358943, there it seems to be soved by PIO update.

But I have a fresh Visual Studio Code (1.50.1) with Platform IO IDE Addon (2.1.0) and therefore it should be up to date.

Where can I search, what library is causing this error?

proddy, 2020-10-16:
which branch, main or dev?

proddy, 2020-10-16:
I tested on a clean Win10 and Ubuntu and can't reproduce. Also using VS Code 1.50.1 and Pio 5.0.1. Try `pio update` and `pio upgrade`. if that fails delete the whole `.pio` folder and restart VSCode.

giovanne123, 2020-10-16:
Thanks, `pio update` does it :-)
I though after installing VSC and PIO IDE Addon, that PIO is up to date but wasn't :-(

giovanne123, 2020-10-16:
solved

closed 2020-10-16 by giovanne123