emsesp/EMS-ESP #560: Support for Nefit Bosch Compress Hybrid 7000i

formadvi, 2020-10-16:
Hello,

I have a Nefit Bosch Compress Hybrid 7000i in combination with Enviline mono block outside unit.
My boiler and thermostat are shown on the dashboard but not the heatpump unit and outside unit.

Are these supported? If not what would be needed to support those?

Regards

[Doc1.pdf](https://github.com/proddy/EMS-ESP/files/5391917/Doc1.pdf)

proddy, 2020-10-16:
Could be that we don't have them in our database yet. Can you try `scan devices`. See instructions at https://emsesp.github.io/docs/#/Troubleshooting?id=not-all-ems-devices-are-recognized-or-capturing-data

formadvi, 2020-10-16:
Hello,

Had to figure out SU password but found it.
Scan devices did not create any output.

Set loglevel to all and retried.
Below output from console.

Please let me know if you need further info. 
Or need me to do some other actions.

regards,
Andre.

ems-esp:/$ scan devices
Command not found
ems-esp:/$ su
Password:
000+01:06:16.533 N 0: [shell] Admin session opened on console
ems-esp:/# scan devices
ems-esp:/# log on
Invalid log level
ems-esp:/# log all
Log level = all
000+01:06:50.549 D 1: [mqtt] Publishing topic ems-esp/boiler_data (#495, attempt                                                                                #1, pid 1)
ems-esp:/# scan devices
000+01:06:53.621 D 2: [telegram] Tx read request to device 0x08 for type ID 0x07
000+01:06:53.771 D 3: [telegram] Sending read Tx [#46], telegram: 0B 88 07 00 20                                                                                A8
000+01:06:53.842 D 4: [emsesp] Last Tx read successful
000+01:06:53.847 D 5: [emsesp] New EMS device detected with ID 0x08. Requesting                                                                                version information.
000+01:06:53.847 D 6: [telegram] Tx read request to device 0x08 for type ID 0x02
000+01:06:53.847 D 7: [emsesp] New EMS device detected with ID 0x09. Requesting                                                                                version information.
000+01:06:53.847 D 8: [telegram] Tx read request to device 0x09 for type ID 0x02
000+01:06:54.106 D 9: [telegram] Sending read Tx [#47], telegram: 0B 88 02 00 20                                                                                BC
000+01:06:54.153 D 10: [emsesp] Last Tx read successful
000+01:06:54.158 D 17: [mqtt] Registering MQTT cmd wwcirculation with topic boil                                                                               er
000+01:06:54.158 D 18: [mqtt] Registering MQTT cmd wwcircmode with topic boiler
000+01:06:54.158 D 19: [mqtt] Registering MQTT cmd flowtemp with topic boiler
000+01:06:54.158 D 20: [mqtt] Registering MQTT cmd wwtemp with topic boiler
000+01:06:54.158 D 21: [mqtt] Registering MQTT cmd heatingactivated with topic b                                                                               oiler
000+01:06:54.158 D 22: [mqtt] Registering MQTT cmd heatingtemp with topic boiler
000+01:06:54.158 D 23: [mqtt] Registering MQTT cmd burnmaxpower with topic boile                                                                               r
000+01:06:54.158 D 24: [mqtt] Registering MQTT cmd burnminpower with topic boile                                                                               r
000+01:06:54.158 D 25: [mqtt] Registering MQTT cmd boilhyston with topic boiler
000+01:06:54.158 D 26: [mqtt] Registering MQTT cmd boilhystoff with topic boiler
000+01:06:54.158 D 27: [mqtt] Registering MQTT cmd burnperiod with topic boiler
000+01:06:54.158 D 28: [mqtt] Registering MQTT cmd pumpdelay with topic boiler
000+01:06:54.158 D 29: [emsesp] Adding new device GBx72/Trendline/Cerapur/Greens                                                                               tar Si/27i (device ID 0x08, product ID 123, version 07.00)
000+01:06:54.158 D 30: [emsesp] Fetching values for device ID 0x08
000+01:06:54.158 D 31: [telegram] Tx read request to device 0x08 for type ID 0x1                                                                               4
000+01:06:54.158 D 32: [telegram] Tx read request to device 0x08 for type ID 0x1                                                                               6
000+01:06:54.158 D 33: [telegram] Tx read request to device 0x08 for type ID 0x1                                                                               9
000+01:06:54.158 D 34: [telegram] Tx read request to device 0x08 for type ID 0x3                                                                               3
000+01:06:54.158 D 35: [telegram] Tx read request to device 0x08 for type ID 0xE                                                                               6
000+01:06:54.158 D 36: [telegram] Tx read request to device 0x08 for type ID 0xE                                                                               A
000+01:06:54.371 D 37: [telegram] Sending read Tx [#48], telegram: 0B 89 02 00 2                                                                               0 B4
000+01:06:54.398 D 38: [emsesp] Last Tx read successful
000+01:06:54.402 D 39: [emsesp] Adding new device BC10 (device ID 0x09, product                                                                                ID 190, version 01.04)
000+01:06:54.402 D 40: [emsesp] Fetching values for device ID 0x09
000+01:06:54.505 D 41: [telegram] Sending read Tx [#49], telegram: 0B 88 14 00

proddy, 2020-10-16:
you need to turn logging on first, and then the `scan devices`

formadvi, 2020-10-16:
Hello 

First time I did not have logging enabled. Therefore I ran command twice. In between I enabled logging Log Level = all
Two lines after log level = all the scan devices command is given.

Please check above.

Regards,
Andre.

formadvi, 2020-10-16:
Did another run for clarity:

ems-esp:/# log all
Log level = all
ems-esp:/# scan devices
000+03:17:50.384 D 2: [telegram] Tx read request to device 0x08 for type ID 0x07
000+03:17:50.560 D 3: [telegram] Sending read Tx [#76], telegram: 0B 88 07 00 20                                                                                                  A8
000+03:17:50.627 D 4: [emsesp] Last Tx read successful
000+03:17:50.632 D 5: [emsesp] New EMS device detected with ID 0x08. Requesting                                                                                                  version information.
000+03:17:50.632 D 6: [telegram] Tx read request to device 0x08 for type ID 0x02
000+03:17:50.632 D 7: [emsesp] New EMS device detected with ID 0x09. Requesting                                                                                                  version information.
000+03:17:50.632 D 8: [telegram] Tx read request to device 0x09 for type ID 0x02
000+03:17:50.854 D 9: [emsesp] No telegram type handler found for ID 0x34 (src 0                                                                                                 x08)
000+03:17:51.027 D 10: [telegram] Sending read Tx [#77], telegram: 0B 88 02 00 2                                                                                                 0 BC
000+03:17:51.079 D 11: [emsesp] Last Tx read successful
000+03:17:51.084 D 18: [mqtt] Registering MQTT cmd wwcirculation with topic boil                                                                                                 er
000+03:17:51.084 D 19: [mqtt] Registering MQTT cmd wwcircmode with topic boiler
000+03:17:51.084 D 20: [mqtt] Registering MQTT cmd flowtemp with topic boiler
000+03:17:51.084 D 21: [mqtt] Registering MQTT cmd wwtemp with topic boiler
000+03:17:51.084 D 22: [mqtt] Registering MQTT cmd heatingactivated with topic b                                                                                                 oiler
000+03:17:51.084 D 23: [mqtt] Registering MQTT cmd heatingtemp with topic boiler
000+03:17:51.084 D 24: [mqtt] Registering MQTT cmd burnmaxpower with topic boile                                                                                                 r
000+03:17:51.084 D 25: [mqtt] Registering MQTT cmd burnminpower with topic boile                                                                                                 r
000+03:17:51.084 D 26: [mqtt] Registering MQTT cmd boilhyston with topic boiler
000+03:17:51.084 D 27: [mqtt] Registering MQTT cmd boilhystoff with topic boiler
000+03:17:51.084 D 28: [mqtt] Registering MQTT cmd burnperiod with topic boiler
000+03:17:51.084 D 29: [mqtt] Registering MQTT cmd pumpdelay with topic boiler
000+03:17:51.084 D 30: [emsesp] Adding new device GBx72/Trendline/Cerapur/Greens                                                                                                 tar Si/27i (device ID 0x08, product ID 123, version 07.00)
000+03:17:51.084 D 31: [emsesp] Fetching values for device ID 0x08
000+03:17:51.084 D 32: [telegram] Tx read request to device 0x08 for type ID 0x1                                                                                                 4
000+03:17:51.084 D 33: [telegram] Tx read request to device 0x08 for type ID 0x1                                                                                                 6
000+03:17:51.084 D 34: [telegram] Tx read request to device 0x08 for type ID 0x1                                                                                                 9
000+03:17:51.084 D 35: [telegram] Tx read request to device 0x08 for type ID 0x3                                                                                                 3
000+03:17:51.084 D 36: [telegram] Tx read request to device 0x08 for type ID 0xE                                                                                                 6
000+03:17:51.084 D 37: [telegram] Tx read request to device 0x08 for type ID 0xE                                                                                                 A
000+03:17:51.132 D 38: [emsesp] New EMS device detected with ID 0x09. Requesting                                                                                                  version information.
000+03:17:51.132 D 39: [telegram] Tx read request to device 0x09 for type ID 0x0                                                                                                 2
000+03:17:51.360 D 40: [emsesp] Received UBAMonitorFast
000+03:17:51.397 D 41: [mqtt] Publishing topic ems-esp/heating_active (#1428, at                                                                                                 tempt #1, pid 1)
000+03:17:51.498 D 42: [mqtt] Publishing topic ems-esp/tapwater_active (#1429, a                                                                                                 ttempt #1, pid 1)
000+03:17:51.561 D 43: [telegram] Sending read Tx [#78], telegram: 0B 89 02 00 2                                                                                                 0 B4
000+03:17:51.594 D 44: [emsesp] Last Tx read successful
000+03:17:51.599 D 45: [emsesp] Adding new device BC10 (device ID 0x09, product                                                                                                  ID 190, version 01.04)
000+03:17:51.599 D 46: [emsesp] Fetching values for device ID 0x09
[... cut ...]
000+03:18:21.163 D 175: [emsesp] Received UBAMonitorFast
000+03:18:21.381 D 176: [mqtt] Publishing topic ems-esp/boiler_data (#1436, attempt #1, pid 1)
000+03:18:21.437 D 177: [emsesp] Received MC10Status
000+03:18:30.860 D 178: [emsesp] Received MC10Status
000+03:18:31.058 D 179: [emsesp] Received UBAMonitorWW
000+03:18:31.315 D 180: [emsesp] Received UBAMonitorFast
000+03:18:31.393 D 181: [mqtt] Publishing topic ems-esp/boiler_data (#1437, attempt #1, pid 1)
ems-esp:/# log err
Log level = err

formadvi, 2020-10-16:
Added also run in debug level:
ems-esp:/# log debug
Log level = debug
ems-esp:/# scan devices
000+03:23:40.964 D 183: [telegram] Tx read request to device 0x08 for type ID 0x07
000+03:23:41.133 D 184: [telegram] Sending read Tx [#124], telegram: 0B 88 07 00 20 A8
000+03:23:41.695 D 185: [emsesp] No telegram type handler found for ID 0x2A (src 0x08)
000+03:23:41.998 D 186: [emsesp] No telegram type handler found for ID 0x34 (src 0x08)
000+03:23:42.165 D 187: [telegram] Sending read Tx [#125], telegram: 0B 88 07 00 20 A8
000+03:23:42.232 D 188: [emsesp] Last Tx read successful
000+03:23:42.236 D 189: [emsesp] New EMS device detected with ID 0x08. Requesting version information.
000+03:23:42.236 D 190: [telegram] Tx read request to device 0x08 for type ID 0x02
000+03:23:42.236 D 191: [emsesp] New EMS device detected with ID 0x09. Requesting version information.
000+03:23:42.236 D 192: [telegram] Tx read request to device 0x09 for type ID 0x02
000+03:23:42.365 D 193: [telegram] Sending read Tx [#126], telegram: 0B 88 02 00 20 BC
000+03:23:42.418 D 194: [emsesp] Last Tx read successful

MichaelDvP, 2020-10-17:
Are you sure you have connected the heatpump to the same ems-bus? This looks like there is only the boiler (and builtin controller) on the bus, no thermostat, no heatpump. To verify that, can you please post the output of `read 8 7` (no special logging required).

proddy, 2020-10-17:
does look that way doesn't it. @MichaelDvP I'm making some core changes today which should help with debugging

MichaelDvP, 2020-10-17:
Or it is only the heatpump identifies itself as product-id 123, ver 7.00, there are a lot of boilers using this product-id and it's not much difference controlling a burner or a compressor.

formadvi, 2020-10-17:
ems-esp:/# read 8 7
000+20:26:36.933 N 1: [emsesp] Boiler(0x08) -> (0x0B), (0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00
ems-esp:/#

I will check how everything is connected tomorrow if possible. I might have to contact the company that installed the equipment. I do see that the thermostat and outdoor sensor cables are going to heatpump now. Originally that was to heater. I looked at the documentation and outdoor unit is connected to canbus connection (Enviline mono block 5KW).. There are 2 EMS bus connectors in heatpump (occording documentation). One EMS (HDI) and normal EMS bus. I expect that from there it connects also to the normal heater (Trendline II). I will check tomorrow and open the heatpump unit.

Note: Currently the EMS board is connected to service connector Trendline II.

Thanks for support so far. I will try to support you as much as possible. If you need further info regarding equipment or want to debug let me know.

MichaelDvP, 2020-10-18:
Indeed according to 0x07-telegram there is nothing on the ems-bus but ems-esp, boiler and builtin controller. 
I think the heatpump is also master (boiler) and there is only start/stop line on clamps EM0 to the old boiler.  
Try to connect ems-esp to the heatpump ems-line and check if this sends on device-id 0x08.

formadvi, 2020-10-18:
Spot on. Old boiler connected On/Off only. No EMS bus. 
[image]

New Heatpump is now recognized but no data available.
[image]

Parameter 7 for all 3 devices:
For device 8 I see lot of errors reported.

ems-esp:/# read 38 7
000+00:09:18.232 N 12: [emsesp] Heat Pump(0x38) -> (0x0B), (0x07), data: <empty>
ems-esp:/#

ems-esp:/# read 8 7
000+00:09:38.346 E 18: [telegram] Rx: F0 88 00 FF 31 03 94 00 00 00 00 00 00 00 CA (CRC CA != 0C)
000+00:09:39.732 E 19: [telegram] Rx: 88 0B 07 00 09 01 00 00 00 00 01 00 00 00 00 FF 00 17 D1 5D 10 (CRC 10 != 75)
000+00:09:40.291 E 20: [telegram] Rx: C0 88 00 FF 00 03 9F 00 00 00 00 00 00 00 00 07 08 00 00 00 00 CC (CRC CC != 03)
000+00:09:41.180 E 21: [telegram] Rx: 40 88 00 FF 00 03 AC 00 1C 1F 2B (CRC 2B != 07)
000+00:09:42.020 E 22: [telegram] Rx: E4 88 00 FF 18 03 8F 80 00 00 3D 01 34 62 (CRC 62 != 4F)
000+00:09:45.941 E 23: [telegram] Rx: D0 88 00 E3 00 01 00 01 00 00 00 00 00 00 00 00 01 2F 1E 64 55 CF (CRC CF != 57)
000+00:09:46.730 E 24: [telegram] Rx: C0 FC 88 00 FF 00 03 A0 00 00 01 6D B7 (CRC B7 != 2E)
000+00:09:50.125 E 25: [telegram] Rx: 40 FC 88 00 FF 00 03 8D 01 00 10 08 01 00 00 01 00 00 00 19 04 00 00 01 00 1E 00 00 00 15 12 F8 (CRC F8 != 44)

ems-esp:/# read 10 7
000+00:10:11.288 N 33: [emsesp] Thermostat(0x10) -> (0x0B), (0x07), data: <empty>
ems-esp:/#

formadvi, 2020-10-18:
More detail: (indoor unit) Boiler is Bosch Compress Hybrid 7000i AW
Heatpump is Enviline A/W Monoblock 5KW

proddy, 2020-10-18:
We haven't implemented fully the heat pumps yet as there is little knowledge and data to go with. That's is why it says "no data available". If you could help us by
* showing what information you would expect to see from the heatpump. Either displayed on the unit itself or via a web page/mobile app if you're using the one provided by the manufacturer.
* providing some logs, from the Console type `watch on 38` and grab about 5 minutes of output.

There might also be some info in the telegram 0xBC (see https://github.com/proddy/EMS-ESP/issues/309). The UBA Master (boiler) sometimes collects the heatpump data.

formadvi, 2020-10-18:
I will do that. Also I will start reading on the EMS protocol and decoding and try to find values back.. 

Just to check, I see lot of CRC messages. Is that because something is not recognized or is there something wrong with communication itself?

000+02:35:09.694 E 359: [telegram] Rx: 80 FA 88 00 E3 00 01 00 01 00 00 00 00 00 00 00 00 01 2E 34 64 55 6F (CRC 6F != 10)
000+02:35:11.196 E 360: [telegram] Rx: C0 88 00 FF 00 03 A0 00 00 01 6D B7 (CRC B7 != 5F)
000+02:35:12.910 E 361: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
000+02:35:13.372 E 362: [telegram] Rx: 88 0B EA 05 01 30 FE 00 19 00 03 41 00 01 12 41 3C 2A 0A 38 64 64 00 56 F8 FF 09 (CRC 09 != D5)
000+02:35:19.694 E 363: [telegram] Rx: C0 88 00 E3 00 01 00 01 00 00 00 00 00 00 00 00 01 2F 34 64 55 67 (CRC 67 != A8)
000+02:35:29.594 E 364: [telegram] Rx: E0 88 00 E3 00 01 00 01 00 00 00 00 00 00 00 00 01 2F 34 64 55 67 (CRC 67 != 06)
000+02:35:30.471 E 365: [telegram] Rx: E0 88 00 FF 18 03 8F 80 00 00 2B 01 36 38 (CRC 38 != 03)
000+02:35:39.618 E 366: [telegram] Rx: F2 88 00 E3 00 01 00 01 00 00 00 00 00 00 00 00 01 2F 34 64 55 67 (CRC 67 != B2)
000+02:35:40.546 E 367: [telegram] Rx: E0 88 00 FF 18 03 8F 80 00 00 2B 01 39 37 (CRC 37 != 0C)
000+02:35:49.592 E 368: [telegram] Rx: C0 88 00 E3 00 01 00 01 00 00 00 00 00 00 00 00 01 2F 34 64 55 67 (CRC 67 != A8)
000+02:35:50.590 E 369: [telegram] Rx: 40 88 00 FF 00 03 8D 01 00 10 08 01 00 00 01 00 00 00 29 04 00 00 00 00 34 00 00 00 13 2E 01 (CRC 01 != D7)
000+02:35:51.195 E 370: [telegram] Rx: 80 FA 88 00 FF 00 03 A0 00 00 01 6D B7 (CRC B7 != DF)
000+02:36:00.065 E 371: [telegram] Rx: 80 88 00 D1 00 00 72 3C (CRC 3C != 37)
000+02:36:00.950 E 372: [telegram] Rx: 40 88 00 E9 00 00 80 00 80 00 00 00 00 00 41 3C 00 00 00 00 00 0

bbqkees, 2020-10-19:
@formadvi Not sure about the Bosch Compress but the related Enviline II unit has the servicejack inside the housing as a cable:
[image]

formadvi, 2020-10-19:
Yesterday when I connected on EMS bus I did see that cable. It did not state its functionality. Tested it and it works like a charm.
Thanks great tip.

Picture for further reference, might help future users.

[image]

formadvi, 2020-10-19:
Logging on device 38 for 5 minutes:

ems-esp:/$ su
Password:
000+01:01:02.514 N 0: [shell] Admin session opened on console
ems-esp:/# watch on 38
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x38
000+01:01:48.029 N 1: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x42B), data: 00 C9 07 D7 01
000+01:01:48.066 N 2: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0A 2E 00 61
000+01:02:01.936 N 3: [emsesp] Heat Pump(0x38) -> (0x0B), (0x47B), data: 0A 2E 00 61
000+01:02:02.225 N 4: [emsesp] Heat Pump(0x38) -> (0x0B), (0x42B), data: 00 C9 07 D7 01
000+01:02:48.044 N 5: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x42B), data: 00 C9 07 D7 01
000+01:02:48.082 N 6: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0A 2E 00 61
000+01:03:01.975 N 7: [emsesp] Heat Pump(0x38) -> (0x0B), (0x47B), data: 0A 2E 00 61
000+01:03:02.217 N 8: [emsesp] Heat Pump(0x38) -> (0x0B), (0x42B), data: 00 C9 07 D7 01
000+01:03:48.467 N 9: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x42B), data: 00 C9 07 D7 01
000+01:03:48.505 N 10: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0A 2E 00 61
000+01:04:02.022 N 11: [emsesp] Heat Pump(0x38) -> (0x0B), (0x47B), data: 0A 2E 00 61
000+01:04:02.312 N 12: [emsesp] Heat Pump(0x38) -> (0x0B), (0x42B), data: 00 C9 07 D7 01
000+01:04:48.091 N 13: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x42B), data: 00 C9 07 D7 01
000+01:04:48.129 N 14: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0A 2E 00 61
000+01:05:01.981 N 15: [emsesp] Heat Pump(0x38) -> (0x0B), (0x47B), data: 0A 2E 00 61
000+01:05:02.288 N 16: [emsesp] Heat Pump(0x38) -> (0x0B), (0x42B), data: 00 C9 07 D7 01
000+01:05:48.026 N 17: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x42B), data: 00 C9 07 D7 01
000+01:05:48.064 N 18: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0A 2E 00 61

FWES70, 2020-10-20:
Hi, I also have a Nefit Enviline heat pump Tower model, with monoblock aouter unit.
Can I help as well? here is what the EMS-ESP sees.
[image]
how do I get the logs?

proddy, 2020-10-20:
@FWES70 look at the documentation. https://emsesp.github.io/docs/#/

FWES70, 2020-10-20:
OK, I watch on 38 for a few minutes, and changed the set temp on the thermostat from 19,0 to 18,0 and later back to 19,0 
ems-esp:/# watch on 38
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x38
003+22:54:01.159 N 1100: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x42B), data: 00 B9 07 37 01
003+22:54:01.196 N 1101: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0C 38 00 73
003+22:54:08.082 N 1102: [emsesp] Heat Pump(0x38) -> (0x0B), (0x47B), data: 0C 38 00 73
003+22:54:08.396 N 1103: [emsesp] Heat Pump(0x38) -> (0x0B), (0x42B), data: 00 B9 07 37 01
003+22:55:01.267 N 1104: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x42B), data: 00 B9 07 3A 01
003+22:55:01.306 N 1105: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0C 38 00 73
003+22:55:07.671 N 1106: [emsesp] Heat Pump(0x38) -> (0x0B), (0x47B), data: 0C 38 00 73
003+22:55:08.208 N 1107: [emsesp] Heat Pump(0x38) -> (0x0B), (0x42B), data: 00 B9 07 3B 01
003+22:56:01.160 N 1108: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x42B), data: 00 B9 07 3B 01
003+22:56:01.198 N 1109: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0C 38 00 73
003+22:56:07.764 N 1110: [emsesp] Heat Pump(0x38) -> (0x0B), (0x47B), data: 0C 38 00 73
003+22:56:08.071 N 1111: [emsesp] Heat Pump(0x38) -> (0x0B), (0x42B), data: 00 B9 07 3B 01
003+22:57:01.277 N 1112: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x42B), data: 00 B9 07 3C 01
003+22:57:01.315 N 1113: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0C 38 00 73
003+22:57:07.617 N 1114: [emsesp] Heat Pump(0x38) -> (0x0B), (0x47B), data: 0C 38 00 73
003+22:57:07.955 N 1115: [emsesp] Heat Pump(0x38) -> (0x0B), (0x42B), data: 00 B9 07 3C 01
003+22:58:01.018 N 1116: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x42B), data: 00 B9 07 3C 01
003+22:58:01.056 N 1117: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0C 38 00 73
003+22:58:07.438 N 1118: [emsesp] Heat Pump(0x38) -> (0x0B), (0x47B), data: 0C 3A 00 77
003+22:58:07.769 N 1119: [emsesp] Heat Pump(0x38) -> (0x0B), (0x42B), data: 00 B9 07 3C 01
003+22:58:10.187 N 1120: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x2B9), data: 24 (offset 10)
003+22:58:51.333 N 1121: [emsesp] Heat Pump(0x38) -> (0x00), (0xBF), data: 38 C8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
003+22:59:01.271 N 1122: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x42B), data: 00 B9 07 3C 01
003+22:59:01.309 N 1123: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0C 39 00 77
003+22:59:07.587 N 1124: [emsesp] Heat Pump(0x38) -> (0x0B), (0x47B), data: 0C 39 00 77
003+22:59:07.885 N 1125: [emsesp] Heat Pump(0x38) -> (0x0B), (0x42B), data: 00 B9 07 3C 01
003+22:59:29.106 N 1126: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x2B9), data: 26 (offset 10)
003+23:00:00.500 N 1127: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x42B), data: 00 B9 07 3C 01
003+23:00:00.538 N 1128: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0C 39 00 77
003+23:00:07.432 N 1129: [emsesp] Heat Pump(0x38) -> (0x0B), (0x47B), data: 0C 39 00 77
003+23:00:07.762 N 1130: [emsesp] Heat Pump(0x38) -> (0x0B), (0x42B), data: 00 B9 07 3C 01

proddy, 2020-10-20:
@FWES70 what kind of data do you want to capture from the Heat Pump. Setting the thermostat data is on the thermostat so we're reading that anyway. For example in your log above the line
`Heat Pump(0x38) -> Thermostat(0x10), (0x2B9), data: 24 (offset 10)` -> is the heatpump telling the thermostat to set the temp to 18 degrees (18 * 2 in hex is 0x24).

proddy, 2020-10-20:
@formadvi the telegram types 0x47B and 0x42B are the ones we need to decipher. The 0x47B doesn't look like it changes so this has probably the configuration settings. And the data from 0x42B like `Heat Pump(0x38) -> (0x0B), (0x42B), data: 00 CA 07 E1 01` must have some interesting. Comparing with FWESA70 I would say 2nd byte (CA) and 4th byte (E1) have values. 0xE1 is 22.5 and 0xCA is 20.2. Does these match? You can do a `watch 42b` and see in real-time how the numbers change when the heatpump is adjusted.

FWES70, 2020-10-20:
@proddy yes, i noticed that too. the heat pump is providing a lot of data that I would like to at least read (like boiler temp (different from actual water temp, which is a calculated value); HP modulation %; position of the 3-way valave etc...) . I did a watch on the HP and put it in excel so it is possible to filter. see att. if you prefer the data in another format pse let me know.
I am still quite new to this, so please tell me what to do in order to crack the codes. I am looking on German sites because I thought I saw a table there once with the explanation of the telegrams.
[watch on 8 for 3 min.xlsx](https://github.com/proddy/EMS-ESP/files/5409370/watch.on.8.for.3.min.xlsx)

FWES70, 2020-10-20:
sorry for the typo in the name of the attachment: "watch on 8" should be "watch on 38"

proddy, 2020-10-20:
@FWES70 the XLS format is really useful, thanks. Except you are actually watching the boiler (8) and not the heatpump (38). In the WebUI what do you see when you click on the Boiler. Are the values you're after in the boiler data perhaps? Also try and upgrade to the latest 2.1 dev build which shows even more device data. What I'm struggling with is what data is available in the Heat Pump versus the Boiler (Enviline is actually a boiler)

FWES70, 2020-10-20:
@proddy you're right, it was the boiler. however, I think the boiler is the "heart" of the system. I'll do a watch on 38, let you know/ this is what I see when I click "boiler"":
[image]
and on "Thermostat"
[image]
when I click on Heat Pump or Gateway Module nothing happens.
I will update to 2.1...
Frans

proddy, 2020-10-20:
There will be no data when you click on the Heat Pump or Gateway Module, because we don't capture any. That's the goal of this GitHub issue. I think you'll find all the data you need from your heatpump will be in the Boiler section.

FWES70, 2020-10-20:
I think so too. in the meantime: here is the data from the HP (38 this time :-)). not so much it seems...
[watch on 38 for 5 min.xlsx](https://github.com/proddy/EMS-ESP/files/5409875/watch.on.38.for.5.min.xlsx)

FWES70, 2020-10-20:
and finally: the data from the KM200 - much busier it seems (1 min, it seemed to repeat)
[watch on 48 for 1 min.xlsx](https://github.com/proddy/EMS-ESP/files/5410012/watch.on.48.for.1.min.xlsx)

formadvi, 2020-10-20:
On my side the 47B is changing. First 2 bytes change 0x0B33  or 0x0B32
2866 or 2867. A temperature? 28.67? Tried to go through monitoring values in maintenance menu but cannot find these values yet.

formadvi, 2020-10-20:
Boiler: Bosch Enviline/Compress 6000AW (DeviceID:0x08 ProductID:172, Version:01.                                    18)
ProdID 172
Version 1.18
This should actually be: Bosch Hybrid 7000i AW

FWES70, 2020-10-20:
@formadvi I installed mine June '19, so it might not be an Envilne II?

FWES70, 2020-10-20:
I'll try again to upgrade to V 2.1 tomorrow, so far no success. maybe then it will change.

proddy, 2020-10-20:
> On my side the 47B is changing. First 2 bytes change 0x0B33 or 0x0B32
> 2866 or 2867. A temperature? 28.67? Tried to go through monitoring values in maintenance menu but cannot find these values yet.

the first byte is flags. 0x0B is in binary b1011 so bits 4, 2 and 1 are set. The temps are usually stored in a single byte and multiplied by 2. So the 2nd byte which is 0x33 means 25.5 degrees.

formadvi, 2020-10-20:
> @formadvi I installed mine June '19, so it might not be an Envilne II?

Device 8 in my case is the indoor unit of the heatpump. The Bosch hybrid 7000i AW. 
I was referring to my system. 
In the software it is reported as Bosch Enviline/Compress 6000AW. But I am wondering if combination of ID and version is enough to uniquely identify it as the 70001AW unit.

formadvi, 2020-10-20:
> the first byte is flags. 0x0B is in binary b1011 so bits 4, 2 and 1 are set. The temps are usually stored in a single byte and multiplied by 2. So the 2nd byte which is 0x33 means 25.5 degrees.

Thanks. That makes sense. I tried to find in software decoding of the existing messages and want to check those against my unit to see which are actually available. Any idea where I can find this?

formadvi, 2020-10-21:
001+05:18:54.901 N 15: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0C 34 00 74

34 = relative air humidity as measured by the thermostat.
0C = Dewpoint

When air humidity changes it is confirmed that these values change
[image]

proddy, 2020-10-21:
nice work @formadvi . I'll go an add this to the code (relative air humidity & dew point). It would be interesting if you're seeing these same values in the Thermostat or the Boiler as a lot of data is passed through.

While you're at it, could you also do a `watch bf`. 0xBF is another telegram type which is broadcasted by the Heat Pump with some interesting values.

FWES70, 2020-10-23:
@proddy I updated the ems-esp to V2.1 (2_1_0b10). I see more info on the boiler and thermostat. however, when I try a watch Putty crashes after a few seconds. any idea what could be the problem? (sometimes the led on the emsesp flashes just before putty stops, sometimes putty just closes). log works fine it seems
thanks!

proddy, 2020-10-23:
looks like a bug - I made some changes with the watch in b10. I'll take a look. Thanks for reporting!

proddy, 2020-10-23:
@FWES70 which watch command were you using? `watch on bf` and it crashes? Can you see if it breaks with a `watch raw` ..

FWES70, 2020-10-23:
[image]
Hi proddy, tried watch raw, putty crashes as well. before, i tried watch on all, then watch on 38.  no difference.

proddy, 2020-10-23:
it looks like you're using an older build. Did you compile yourself or take from the GitHub firmware branch?

FWES70, 2020-10-23:
I took it from Github

formadvi, 2020-10-23:
I am monitoring the other values. BF is not send frequently and only first 2 bytes seem to be populated. Cannot find matching values yet at this point of time.

The Thermostat (device 10, CW400) is the controller embedded in heatpump indoor unit. I am monitoring traffic to that controller and there seem to be a lot of telegrams that send data to this device. I will continue to map this traffic. And map to actual values in CW400. This might take some time.

Telegrams: 0x2A5, 0x2B9, 0x2F5, 0x201, 0x240, 0x291,0x29B, 0x301, 0x2E0, 0x2EA, 0x61E, 0x267
Do you already have some info on thos telegrams as that might help reverse engineer these telegrams?

proddy, 2020-10-23:
I'm working on a new build to fix the watch issue. I'll let you know when its done.

Some of those telegrams we already support like 2A* and 2B*. The others in your list I don't know.

FWES70, 2020-10-23:
I took pictures of the different screens of the HP, and did a watch (shortly) on all. tried to convert hex to dec, but I do not recognize any value...

[Watch_all_actual data.xlsx](https://github.com/proddy/EMS-ESP/files/5429336/Watch_all_actual.data.xlsx)

FWES70, 2020-10-23:
I found the installation guide for the settings from Nefit, can this help figure out some of the telegrams?
[pdf_nefit_warmtepompen_-_hmc300_instellingen_def.pdf](https://github.com/proddy/EMS-ESP/files/5429383/pdf_nefit_warmtepompen_-_hmc300_instellingen_def.pdf)

proddy, 2020-10-24:
sadly not much info in that PDF I'm afraid.

I'm still trying to figure out why ESM-ESP is crashing when watching telegrams in putty. Can you confirm
 - it only crashes when do you a `watch` ?
 - it crashes when you do either a `watch on` and `watch raw` ?
 - it also crashes when you do a `watch on 38`

MichaelDvP, 2020-10-26:
@formadvi 
> 001+05:18:54.901 N 15: [emsesp] Heat Pump(0x38) -> Thermostat(0x10), (0x47B), data: 0C 34 00 74
34 = relative air humidity as measured by the thermostat.
0C = Dewpoint

To calculate the dewpoint the temperature is needed, 0074 looks like the outdoor-temperature 11.6°C, 
But maybe outdoortemperature  is already published by the boiler in type-id 0x19?

formadvi, 2020-10-26:
004+03:53:20.763 N 2: [emsesp] Heat Pump(0x38) -> (0x0B), (0x47B), data: 0A 2C 00 61
This is current: Dewpoint 10, humidity 44%
Outside temp is reported as 12.5C
Does not match with 61 (would be 9.7C)

I am monitoring multiple records over longer period of time to find more matching data. Hope to report more later this week.
Will check if I can find the outside temperature.

FWES70, 2020-10-27:
> sadly not much info in that PDF I'm afraid.
>
> I'm still trying to figure out why ESM-ESP is crashing when watching telegrams in putty. Can you confirm
>
>     * it only crashes when do you a `watch` [...]

@proddy putty crashes whenever I try a watch (wether raw, or on) for "log" it does not crash.
let me know if you need more info or if I can support better.

proddy, 2020-10-27:
there is one message that comes in that is causing the crash. most probably an out-of-bounds error where the buffer is overrun (too much data for the allocated storage). it's hard to find and reproduce without using your setup and data. Are you able to compile and build the firmware yourself? I'll need to add some extra debug statements just for you

proddy, 2020-11-18:
@FWES70 can you check if this is still happening with the latest 2.1.1 ?

proddy, 2020-12-01:
@formadvi are you still seeing discrepancies with the data heat pump data?

formadvi, 2020-12-01:
No not at this moment. Currently not connected because I need to work on other topics. Will pick up later.

closed 2020-12-17 by proddy