emsesp/EMS-ESP #561: Problem with new v2.0.1 .I can't send new temp , change mode.

pitulek, 2020-10-16:
Hi, i cannot change temp , mode values in thermostat.In console always is too many arguments.Mqtt does not working too thermostat_cmd_temp1.
[image]

*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/proddy/EMS-ESP/issues) (both open and closed)*
* *searched the [wiki help pages](https://bbqkees-electronics.nl/wiki/gateway/troubleshooting.html)*

*Completing this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from the telnet session with the logging set to Verbose mode.*

**Additional context**
*Add any other context about the problem here.*

proddy, 2020-10-16:
Take a look at https://emsesp.github.io/docs/#/Troubleshooting?id=some-commands-via-mqtt-are-not-working

Basically, use the Console with debug logging to see what is happening to trace where it is failing. Also recommend you upgrade to the latest 2.1 version which has many fixes in this area.

pitulek, 2020-10-16:
i did it su
call thermostat temp 15 
command not found
[image]

pitulek, 2020-10-16:
where i can download the other versions?

MichaelDvP, 2020-10-16:
@pitulek i think you are not in su mode. Read [https://github.com/proddy/EMS-ESP#using-the-console](https://github.com/proddy/EMS-ESP#using-the-console)
The 2.1.beta firmware is [here](https://github.com/proddy/EMS-ESP/tree/firmware/firmware)

pitulek, 2020-10-16:
Thank you for your help. It change the firmware and it is working from console with command "call thermostat nighttemp 15"  or daytemp.
What format should i usen for mqtt  :  ems-esp/thermostat {"cmd":nighttemp, "data": 15} it dose not working?

proddy, 2020-10-16:
see https://emsesp.github.io/docs/#/MQTT?id=receiving-topics and note the text "`cmd is one of the commands listed in the Commands and must be enclosed in quotes`"

pitulek, 2020-10-16:
ems-esp/thermostat    {"cmd":"nighttemp", "data":15, "id":"hc1"} 
not working, i do not know where is the mistake.

MichaelDvP, 2020-10-16:
`{"cmd":"nighttemp", "data":15, "hc":1}` or `{"cmd":"nighttemp", "data":15, "id":1}` or `{"cmd":"nighttemp", "data":15}`

pitulek, 2020-10-16:
thank you,{  "cmd":"nighttemp", "data":15, "id": "hc1" } working .  Thank you very much for your help!

closed 2020-10-16 by proddy