emsesp/EMS-ESP #562: inconsistency for data and temperature sensor error

florin-laurentiu, 2020-10-17:
Hello everyone,

return to area,after some time of using EMS-ESP interface, with 2 problems that i will expose after remaind my system:
Buderus GB 162 heating system 35 kw, with RC 35 controller, SM10 module connected on EMS bus, external sensor for temperature connected to RC 35 and 2 solar thermal panels SKN 4.0 with the necessary sensor, which is 20 kohm value .
1. first problem that appear is that graphical logger on the RC 35 is not continuously, is as is broken in many parts for the recorded values.  this problem is appear only with EMS-esp interface connected- as i disconnect the interface, all is working fine. 
2. most of time , is coming error A51 which is related to the temperature sensors of the solar thermal panels which are SKN 4.0 
i disconnect the sensor from SM 10 and tested with multimeter and everything is OK ; after reconnected the sensor , is appear same error: A51. 
real problem is that EMS esp interface is not reporting any issue for this sensor, is only heating GB 162 who is reporting this error which is not possible to be cancelled. Also in this case, if a remove from EMS bus the interface, everything is returning to good condition of heating system, with no error - no A51 error!
EMS-ESP interface report those for my heating system:

Type	Brand	Model	Device ID	Product ID	Version
Boiler	Buderus	Topline/GB162	0x08	115	03.06
Controller	---	BC10/RFM20	0x09	68	02.03
Solar Module	---	SM10	0x30	73	02.00
Thermostat	---	RC35	0x10	86	31.07

thank you if you can give me some help !
laurentiu

MichaelDvP, 2020-10-17:
A51 can have some causes, what is the errorcode?  The actual 2.1.0b7 should list it in thermostat. I suppose its A51/816, meaning a communication error to SM10. Is ems-esp buspowered? Where is it connected. I think best is on the right ems-out-clamps of the SM10. Or try powering from service-jack.

florin-laurentiu, 2020-10-17:
hi Michael,

interface is connected on SM10  on EMS bus clamp, directly; first time i
try to connect by service jack but this give me error for entire GB162:
call service!
so, after restart my system, i was connecting interface on SM 10 directly.
please, A51 error is appear only with interface connected;
without interface connected, all is fine.
just to remember, i also tell that the graphical records of temperature on
RC 35 is appear broken when interface is connected and is OK , means
continuously when is not connected.

laurentiu

MichaelDvP, 2020-10-17:
Also the broken records seems to be a communication issue. 
What ems-esp-hardware do you use, what software-version, have you checked the errorcode on boiler or emsesp, is it 816 or another number? Does ems-esp register communication errors? Can you do a `log debug` on terminal and post the output? We need a some more information about your system to help (or send me a working crystal ball).

florin-laurentiu, 2020-10-17:
hello Michael,

i m coming to you with the information that i have in my hand:
- EMS-ESP v2.0.1 software; interface is buyed in september 2020 from:
https://bbqkees-electronics.nl/ . it was buyed by one of my colleague do to
the fact that i can t pay with my credit card.
- error code is 816, but appear only with interface connected  !

""EMS Status
EMS Connection Status
Connected

Statistic # Telegrams
(Rx) Received telegrams 3,294
(Rx) Incomplete telegrams 179
(Tx) Successfully sent telegrams 760
(Tx) Send Errors 65""   this is coming from errors reported by interface;
usual Tx ERRORS are more, what we have just now is after an restart of
interface. TX mode is 2 ; already try for one day in 1 and for another day
in 3 , but not any difference.
if you guide me, i will follow your instruction to do a log debug.

laurentiu

florin-laurentiu, 2020-10-17:
Hello again Michael,

just a small question that i have in my mind:  do you thing that can be a
problem related to language setting for my RC35 ? actual, language setting
for my RC 35 is ROMANIAN, because i m Romanian citizen.  is any possibility
to need to be changed on ENGLISH?

laurentiu

MichaelDvP, 2020-10-17:
No, language is not the problem. It's clearly a bus communication issue. 
Maybe @bbqkees has more experience and can help. Check the power settings of ems-esp, try another ems-connection on the boiler. Also retry with service-jack and jack powered, the error message was due to short circuit of the bus. Make sure the connector on ems-esp is proper fit in before connecting to service-jack on the boiler. I've read with some cables the hole in ems-esp housing is a little to small, giving sometimes contact issues, check that.

florin-laurentiu, 2020-10-17:
OK, i will follow your advice; i will return with information
tomorrow morning.

bbqkees, 2020-10-17:
@florin-laurentiu some Gateways from september had a too thick cable in the box. Those did not properly fit the hole in the Gateway.
But your's is now attached to the EMS bus itself so that's likely not the issue.
Can you try if the problems go away if you connect a USB adapter to the Gateway?
For this its best to remove the yellow jumper.
If there are too many devices on the bus there is not enough power. This can also give a range of errors.

Almost all short circuit problems have to do with the cables.

florin-laurentiu, 2020-10-17:
Hello Kees,

i will follow your instruction, beginning with Sunday morning 18 of October.

laurentiu

florin-laurentiu, 2020-10-18:
Hello everyone,

for today , just fix the problem with jack cable:
- due to the construction of the case, the hole for jack is too small and
jack can not be connected properly so the result was the malfunction of my
heating system.
fix the problem with case and connect interface on jack service part: now
is working the interface powered in this way.
for the moment is not appear A51 error, but looking to the number of errors
on communication, is appear to be very big for me - can you advice?

laurentiu

bbqkees, 2020-10-18:
I always order the same article number of cable but somewhere in September they changed the cable inside the bag with that same article number.

The new cable (left) has a slightly thicker plug than the old done (right) so it won't fit properly in the recessed hole in the Gateway.
I only noticed after about 20 Gateways were shipped and customers started sending emails about shortcuts and strange bus issues.

[image]

All Gateways now have a larger hole in the case and of next week there will be a cable included with an extended bus so this won't happen again.

florin-laurentiu, 2020-10-24:
hello,

return in picture with new information:
- for the connection of gateway on service jack, everything is OK , no more error messages till today
- for the gateway, there is a strange behaviour: each time in each morning when i try to access gateway via internet/network is take a long time - more than 5 minutes till is responding; most of time i need to ping gateway - it have a fix IP- in order to be able to access via browser as Chrome or Microsoft Edge.
- there is false information about solar pump as total numbering of function hour:

Collector temperature (TS1)	60.6°C
Tank bottom temperature (TS2)	41.9°C
Solar pump modulation (PS1)	35%
Solar Pump (PS1) active	on
**Pump working time	162290 days 11 hours 46 minutes**
- time to time there are information on the interface/gateway that are apparently frozen, related to pump of heating running or not and percentage of pump modulation as also frozen information during night time related to solar pump working- solar pump can t work during night as there is no sun on the sky....

MichaelDvP, 2020-10-24:
The pump working time is a bug in solar.cpp, line 268 reads 4 bytes instead of 3, easy to fix. If you divide the value by 256 you get the right time (633 days).

The slow access, do you mean access via web-interface? Loading the interface hangs if memory is low. This can happen if a telnet connection is open, or the mqtt queue is full. Is the mqtt-broker down over night?

proddy, 2020-10-24:
@MichaelDvP do you want to fix solar.cpp?

MichaelDvP, 2020-10-24:
done

florin-laurentiu, 2020-10-24:
> 
> The slow access, do you mean access via web-interface? Loading the interface hangs if memory is low. This can happen if a telnet connection is open, or the mqtt queue is full. Is the mqtt-broker down over night?

hi,

no, there is not mqtt running - even day or night.

florin-laurentiu, 2020-10-24:
by mistake i was closed this thread.
Please, considered OPEN!

MichaelDvP, 2020-10-24:
> no, there is not mqtt running - even day or night.

And is mqtt also disabled on ems-esp?
What is the output of  
`http://ems-esp.local/api?device=system&cmd=report` (if not resolved ems-esp.local -> your ip)

florin-laurentiu, 2020-10-24:
> http://ems-esp.local/api?device=system&cmd=report

hi,

true, mqtt is disabled on ems-esp.
answer to that command in browser, taking long time and finally is the web page or interface; please, note that this command is practically restart the interface because checking numbers of telegrams sended to heating system is very small as numbers...
as you can have more question, i can give you access to my system via Team Viewer  .

laurentiu

florin-laurentiu, 2020-10-30:
hello all,

today, i just removed EMS-ESP interface, for next reasons:
-  first of all, begin to have error message: connection with BC 10 ERROR, call SERVICE.
- each time when need to connect to EMS-ESP interface, wia network/internet, take a long time - 5 to 10 minutes and each time is starting from beginning with connection on bus...
Can someone please, send/show the electric drawing to can check interface of EMS-ESP on bus of Buderus system ?
i will try to check at hardware level -  if i will find electric drawing of interface - if  is something wrong/defect.
thank you!

proddy, 2020-10-30:
sorry to hear that. There's a simple schematic on the documentation site but bbqkees has more details. What you could try is running EMS-ESP in 'listen only' mode by setting txmode to 0 (off) so its only listening to the EMS bus (Rx) and not sending anything. This will rule out if EMS-ESP is transmitting bogus telegrams which is confusing the EMS bus.

florin-laurentiu, 2020-10-30:
hi,

could you help me to have electric drawing of interface for checking the
existing hardware of interface?
i will try also to put in listen mode to see what is happened.

thank you,
laurentiu

bbqkees, 2020-10-30:
@florin-laurentiu can you send me an email via the contact form to provide me your order ID?
https://bbqkees-electronics.nl/contact/
I'll just send you a new board.

florin-laurentiu, 2020-10-30:
i will send you reference number and please remember that it was ordered by
one of my friend due to the paiement :

32 of 3,702
FW: Your order 5326
Inbox

Liviu Amarandei
Mon, 28 Sep, 09:56
to me

Cred ca pachetul vine azi

From: BBQKees Electronics [mailto:noreply@sendcloud.sc]
Sent: September 22, 2020 9:54
To: Liviu Amarandei
Subject: Your order 5326 is ready to be shipped

BBQKees Electronics

Your order 5326 is ready to be shipped

Track & Trace

Delivery address

47 Independentei, bl 40, sc. A

ap. 7

905600 Medgidia

Romania

Liviu Amarandei,

Your order 5326 is packed and will be shipped by DHL. You can track your
shipment by clicking the button above.

Kind regards,
BBQKees Electronics

# Order number

5326

Track & Trace

JVGL0593131600970118

Tracking email is powered by Sendcloud

What is this?Thank you.Done.

MichaelDvP, 2020-10-30:
@florin-laurentiu when connecting to service jack, have you switched (jumpered) the power-setting to service-jack too?

florin-laurentiu, 2020-10-30:
My name is laurentiu, so you can tell me as this.

when EMS-ESP is connected via service jack,jumper for power supply is
in top position, when:
- box of interface is open, with board top
- jack connector on left part as also connector/screw for bus line
 yellow jumper is in top.

florin-laurentiu, 2020-10-30:
attached is a picture with that switch.

laurentiu

florin-laurentiu, 2020-11-04:
hi,

for the latest firmware:
- for the RC 35 controller":
(hc 1) Nofrost temperature 254 °C  !!!!

proddy, 2020-11-05:
Hi @florin-laurentiu , could you create a new issue for the nofrost problem. This one we'll mark as resolved.

closed 2020-11-05 by proddy

florin-laurentiu, 2020-11-05:
ok