emsesp/EMS-ESP #563: What command for onetime water

pitulek, 2020-10-17:
Hi!

What command is for onetimewater heat , is that  wwmode?

Is there any command for reset the boiler, because my sometimes hang on and i have to reset it with the button reset in the basement.

*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/proddy/EMS-ESP/issues) (both open and closed)*
* *searched the [wiki help pages](https://bbqkees-electronics.nl/wiki/gateway/troubleshooting.html)*

*Completing this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from the telnet session with the logging set to Verbose mode.*

**Additional context**
*Add any other context about the problem here.*

proddy, 2020-10-17:
boiler reset is hard to implement and depends on the model. see https://github.com/proddy/EMS-ESP/issues/306

onetime command is `wwonetime`. I noticed the documentation at https://emsesp.github.io/docs/#/API?id=device-boiler is not up to date so need to work on that

proddy, 2020-10-17:
go into the Console, type `su` and the password (ems-esp-neo unless you changed it). Then type in `call` and you'll see all the possible commands per device. `wwonetime` should be in there.

[image]

pitulek, 2020-10-17:
Yes there is wwonetime. Thank you!

closed 2020-10-20 by proddy