emsesp/EMS-ESP #564: Support for RC10 Thermostat (USA Version)

michalsteyn, 2020-10-18:
**Bug description**
I have an RC10 Moduline 100 Thermostat. Opening the unit lists the firmware version as v1.01.

EMS-ESP does not seem to correctly detect my thermostat. 
I get the following entry in Telnet:
`[emsesp] Unrecognized EMS device with device ID 0x17 with product ID 90. Please report on GitHub.`

From looking at other issues I believe the ID 0x17 is for an RC 10. However, my Product ID seems different from others.

Is there any easy way for me to configure the system to force it to detect this as an RC10? Or does this require a code change?

**Device information**
```
Uptime:        000+00:58:41.372
Chip ID:       0x00ac2221
SDK version:   2.2.2-dev(38a443e)
Core version:  2_7_4
Full version:  SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
Boot version:  6
Boot mode:     1
CPU frequency: 160 MHz
Flash chip:    0x00164020 (4194304 bytes)
Reset reason:  Exception
Reset info:    Fatal exception:9 flag:2 (Exception) epc1:0x4010576c epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000003 depc:0x00000000
Free heap:                14472 bytes
Free mem:                 56 %
Maximum free block size:  14104 bytes
Heap fragmentation:       3 %
Free continuations stack: 1232 bytes
Sketch size:   913648 bytes (2228224 bytes free)
Free heap:                14240 bytes
Free mem:                 55  %

WiFi: Connected
SSID: SteynHome
BSSID: 1C:F2:9A:7E:53:C2
RSSI: -70 dBm (60 %)
MAC address: 84:0D:8E:AC:22:21
Hostname: ems-esp
IPv4 address: 192.168.1.132/255.255.255.0
IPv4 gateway: 192.168.1.1
IPv4 nameserver: 192.168.1.1

Syslog:
 Host = <unset>
 Log level = off
 Mark interval = 0s
```

Devices:
```
ems-esp:/$ show devices
These EMS devices are currently active:

Boiler: BK13/BK15/Smartline/GB1x2 (DeviceID:0x08, ProductID:64, Version:02.07)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE9

Controller: BC10 GB142 (DeviceID:0x09, ProductID:89, Version:01.01)
```

**Additional context**
Watching the Telegrams Produces this:
```
ems-esp:/$ [K000+00:35:36.661 N 28: [emsesp] Boiler(0x08) -> (0x00), (0x34), data: 3C 02 4F 02 4F 21 00 01 03 00 06 31 36 00 A6 DB
ems-esp:/$ [K000+00:35:36.895 N 29: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 36 01 FA 64 5B 09 01 25 60 80 00 02 4F 01 AB 00 2B 0C 2D 48 00 C8 00 02 00
ems-esp:/$ [K000+00:35:41.678 N 30: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 36 01 FD 64 5C 09 01 25 60 80 00 02 4F 01 AD 00 2B 0C 2D 48 00 C8 00 02 00
ems-esp:/$ [K000+00:35:46.651 N 31: [emsesp] Boiler(0x08) -> (0x00), (0x34), data: 3C 02 4F 02 4F 21 00 01 03 00 06 31 36 00 A6 DB
ems-esp:/$ [K000+00:35:46.860 N 32: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 36 01 FF 64 5C 09 01 25 60 80 00 02 4F 01 AE 00 2B 0C 2D 48 00 C8 00 02 00
ems-esp:/$ [K000+00:35:50.186 N 33: [emsesp] (0x17) -> (0x00), (0xAD), data: 27 (offset 2)
ems-esp:/$ [K000+00:35:50.437 N 34: [emsesp] (0x17) -> Boiler(0x08), (0x1A), data: 2D 64 64 00
ems-esp:/$ [K000+00:35:51.569 N 35: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 2D 02 01 64 5C 09 01 24 60 80 00 02 4F 01 AF 00 2F 0C 30 59 00 CC 00 02 00
ems-esp:/$ [K000+00:35:52.615 N 36: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 2D 02 01 64 00 01 01 24 60 80 00 02 4F 01 B1 00 01 0C 30 59 00 CC 00 00 00
ems-esp:/$ [K000+00:35:54.559 N 37: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 2D 01 FD 64 00 01 01 24 60 80 00 02 4F 01 B1 00 01 0C 30 59 00 CC 00 00 00
ems-esp:/$ [K000+00:35:55.307 N 38: [emsesp] (0x17) -> Controller(0x09), (0x29), data: 01
ems-esp:/$ [K000+00:35:55.325 N 39: [emsesp] Controller(0x09) -> (0x17), (0x29), data: 7B
ems-esp:/$ [K000+00:35:56.602 N 40: [emsesp] Boiler(0x08) -> (0x00), (0x19), data: 80 00 01 E8 80 00 00 00 00 5F 01 4C 13 1D A3 CA 00 00 00 17 72 94 00 A5 38
ems-esp:/$ [K000+00:35:56.816 N 41: [emsesp] Boiler(0x08) -> (0x00), (0x34), data: 3C 02 4F 02 4F 21 00 01 03 00 06 31 36 00 A6 DB
ems-esp:/$ [K000+00:35:57.087 N 42: [emsesp] Boiler(0x08) -> (0x00), (0x07), data: 0B 80 00 00 00 00 00 00 00 00 00 00 00
ems-esp:/$ [K000+00:35:57.320 N 43: [emsesp] (0x17) -> (0x0B), Version(0x02), data: 5A 01 01
000+00:35:57.320 N 44: [emsesp] Unrecognized EMS device with device ID 0x17 with product ID 90. Please report on GitHub.
ems-esp:/$ [K000+00:35:57.399 N 45: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 2D 01 EC 64 00 01 01 24 60 80 00 02 4F 01 B0 00 00 0C 30 59 00 CC 00 00 00
ems-esp:/$ [K000+00:35:57.623 N 46: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 2D 01 EB 64 00 01 01 24 60 80 00 02 4F 01 B0 00 00 0C 30 59 00 CC 00 00 00
ems-esp:/$ [K000+00:35:58.595 N 47: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 2D 01 E5 64 00 01 01 24 60 80 00 02 4F 01 AD 00 00 0C 30 59 00 CC 00 00 00
ems-esp:/$ [K000+00:36:00.242 N 48: [emsesp] Boiler(0x08) -> (0x0B), (0x19), data: 80 00 01 DB 80 00 00 00 00 5F 01 4C 13 1D A3 CA 00 00 00 17 72 94 00 A5 38
ems-esp:/$ [K000+00:36:00.476 N 49: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 FF 3C FB 00 28 FF 02 46 D8 00
ems-esp:/$ [K000+00:36:00.638 N 50: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 2D 01 DA 64 00 01 01 24 60 80 00 02 4F 01 A6 00 00 0C 30 59 00 CC 00 00 00
ems-esp:/$ [K000+00:36:00.941 N 51: [emsesp] Boiler(0x08) -> (0x0B), (0x14), data: 76 6C D6
ems-esp:/$ [K000+00:36:01.149 N 52: [emsesp] Boiler(0x08) -> (0x0B), (0x16), data: FF 36 64 00 06 FA 0A 01 09 5F 32 02
ems-esp:/$ [K000+00:36:01.584 N 53: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 2D 01 D4 64 00 01 01 24 60 80 00 02 4F 01 A3 00 00 0C 30 59 00 CC 00 00 00
ems-esp:/$ [K000+00:36:03.578 N 54: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 2D 01 C9 64 00 01 01 24 60 80 00 02 4F 01 9C 00 00 0C 30 59 00 CC 00 00 00
ems-esp:/$ [K000+00:36:04.218 N 55: [emsesp] (0x17) -> (0x00), (0xAE), data: 80 02 27 00 C6 00 00 64
ems-esp:/$ [K000+00:36:04.455 N 56: [emsesp] (0x17) -> Boiler(0x08), (0x35), data: 11 11
ems-esp:/$ [K000+00:36:05.571 N 57: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 2D 01 BE 64 00 01 01 24 60 80 00 02 4F 01 96 00 00 0C 30 59 00 CC 00 00 00
```

MichaelDvP, 2020-10-18:
Thanks for the complete description including watch, the thermostat uses 0xAD, 0xAE for settings and monitor and is compatible to newer RC20. 
I'll add it to database.

proddy, 2020-10-18:
Thanks for adding that Michael. This issue would have been a good candidate to test the changes I pushed yesterday. The RC10 would have been shown in the WebUI as an 'unknown' device. Followed by a `http://ems-esp/api?device=system&cmd=report` would have given us the details to add it.

MichaelDvP, 2020-10-18:
Ups, if you want to test this first, remove the database-entry i've made already.

michalsteyn, 2020-10-19:
I tested the latest dev firmware and it works... Thanks so much!

closed 2020-10-19 by proddy