emsesp/EMS-ESP #568: Command wwonetime

pitulek, 2020-10-20:
Hi !

i have a question.In version 1.9 there was commad for one wwonetime  "0" or "1"  mqtt, so now is  wwonetime <temp>?

'''one_time_water_on:
  sequence:
  - data:
      payload: '1'
      topic: ems-esp/boiler_cmd_wwonetime
    service: mqtt.publish  

one_time_water_off:
  sequence:
  - data:
      payload: '0'
      topic: ems-esp/boiler_cmd_wwonetime
    service: mqtt.publish''' 
   it worked fine , the same like thermostat button, i could on and off thet action.
 If the new commmand is wwonetime  <temp> , how i can off ?

*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/proddy/EMS-ESP/issues) (both open and closed)*
* *searched the [wiki help pages](https://bbqkees-electronics.nl/wiki/gateway/troubleshooting.html)*

*Completing this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from the telnet session with the logging set to Verbose mode.*

**Additional context**
*Add any other context about the problem here.*

proddy, 2020-10-20:
Yes in 2.1 (the dev build) there's a new format for commands. see https://emsesp.github.io/docs/#/MQTT?id=receiving-topics. These even an example for using wwonetime at https://emsesp.github.io/docs/#/Home-Assistant?id=example-activating-one-time-hot-water-charging-dhw-once

pitulek, 2020-10-20:
so the data is 0 or 1, thank you!

closed 2020-10-20 by pitulek

proddy, 2020-10-21:
@pitulek correct. data can be either `0`, `1`, `"1"`, `"0"`, `true`, `false`, `"true"`, `"false"`, `"on"` or `"off"`