emsesp/EMS-ESP #569: Difference in Free Memory Size and Fragmentation values between MQTT, Web and API

kwertie01, 2020-10-20:
First a compliment to Proddy and al the other contributors of EMS-ESP. I'm now using this project for quite some time now and it is getting better every time :-)
 
I notice a difference in the values of Free Memory Size and Fragmentation in the Heartbeat MQTT message the Website and the API report. The values between the website and the API (report) are more equal but the values in the heartbeat MQTT message are more different.

I made this screenshot to show the difference
[image]

here is the output of the API:
{
  "System": {
    "version": "2.1.0b9",
    "uptime": "000+05:02:03.428",
    **"freemem": 45,
    "fragmem": 6**
  }
Do I misinterpret these values or is this not ok?

proddy, 2020-10-20:
yes this is normal. the free mem and frag and momentary measurements. The ones from the web are taken when the URL is loaded and before the resources are free'd up. I did experiment keeping these checks separate but they are also quite CPU intensive. Perhaps check every 30 seconds or so. Leave this issue open and I'll try out a few things.

proddy, 2021-01-05:
I've removed this for 2.3 as it's misleading

closed 2021-01-05 by proddy