emsesp/EMS-ESP #570: Can we cause wearout of the eeproms built into the buderus electronics

hamster65, 2020-10-21:
Just thinking: What happens if we use home automation and ems-esp to issue lot's of commands that change values that are stored e.g. in the rc35 eeprom. Like switching between night and day mode, or setting the target temperature every minute as part of a regulator... would that be an issue and damage the buderus electronics (eeproms) over time?

bbqkees, 2020-10-22:
No most parameter changes will never end up in an EEPROM.
A boiler and thermostat are a control system with the inherent property of continuous state changes over its life time so it's designed to cope with that.

hamster65, 2020-10-22:
So I could change settings like target temperature, summer/winter mode every second and it would not wear out the eeprom? How does rc35 retain those settings during a reset or power failure? I guess there is a battery or supercam inside of the rc35 unit for the clock, but I am not sure if that's used for settings, too.

closed 2020-11-13 by proddy