emsesp/EMS-ESP #573: (v2) Question - HA example config for climate, mqtt?

giovanne123, 2020-10-23:
Question regarding the usage of the HA climate / thermostat within EMS-ESP v2.1/v2.1.0b10

[image]

How to send changes for mode and temp within climate/mqtt?
How does a config look like for, e.g. "mode_command_topic"?
Maybe I miss something because I haven't send commands before to the EMS-ESP.
I am not using HA discover!

climate.yaml:
```
- platform: mqtt
  name: Heating Circuit 1 (HC1)
  modes:
    - "auto"
    - "heat"
    - "off"

  mode_command_topic: "ems-esp/thermostat"
  temperature_command_topic: "ems-esp/thermostat"

  mode_state_topic: "ems-esp/thermostat_data1"
  current_temperature_topic: "ems-esp/sensor_10-7E42-7001-0xxx-xx"
  temperature_state_topic: "ems-esp/thermostat_data1"

  mode_state_template: "{% if value_json.mode in ['manual', 'day'] %} heat {% elif value_json.mode in ['night', 'off'] %} off {% else %} auto {% endif %}"

  current_temperature_template: "{{ value_json.temp }}"
  temperature_state_template: "{{ value_json.seltemp }}"

  temp_step: 0.5
```

With this **mode_command_topic: "ems-esp/thermostat"** I publish mqtt with e.g.:
_topic: ems-esp/thermostat
data/payload: heat_

Do I need to generate also automations in automations.yaml or how **to get not only "heat" but {"cmd":"mode", "data": "day", "id": 1 }** into the data/payload to send?

Does someone maybe have an example config for HA?

(As workaround I already tested this successful with and input.select and automation: payload {"cmd":"mode", "data": "{{states("input_select.thermostat_mode_hc1")}}", "id": 1 })

giovanne123, 2020-10-23:
Hm, ok seems because stock climate in HA has no mode_command_template I have to do the workaround with temporairy mqtt-topic and automation like described here: https://community.home-assistant.io/t/enhanced-version-of-mqtt-hvac-climate-platform-with-proper-history-chart/75304/31

I will give it a try...

proddy, 2020-10-23:
@giovanne123 I would recommend you use the HA MQTT format in ESP-ESP. It was designed to solve this problem. Of course you can stick the traditional yaml files and use the topics thermostat_hc1 with either the mode or the temperature value. See https://emsesp.github.io/docs/#/MQTT?id=sending-commands

giovanne123, 2020-10-23:
Ok, than maybe I'm doing it to complicated with the workaround above, but it is working ;-)
I will have a look to https://emsesp.github.io/docs/#/MQTT?id=sending-commands and _"With Home Assistant, Thermostat commands can also be sent to control individual heating circuits via sending a mode string or temperature number to a topic thermostat_hc<n>."_
But how looks the payload for the topic ems-esp/thermostat_hc1 ??

proddy, 2020-10-23:
in your yaml file the `mode_command_topic` and `mode_state_topic` can both be `ems-esp/thermostat_hc<n>`. EMS-ESP will work it whether its a numerical temperature value or a mode string.

giovanne123, 2020-10-24:
Thanks I will try it. 
I thought that I have to pass not only the value (mode string or numeric temperature) to the topic. 
That means that EMS-ESP automatic decides If there is a numeric value then it must be temperature otherweise if a string it must be a mode.

proddy, 2020-10-24:
> That means that EMS-ESP automatic decides If there is a numeric value then it must be temperature otherwise if a string it must be a mode.

correct. The limitation of Home Assistant's climate component doesn't except value templates so I had to implement a new topic and build the logic into EMS-ESP.

giovanne123, 2020-10-27:
hm, can't get it to work (2.1.0b12),
publishing:
[image]

does not popup in console with "log debug".

---------
Where my workaround, s. above, does popup in console:
topic: ems-esp/thermostat
data: {"cmd":"mode", "data": "auto", "id": 1 }

proddy, 2020-10-27:
works for me.....

[image]

is it the topic subscribed?

giovanne123, 2020-10-27:
no there is no topics ems-esp/thermostat_hc1 (or ems-esp/thermostat_hc2)
[image]
(also after restarting HA, EMS-ESP, or sending via MQTT Explorer)

Hm, where is my fault (other than still not using discovery)?

proddy, 2020-10-27:
ah, I only enable the topic thermostat_hc1 if the MQTT format is set to HA.

giovanne123, 2020-10-28:
ok, so I finally have to switch to HA MQTT format...

giovanne123, 2020-10-30:
now switched to HA MQTT format, still without HA discovery.

[image]

both telegram types working:
 ems-esp/thermostat
 ems-esp/thermostat_hc1
 ems-esp/thermostat_hc2

Thanks

closed 2020-10-30 by giovanne123

proddy, 2020-10-30:
great!