emsesp/EMS-ESP #574: Topic names

MichaelDvP, 2020-10-24:
Can we change the topic names to **one scheme**, to me it looks more and more chaotic?

For thermostat publishes we have `thermostat_data` for nested and in single it splits to `thermostat_data`, `thermostat_data1`, etc.
Also the mixer has suffixes `_data`, `_data1`. `_data2`, ..., `_data10`
But now the boiler splits in `_main`and `_ww`, a complete new scheme with underscore and names. 
Also the subscriptions are always `boiler`, `thermostat`, but for HA there are additional `thermostat_hc1`, ...

I suggest to to skip the `_main` suffix from boiler and make all split-messages to a suffix with underscore and name. 
`boiler_data`
`boiler_data_ww`
`thermostat_data`
`thermostat_data_hc1`
`thermostat_data_hc2`
`mixer_data`      (=> see below)
`mixer_data_hc1` 
`mixer_data_wwc1`

this also fits to the subscriptions with
`boiler`
`thermostat`
`thermostat_hc1`

 
Another thing is the name `mixing`, sometimes called `mixing device`. Please change this to `mixer`, since it's also not called `boiling device`  or `controlling device`. Also skip the `device` from `solar`. 

Also new is a renaming of `sensor` to `dallassensor` for the api, but mqtt stay on `sensor_data`, This should always use the same keyword.

giovanne123, 2020-10-24:
I agree and would be fine to not to often change the topics because from 2.1.0b7 to b10 I have already modified my HA configs 😁

bbqkees, 2020-10-24:
I think we should settle now on one convention and not change it anymore.
Every time it changes it breaks plugin compatibility somewhere.

proddy, 2020-10-24:
I agree, it's becoming a bit of a mess and in need of some standardizing. Let's go with what Michael suggested. mixing to mixer, sensor to dallassensor all the way including the web interface.

By the way, the dev branch is for the project's contributors for adding and testing new code and ideas. Things will change and we made it clear that these builds are experimental. If you don't want to make changes to the plugins wait for the official 2.1 release.

giovanne123, 2020-10-24:
Btw. @proddy, I know that dev ist experimental. Was no criticism. Was only a hint because I only noticed about the change by listing the mqtt and searching the git log and issues. Maybe helpful for other contributors. OK, I have to wait for the official 2.1 and its doc 😁
But than I think I am no contributor only a tester asking for issues/enhancements, after testing I have to flash back to 2.0.1  🙄

proddy, 2020-10-25:
i get your point. A lot things were modified, broke, fixed and re-modified in 2.1. It's quite a big change to what I had originally expected. I really want to get 2.1 out but we keep finding new issues and tweaking things. But we're close. When I change things I try and keep the CHANGELOG.md file up to date so you shouldn't need to go through all the issues and git logs.

MichaelDvP, 2020-10-25:
I'm very happy with the topics now, closing this as done.

closed 2020-10-25 by MichaelDvP