emsesp/EMS-ESP #575: Is there an easy way to reset to factory default?

bbqkees, 2020-10-24:
Is there a (simple) way to do a factory reset without using pio erase?
(I could not find it but maybe its there.)

proddy, 2020-10-24:
There's the console's format command. I think it's still there.

proddy, 2020-10-24:
And also from the web ui.

bbqkees, 2020-10-24:
Yes but I got someone who forgot the password.

proddy, 2020-10-24:
Doh. No easy way I'm afraid. Take a copy of the settings using the web api system info and start again.

MichaelDvP, 2020-10-24:
@bbqkees : The su-password is different from the admin-password in web. Does he forgot both passwords? 
Maybe a make version that clears the settings on espota-update, or is the espota-password also changed and forgotten or espota disabled?

Victor-Mo, 2020-10-24:
Maybe have an option that 30 seconds after power-up you can do a factory reset from telnet without being logged in ? Or maybe a special password that can trigger it (in the first 30s). If you want it more secure, it only works when not connected to the EMS bus ?

proddy, 2020-10-24:
version 1 had that feature. When it was booting and the LED was flashing for 5 seconds if you reset the ESP it would go into safe mode where you could reset to factory settings.

But as Michael pointed out if you forget the web admin password AND the console su password, then you better stick to gardening

closed 2020-10-25 by proddy