emsesp/EMS-ESP #579: not all sensors report to mqtt

jtonk, 2020-10-26:
**Bug description**
`Heating Active` status is not resported correctly to home assistant.

**Steps to reproduce**
comporate ems-esp http interface with home assistant sensor value

**Expected behavior**
home assistant value to follow the ems-esp value

**Screenshots**

from HA
[image]

from http interface
[image]

**Device information**
Boiler: Topline/GB162 (DeviceID:0x08, ProductID:115, Version:05.09)
Thermostat: RC20/Moduline 300 (DeviceID:0x17, ProductID:77, Version:03.06)

**Additional context**
n/a

proddy, 2020-10-26:
thanks for reporting. Which version of EMS-ESP are you on? Can you send the contents of http://ems-esp/api?device=boiler&cmd=info as that will list all the available values you have. Note HA will create entities for all values, even if you don't have them with your Boiler/Thermostat. You just need to remove them from Home Assistant manually.

proddy, 2020-10-26:
and for the Heating Active, also check with the Console if this is showing the correct value.

jtonk, 2020-10-27:
see below the contents of the url:
{
	"heatingActive": "on",
	"tapwaterActive": "off",
	"wWComfort": "Eco",
	"wWSelTemp": 60,
	"wWSetTemp": 62,
	"wWDisinfectionTemp": 70,
	"selFlowTemp": 60,
	"selBurnPow": 80,
	"curBurnPow": 22,
	"pumpMod": 30,
	"wWType": "buffer",
	"wWChargeType": "3-way valve",
	"wWCircPump": "on",
	"wWCircPumpMode": "2x3min",
	"wWCirc": "off",
	"wWCurTemp": 61.7,
	"wWCurTemp2": 61.7,
	"wWCurFlow": 0,
	"curFlowTemp": 60.6,
	"retTemp": 48.2,
	"switchTemp": 0,
	"sysPress": 2.3,
	"boilTemp": 56.3,
	"wwStorageTemp1": 46.8,
	"wwStorageTemp2": 61.7,
	"wWActivated": "on",
	"wWOneTime": "off",
	"wWDisinfecting": "off",
	"wWCharging": "off",
	"wWRecharging": "off",
	"wWTempOK": "on",
	"wWActive": "off",
	"burnGas": "on",
	"flameCurr": 16.3,
	"heatPump": "on",
	"fanWork": "on",
	"ignWork": "off",
	"wWHeat": "off",
	"heatingActivated": "on",
	"heatingTemp": 60,
	"pumpModMax": 100,
	"pumpModMin": 30,
	"pumpDelay": 5,
	"burnMinPeriod": 10,
	"burnMinPower": 0,
	"burnMaxPower": 80,
	"boilHystOn": -6,
	"boilHystOff": 6,
	"wWStarts": 8618,
	"wWWorkM": 31310,
	"UBAuptime": 2021381,
	"burnStarts": 32152,
	"burnWorkMin": 558028,
	"heatWorkMin": 526718,
	"serviceCode": "-H",
	"serviceCodeNumber": 200
}
------------------

from the console

Boiler: Topline/GB162 (DeviceID:0x08, ProductID:115, Version:05.09)
  Heating active: on
  Warm water/DHW active: off
  Service code: -H
  Service code number: 200
  Warm water selected temperature: 60 °C
  Warm water set temperature: 62 °C
  Warm water disinfection temperature: 70 °C
  Selected flow temperature: 60 °C
  Burner selected max power: 75 %
  Burner current power: 22 %
  Pump modulation: 30 %
  Warm water type: buffer
  Warm water charging type: 3-way valve
  Warm water circulation pump available: on
  Warm water circulation pump freq: 2x3min
  Warm water circulation active: off
  Warm water current temperature (intern): 61.5 °C
  Warm water current temperature (extern): 61.5 °C
  Warm water current tap water flow: 0.0 l/min
  Current flow temperature: 60.5 °C
  Return temperature: 48.2 °C
  Mixing switch temperature: 0.0 °C
  System pressure: 2.2
  Max boiler temperature: 56.2 °C
  Warm water storage temperature (intern): 46.5 °C
  Warm water storage temperature (extern): 61.5 °C
  Warm water activated: on
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: off
  Warm water recharging: off
  Warm water temperature ok: on
  Warm water activated: off
  Gas: on
  Flame current: 16.6 uA
  Boiler pump: on
  Fan: on
  Ignition: off
  Warm water heating: off
  Heating activated: on
  Heating temperature setting on the boiler: 60 °C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 30 %
  Boiler circuit pump delay time: 5 min
  Boiler burner min period: 10 min
  Boiler burner min power: 0 %
  Boiler burner max power: 80 %
  Boiler temperature hysteresis on: -6 °C
  Boiler temperature hysteresis off: 6 °C
  Warm water # starts: 8618
  Warm water active time: 31310
  Burner # starts: 32152
  Warm Water active time: 21 days 17 hours 50 minutes
  Total burner operating time: 387 days 12 hours 32 minutes
  Total heat operating time: 365 days 18 hours 42 minutes
  Total UBA working time: 1403 days 17 hours 46 minutes

Thermostat: RC20/Moduline 300 (DeviceID:0x17, ProductID:77, Version:03.06)
  Time: 08:36:31 27/10/2020
  Error code: (0)
  Heating circuit 1:
    Setpoint room temperature: 20.5 °C
    Current room temperature: 19.5 °C
    Mode: auto

----------------

jtonk, 2020-10-27:
sorry accidentally closed the issue.

I've noticed I have 2 heating active topics in Mqtt
heating_active and heatingActive. If I remove both will ems-esp regenerate the HA auto discovery topics?

[image]

jtonk, 2020-10-27:
Ok, the issue was due to old Mqtt discovery topics. I deleted all, rebooted ems-esp, and not it appears that all (supported) sensors are working again.

sorry for your time, I could have figured this out myself. Closing this topic.

closed 2020-10-27 by jtonk

proddy, 2020-10-27:
no worries. you actually raised a good point. Since the MQTT topic names have changed there will be "retained" messages left over in the MQTT broker that need to be removed. I'll mention that in the documentation.