emsesp/EMS-ESP #580: ESP8266 on Tx don't send data

misu1lica, 2020-10-26:
Hi
I have a Buderus Logamax and I am trying to connect it via EMS with the EMS Buss gateway Calduina schematic, connected to ESP8266 nodemcu v3.
I updated the firmware, V2.01 for ESP.
I configured ESP, I checked Tx with the oscilloscope, but it doesn't transmit anything.
I set all the Tx Mode variants from the settings page.
On the page I see only the temperature for Dallas sensor.
Why doesn't the Tx transmission from the ESP8266 start?
Misu

proddy, 2020-10-26:
if you don't see anything on the Tx line using the scope then the problem is most likely with the circuit. I suggest you first use Calduina's software (I think you mean https://github.com/danimaciasperea/Calduino) because that should work your circuit design. And when that's working you can migrate over to EMS-ESP.

Also consider buying a circuit from bbqkees which are guaranteed to work. 

See https://emsesp.github.io/docs/#/Troubleshooting

MichaelDvP, 2020-10-27:
@misu1lica: the ems-clients can only send after they receive a poll from the bus-master. Check rx-line first, if there is something wrong ems-esp will never send anything.

bbqkees, 2020-10-27:
First check if you are using the correct pins for RX and TX on your ESP.
EMS-ESP uses the serial pin swap function on the ESP8266 so it does not use the default set of IO pins.

Getting TX to work with EMS can be tricky.
Make sure RX works first and the ESP can see the data uncorrupted, otherwise nothing will ever get sent.
In general if RX works but TX does not you have to check the TX resistor value.
Somewhere around 170-225 Ohms is a good value if you use an ESP that is powered via USB. 
It also needs to be a 1W or equivalent resistor.
And make sure you have a pull up resistor in front of the TX transistor.
What also helps is a proper ground plane.

There are some more hints here:
https://github.com/bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz/tree/master/Documentation

mciupak, 2020-11-04:
I have the same issue and I do confirm that RX/TX **does not** work on NodeMcu V3 with RX/TX on GPIO13/15.

It **does** work on RX/TX GPIO3/1.

proddy, 2020-11-04:
also configurable via the Settings page.

closed 2020-11-04 by proddy

mciupak, 2020-11-04:
@proddy: despite changing configuration via the Settings page, communication over GPIO13/15 does not work on NodeMcu V3.

proddy, 2020-11-04:
oh, that is strange. Thanks for pointing it out.

realthk, 2020-11-11:
I wonder if rf noise from the wifi module might affect the communication?
I've assembled the circuit on a prototype board with a Wemos Mini powered from the EMS bus (no service connector on my Bosch 2300 boiler) with an external 5V buck converter. 
After powering-up it works for a while, recognises my boiler and its parameters, but very soon both TX and RX errors starts to rise so high, the connection is finally lost.

Because some of the resistors and the opamp is quite close to the Wemos on my test board, I'll try moving the wifi module away with jumper cables to see if that causes the problem.

bbqkees, 2020-11-11:
'WiFi noise' is usually not an issue. On my Gateway boards the Wemos antenna is directly over the comparator.

Try powering the Wemos via USB and see if that solves the problem. I think it might be a EMS bus power issue. 
What capacitor size did you use?

realthk, 2020-11-11:
2200 uF, I guess it should be enough. I measured somewhat below 14V (13.8V or so) on it after power-up. I though it might be safer for the boiler using the EMS bus only instead of an external power source, but then I'll first try removing the buck converter and power the Wemos from a USB charger.

bbqkees, 2020-11-11:
2200uF is perfect. 1600uF is the bare minimum, 1800uF is alright but I use 2200uF as well.
Powering from the bus is not always easy to get working properly. In many cases TX will not work or won't work well.
So first try a USB adapter. If it still has the same problems with a USB adapter its really your circuit or wiring.

realthk, 2020-11-11:
Basically this is how I've built the circuit, with small differences because of the raster distance of the prototype board:
[image]
But let's hope it's simply EMS power issue indeed, I'll try it tonight.

bbqkees, 2020-11-11:
If you power from the bus 2x470 Ohm in parallel might not be low enough. Try to reach around 170~180 Ohms. You can add a third 470 in parallel with the other two, that will result in 150 Ohms that will definitely do.

realthk, 2020-11-11:
Knocking on wood, it seems your assumption was right Kees: after a few thousands of received, hundreds of sent telegrams, both RX and TX quality is still at 100% with powering the circuit from USB now instead of the EMS bus!
 (though it was a bit scary to see when first testing just the board, without connecting it to the bus at all, that it already "receives" false telegrams, with error rates in the sky - but perhaps it's normal to have such noise with those lines floating)
Still not too keen on having externally powered electronics connected to the boiler (I have put 250 mA fuses on the board, but those protect only against current, not over-voltage in an unlikely but possible case the USB adapter goes berserk) so I might experiment later with powering from the EMS bus again, but for now it's fine, thanks!