emsesp/EMS-ESP #581: (v2.1x) New dallassensor mqtt topic usage?

giovanne123, 2020-10-26:
Tested #540 with 2.1.0b12, esp8266
Manual edit of climate.yaml, no usage of Auto discover.
Do I something wrong in the value_json?
See error in HA (0.116.4) below:

Mqtt topic and data is received as expected, controlled in mqtt-explorer.

```
current_temperature_topic: "ems-esp/dallassensor_data"                       

current_temperature_template: "{{ value_json.10-7E42-7001-0800 }}"
```

`2020-10-26 22:16:46 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: dict object has no element 10 (value: {"10-7E42-7001-0800":20.2}, template: {{ value_json.10-7E42-7001-0800 }})
`

proddy, 2020-10-27:
The json key has special characters so you can't use the value_json.* syntax. Use [key] instead. See https://www.home-assistant.io/docs/configuration/templating/#processing-incoming-data

Also you can test yourself in HA using Developer Tools->Template and something like

```yaml
{% set test = {
    "28-EA41-9497-0E03": 18.2
} %}

{{ test['28-EA41-9497-0E03'] }}
```

proddy, 2020-10-27:
and you really should be using the HA format if using Home Assistant. If you don't want to use MQTT discovery turn it off in configuration.yaml.

giovanne123, 2020-10-27:
Thanks, is working.
I will give the HA format a try. Wasn't aware that there is a separate MQTT Option in EMS-ESP. 
Also will have a look to the MQTT discovery. For the moment I edited manually to understand more and more, but maybe not the right way ;-)

giovanne123, 2020-10-27:
@proddy, one question before I start to modify all my setup to mqtt HA option and discovery:
Is this https://github.com/proddy/EMS-ESP/issues/438#issuecomment-690944291 still fact?
Because my BuderusGB132/RC30 is also configured to only be outdoortemp controlled and there isn't a "currtemp" reported therefore I currently using the external DallasSensor (attached to EMS-ESP) which is manually edited in the climate.yaml for "currtemp".
Won't this work with discovery setup?

proddy, 2020-10-27:
good point, it will most likely show an error in the HA logs. I'm not sure how bad that is. I'll have to try it out myself and see.

proddy, 2020-10-27:
@giovanne123 see https://github.com/proddy/EMS-ESP/issues/582

closed 2020-10-27 by proddy