emsesp/EMS-ESP #582: Prevent HA MQTT config messages for thermostat that has no 'currtemp'

proddy, 2020-10-27:
Some thermostats do not output the current room temperature and with HA's MQTT Discovery it will continuously show errors in the log like:

[image]

The the solution is to check the setting on the Thermostat to see if its "outdoortemp controlled" and if so prevent these HA config messages from being sent.

An HA climate component can still be used by manually adding a climate.yaml and picking which current temperature value you want to use e.g. the outdoor temp or one from the dallas sensors if attached.

proddy, 2020-10-27:
@MichaelDvP any idea how to check if (for example an RC30) is configured not to show current room temperature?

MichaelDvP, 2020-10-27:
For the RC35 there is a value in set_typeids in offset 33, 0 for weathercontroled, 1 for roomcontrolled. But if the setting is weathercontrolled with roominfluence, you should also check the roominfluence (offset 4) and the control (offset26). I did not find where a "mounted on boiler" value is published.  

I think it's much easier to check the current roomtemperature in monitor-message, this is 0x7D00 if the circuit is only weather controlled.

If the RC35 is weathercontrolled, mounted on boiler and a roominfluence is given by a remote RC20(RF) or the internal remote-simulation, RC35 publishes the roomtemp in the monitor-message.

proddy, 2020-11-02:
@PhillyGilly is getting the same errors. He doesn't have a room thermostat connected, his boiler fires up when the Underfloor Heating Manifold tells it to via a 240v signal.

EMS-ESP reads the values from the telegrams and classifies them either as a real value (integer, float to boolean) or what we called 'unset' which means it has yet to receive a value. I'll extend the code to support an additional condition "not active" which we can use to detect when sensors are missing. Then prevent the HA climate config from being created which is causing the errors.

proddy, 2020-11-17:
the creation of the homeassistant/climate topic is skipped for a heating circuit that has no active room temperature

Using the weather controlled sensor can be new GitHub enhancement issue if this is needed

joanwa, 2020-11-18:
also #438

proddy, 2020-11-18:
realized that the trick i did is not optimal. We really should be using outdoor temp is it's available.

PhillyGilly, 2020-11-18:
I'm not clear why a "wrong" curtemp is better than a nul value. I think my thermostat must assume a low value if the roomstat isn't there. I can investigate locally tomorrow and see if there is another alternative value.

MichaelDvP, 2020-11-19:
I'm not familiar with HA, corrcet me if i'm wrong: There is a common used climate widget, that is not very flexible with the values. 
It's used for setting temperature and mode and shows also the current room temperature. If there is no room temperature it does not work. To have this widget work we need to publish fixed modes (off, heat ,auto)  and also subscribe to an extra topic `thermostat_hcx`
If we don't have a room temperature we have to send a number to the widget to make it work. 

My suggestion: 
create a value `haTemp` in the mqtt and set the HA-climate to uses this as roomtemperature. 
If we have currtemp (roomthermostat) , publish it also as haTemp, If thermostat is weather-controlled use the setpoint_roomtemperature also as haTemp. That's not completly wrong, the thermostat internally sets the flowtemperature to reach this roomtemperature and assumes always that setpoint and roomtemp are identical (if not: heatingcurve is wrong).
As a third option we can add a command `roomtemp` and send a value to ems-esp that is republished as haTemp. This allows users to measure roomtemperature with a aqara-sensor, a sonoff-T1 or something else and inject the temperatue to the climate-widget.

proddy, 2020-11-19:
I was thinking along the same lines. Your hypothesis of HA is correct, the Home Assistant climate component (which is just a fancy dial) only works when there is a current temp, setpoint temp and those fixed modes you mentioned. If they are not there the component will not be shown in a HA dashboard, which is fine, but the log file with fill up with errors because it cannot find the values in the MQTT. In EMS-ESP 2.1.1 I prevent the climate component from being created, which is also not very useful.

I agree on your suggestion. Have an option to configure which current room temperature sensor to use
 - "none": which will use a value of 0
 - "thermostat": which will use the thermostat currtemp (if it exists, otherwise 0)
 - "outside": using the thermostat supported outdoor/outside/weather sensor (if it exists, otherwise 0)
 - "other": which points to an MQTT topic which has the value, like your sonoff example

this would be a pull-down select setting in the WebUI for the Master Thermostat.

not sure though how to handle the different heating circuits?

MichaelDvP, 2020-11-19:
Why not select automatically by checking the valid value in hierarchical order:
```
           if (Mqtt::mqtt_format() == Mqtt::Format::HA) {
                if (Helpers::hasValue(hc->ha_temp)) {
                    dataThermostat["hatemp"] = Helpers::round2((float)hc->ha_temp / 10);
                } else if (Helpers::hasValue(hc->curr_roomTemp)) {
                    dataThermostat["hatemp"] = Helpers::round2((float)hc->curr_roomTemp / curr_temp_divider);
                } else {
                    dataThermostat["hatemp"] = Helpers::round2((float)hc->setpoint_roomTemp / setpoint_temp_divider);
                }
            }
```
and set the ha_temp by 
```
bool Thermostat::set_roomtemp(const char * value, const int8_t id) {
    float f = 0;
    if (!Helpers::value2float(value, f)) {
        LOG_WARNING(F("Set roomtemperature: Invalid value"));
        return false;
    }
    uint8_t                                     hc_num = (id == -1) ? AUTO_HEATING_CIRCUIT : id;
    std::shared_ptr<Thermostat::HeatingCircuit> hc     = heating_circuit(hc_num);
    if (hc == nullptr) {
        return false;
    }
    if (f > 100 || f < 0) {
        hc->ha_temp = EMS_VALUE_SHORT_NOTSET;
    } else {
        hc->ha_temp = (int16_t)(f * 10);
    }
    return true;
}
```

PhillyGilly, 2020-11-19:
@proddy I like the four option plan and whilst @MichaelDvP's hierachy is neat it may have unintended consequences. I'll post up a few pictures to throw in some of my findings below.

PhillyGilly, 2020-11-19:
OK. I'm trying to work out my question, but it is something like "if my system runs without a room stat haven't Bosch already done the work here?"
This is my Sense II showing Flow Temperature and Outside Temperature
[image]
The smaller temperature display is configurable as shown below. 
[image]
The bigger temperature display seems to depend on where the controller is mounted:
[image]
My Sense II is in the boiler which is in my plant room in the loft. As this room is unheated it is generally colder than the house (except in summer when we don't need any heating!) so I set the target room temperature to 30. 
[image]
The heating comes on and I'm assuming that this is because set_Temp > curr_Temp even though there is no measured value. (Is there any merit in reducing the set_Temp or in moving the Sense II to a wall mount?)
Another post follows...

PhillyGilly, 2020-11-19:
The question here is something like "how does the solution deal with multizone heating systems?"
My heating system may not be typical but it cannot be unique and the problems that I have faced with occupancy going from two people upto ten people and high solar gain in some but not all rooms means that I have almost every room in its own temperature controlled zones.
This is my heating manifold with the valves that control the flow into UFH in all the rooms.
[image]
The white valves are controlled individually (or in pairs) by thermostats in each zone which communicate in zigbee to the Heatmiser neohub above the manifold. The hub sends a 240V live signal to the boiler when set_temp > curr_Temp in any zone. The hub also runs the manifold mounted pump to circulate hot water through any active zones.
The boiler responds to the "call for heat" signal and heats the water according to the external temperature curve and runs its internal pump to send hot water to the manifold through the red valve. The boiler temperature increases and a heat cool cycle starts. This is shown here.
[image]
Each of the thermostats is displayed in HA as a climate entity in its own right e.g.
[image]
I have 8 current temperatures which are in HA but not in MQTT all of which might be the controlling value.  So if I wanted to pass a value back I'm not sure what it would be, how I would do it, or indeed what I would gain from that?

MichaelDvP, 2020-11-19:
> I'm assuming that this is because set_Temp > curr_Temp

No! The thermostat don' t know the curr_Temp. It's simple the calculation from heating curve:
for a given set_Temp look up the curve value at position of outside temperature.

PhillyGilly, 2020-11-19:
> > I'm assuming that this is because set_Temp > curr_Temp
> 
> No! The thermostat don' t know the curr_Temp. It's simple the calculation from heating curve:
> for a given set_Temp look up the curve value at position of outside temperature.

I think we must have a jargon breakdown here.
[image]
The set Flow Temperature TFL is derived from the look up curve: but the set Room Temperature must be set elsewhere by the user?

MichaelDvP, 2020-11-19:
Right, the set (required/desired) Room temperature is user input and causes a parallel displacement [4] and [5]. 
What i want to express is, that the current roomtemperature is not used in this control. The thermostat assumes that the current roomtemperature is identical to the setpoint if the curve is set correctly and calculated only the flowtemp.

PhillyGilly, 2020-11-19:
I think I understand what your saying - at least in part - I have toyed with the 4/5 displacement  settings as  a way of improving heating times albeit at the cost of overall efficiency. The bit I don't get is does the boiler know when to turn off because the room is now warm enough? I envisaged that the TFL is set as in Fig 23, but wrapping around that is another go/no-go control comparing the set and measured room temperatures. In my system this calculation is done in the Heatmiser neohhub. There is information on how to set up the Sense II to do this too if it is wall mounted in a habitable zone.
In this picture the TFL control is running where the green line appears to be thicker, but it  is off where the line appears to be thinner. This is confirmed by boiler hours in history. 
[image]

glitter-ball, 2020-11-21:
Coming to this a bit late, but...

I thought you could use a SENSE I as a remote sensor / controller if you had the SENSE II fitted in the boiler fascia?

With your SENSE II in the boiler fascia, I think the large temperature displayed on it will be boiler flow temp, as the Bosch manual extract you included suggests. So, it's not sensing the plant room temperature at all and the 30ºC setting is something to do with the heat curve selection?

If you moved the SENSE II to the heated space on a wall mount, I don't think you'd lose any functionality and you'd then be able to monitor the temperature of the heated space. You'd also be able to enable weather compensation with room compensation - this tweaks the calculated flow temperature up or down based on the error between the measured room temp and the set temperature. So I have pretty much the default curve set with (from memory) an 8K room compensation value. As the room heats up, the boiler drops its flow temp:

[image]

Here (ignoring the two DHW bursts...) you can see the target flow temp drop as the space heats up. I think this is probably the best control mode - you don't have to get the heat curve perfect and the system will respond to solar or other heat gains (cooking, open fire / stove) by dropping heat input. You can also see it crank up if someone leaves the front door open *8( !

So... if you can get some EMS bus wiring from your boiler to a representative location in the heated space, I'd give it a try.

glitter-ball, 2020-11-21:
@PhillyGilly - curious what the error is on your Sense II display? I had issues with the software in my boiler being incompatible with the Sense II. It was moaning the water pressure sensor was faulty, when the boiler doesn't have one! W-B came an fitted a new HCM (heat control module - boiler firmware in what looks like a weird USB key) and that fixed all the errors.

glitter-ball, 2020-11-22:
@PhillyGilly - The 'B' on your default display is interesting too. Is that because your Heatmiser system is wired into the boiler's external cut-off circuit? I had loads of fun trying to decipher the manual to sort this out on my system.
[image]
I also had to disable the basic weather compensation in the boiler (Menu 5.W1) to get the Sense II advanced compensation going. It was a mare! None of it immediately obvious and a lot of trial and error. No wonder installers shy away from this more complex stuff...

proddy, 2020-12-13:
the implementation will be part of #644

closed 2020-12-13 by proddy