emsesp/EMS-ESP #583: Recover corrected/damped external temp from Sense II / CW400

glitter-ball, 2020-10-27:


glitter-ball, 2020-10-27:
Sorry - have somehow  managed to create this with no labels *8( Tried to add a feature request one...

MichaelDvP, 2020-10-27:
Try with `read 10 A3` if there is a reply.

proddy, 2020-10-27:
we only render the damped external temp (`dampedoutdoortemp_`) if the thermostat model is an EMS_DEVICE_FLAG_RC35. Not an EMS_DEVICE_FLAG_RC300 so we'll need to adjust the code.

glitter-ball, 2020-10-28:
```
ems-esp:/# read 10 A3
000+22:45:44.388 N 3: [emsesp] Thermostat(0x10) -> Me(0x0B), RCOutdoorTemp(0xA3), data: <empty>
ems-esp:/#
```
I suppose I should have a rummage in other thermostat telegrams, then, to look for it?

glitter-ball, 2020-10-28:
Thermostat shows corrected external temp as 10.8ºC so put a watch on and went looking for 6C hex (6C = 108, assuming temperature is divided by 10 in this telegram) but haven't spotted it yet.

glitter-ball, 2020-10-28:
I can see the external temp in the UBAMonitorSlow telegrams from the boiler - no surprises as the boiler is doing the external temp measurement. But nothing for the corrected temp from the thermostat - unless the vit's in a different format. There's always the possibility that the thermostat doesn't broadcast it because there's no real need for it to do so.

Here's a bit more capture...

```
ems-esp:/# watch on 10
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x10
000+23:04:50.165 N 163: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 01 85 01 0F (offset 13)
000+23:05:03.222 N 164: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 B3 21 00 00 00 00 25 01 85 01 01 03 01 85 01 0F 00 00 11 01 00 07 01 00
000+23:05:03.527 N 165: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 3C 01 FF 01 02 (offset 25)
000+23:05:03.812 N 166: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 1E 02 2A FF 20 00 25 01 E1 20 01 0F 05 00 00 02 21
000+23:05:04.137 N 167: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
000+23:05:04.906 N 168: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 28
000+23:05:05.277 N 169: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0A 0C 1C 2E 10 02 00 10 FF FE
000+23:05:18.082 N 170: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+23:05:18.132 N 171: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
000+23:05:18.368 N 172: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 B3 21 00 00 00 00 25 01 85 01 01 03 01 85 01 0F 00 00 11 01 00 07 01 00
000+23:05:18.778 N 173: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 3C 01 FF 01 02 (offset 25)
000+23:05:18.976 N 174: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 09 07
000+23:05:49.684 N 175: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 01 84 01 10 (offset 13)
000+23:06:03.253 N 176: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 B3 21 00 00 00 00 25 01 84 01 01 03 01 84 01 10 00 00 11 01 00 07 01 00
000+23:06:03.526 N 177: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 3C 01 FF 01 02 (offset 25)
000+23:06:03.859 N 178: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 1E 02 2A FF 20 00 25 01 E1 20 01 0F 05 00 00 02 21
000+23:06:04.136 N 179: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
000+23:06:04.976 N 180: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 28
000+23:06:06.125 N 181: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0A 0C 1C 2F 11 02 00 10 FF FE
000+23:06:07.002 N 182: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 01
000+23:06:07.056 N 183: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 28 (offset 3)
000+23:06:10.300 N 184: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBADevices(0x07), data: 0F
000+23:06:10.321 N 185: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00
000+23:06:18.033 N 186: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+23:06:18.083 N 187: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
000+23:06:18.279 N 188: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 09 07
000+23:06:52.637 N 189: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 01 83 01 11 (offset 13)
000+23:07:03.420 N 190: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 B3 21 00 00 00 00 25 01 83 01 01 03 01 83 01 11 00 00 11 01 00 07 01 00
000+23:07:03.725 N 191: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 3C 01 FF 01 02 (offset 25)
000+23:07:04.010 N 192: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 1E 02 2A FF 20 00 25 01 E1 20 01 0F 05 00 00 02 21
000+23:07:04.687 N 193: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
000+23:07:05.504 N 194: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 28
000+23:07:06.324 N 195: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0A 0C 1C 30 11 02 00 10 FF FE
000+23:07:18.280 N 196: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+23:07:18.330 N 197: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
000+23:07:18.566 N 198: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 B3 21 00 00 00 00 25 01 83 01 01 03 01 83 01 11 00 00 11 01 00 07 01 00
000+23:07:18.976 N 199: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 3C 01 FF 01 02 (offset 25)
000+23:07:19.174 N 200: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 09 07
ems-esp:/# watch off
Watching telegrams is off
ems-esp:/#

```

MichaelDvP, 2020-10-28:
In RC35 the damped temperature is published in type 0xA3 as a singel byte in degrees without decimals. Your thermostat does not have this telegram, so it's published somewhere else. Also norbert lists only the hc-dependend telegrams, not the device specific. You can watch the telegrams from thermostat and change the building-type on thermostat, this gives a clue what telegrams are used for device settings. 

The measured outdoor-temperature with decimals is published by boiler, do you have this?  (`read 08 19`, first two bytes).

glitter-ball, 2020-10-28:
```
ems-esp:/# read 08 19
000+23:11:38.932 N 201: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 00 69 80 00 80 00 00 00 00 00 00 33 E2 02 68 BD 00 00 00 01 D0 E4 00 25 7F 80 00
ems-esp:/#
```

So, outdoor temp is 69 hex, 105 decimal, 10.5ºC, so that's all good. It's always possible the thermostat doesn't publish the damped external temperature...

MichaelDvP, 2020-10-28:
267 is listed as floordrying on norberts list. 2F5 is implemented as RC300WWmode, i think you are using an older software.
Device specific settings are not published frequntly, they are published if something changed. To get these telegrams you have to change sttings on the thermostat and search the log for new telegrams and these values.

glitter-ball, 2020-10-28:
A bit more capture of the mystery telegrams which aren't giving much away! 

```
ems-esp:/# watch on 2f5
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0x2F5
000+23:16:04.246 N 208: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
000+23:17:04.197 N 209: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
000+23:18:04.172 N 210: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
000+23:19:04.243 N 211: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
ems-esp:/# watch off
Watching telegrams is off
ems-esp:/# watch on 267
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0x267
000+23:20:18.468 N 212: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
000+23:21:18.467 N 213: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
000+23:22:18.667 N 214: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
000+23:23:18.490 N 215: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
ems-esp:/# watch off
Watching telegrams is off
ems-esp:/#
```

I'll play with the hot water and see if 2F5 changes...

glitter-ball, 2020-10-28:
Can you post a link to Norbert's list please? Struggling to find it. Thanks.

glitter-ball, 2020-10-28:
@MichaelDvP - you mention 2F5 in #518 which is also for CW400.

MichaelDvP, 2020-10-28:
Known in 2F5 is pos 2: wwmode 0=off, 1=low, 2=high, 3=auto, 4=own prog (in v2.1.0b12)
and maybe circulation mode in pos 1, 0=off, ?? (we havn't found someone with circulation pump attached)

[Norberts documentation](https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf)

glitter-ball, 2020-10-28:
> 2F5 is implemented as RC300WWmode, i think you are using an older software.

I have a Sense II (CW400 sold in UK) with firmware NF 33.03. I've seen units for sale on eBay now with NF 33.04.

MichaelDvP, 2020-10-28:
I've meant the ems-esp software, the actual beta-version shows 2F5 as know and named telegram.

glitter-ball, 2020-10-28:
> I've meant the ems-esp software, the actual beta-version shows 2F5 as know and named telegram.

Hmm... that's weird. I'm running 
```

┌──────────────────────────────────────────┐
│ EMS-ESP version 2.1.0b12                 │
│ https://github.com/proddy/EMS-ESP        │
│                                          │
│ type help to show available commands     │
└──────────────────────────────────────────┘

```

MichaelDvP, 2020-10-28:
Oh, i see it's a bug in type-names, all telegrams with type-id xFx are wrong displayed. 
I fix this.

glitter-ball, 2020-10-28:
@MichaelDvP - thanks. Glad to help track that down. That sorts the 0x2F5 telegram. 0x267 is, as you say, floor-drying (i.e. controlled curing of concrete screed around underfloor heating piping) which is pretty niche stuff and would probably only get used once for each heating circuit. So not top priority for decoding!

As to my original question... not sure the thermostat is publishing the damped temperature then. I guess it's getting the realtime temperature from the boiler and calculating the damped temperature internally based on building type setting. There is nothing else that's going to need it for flow temperature calculations (the thermostat does it all for up to four heating circuits) so there's just no point publishing the damped temperature on the bus.

So, unless I'm missing something, this may not get anywhere!

MichaelDvP, 2020-10-29:
The 267 is a short message, only status of floordrying and temperature, but i think it is hc-dependent, do you get messages for `read 10 268`, `read 10 269`, `read 10 26A`? If you need this message i can add it. According to Norbert there are telegrams 29B, ff. for the settings "enable floordrying" and "heating_type", etc. Do you get these messages (`read 10 29B`)? 

The internal values of the thermostat are not published frequently. Only if you change a setting like building, the thermostat publish the change. For the RC300 compatible thermostats there is no documentation about these telegrams, they have to be discovered by someone with this thermostat.

glitter-ball, 2020-10-29:
Results...
```
ems-esp:/# read 10 267
001+21:27:40.113 N 5: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x267), data: 00 00
ems-esp:/# read 10 268
001+21:27:43.606 N 6: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x268), data: <empty>
ems-esp:/# read 10 269
001+21:27:47.668 N 7: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x269), data: 09 01 01 09 01 01 02 08 02 FF 00 00 00 00 00 00 00 FF 00
ems-esp:/# read 10 26A
001+21:27:52.964 N 8: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x26A), data: 09 01 01 09 01 01 04 11 02 00 00 00 00 00 00 00 00 00 00
ems-esp:/# read 10 29B
001+21:27:59.703 N 9: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x29B), data: 01 01 00 FF FF 01 05 30 4B
```
Surprised, I think. If we think 267 is screed drying for hc1, 268 for hc2, 269 for hc3 and 26A for hc4, then this looks odd. I only have hc1 defined. That would explain, maybe, why 268 is empty. But why is there data for 269 and 26A?! Not sure about 29B.

I'll try changing building type and see what the thermostat publishes.

glitter-ball, 2020-10-29:
OK...
```
001+21:32:48.563 N 27: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x240), data: 03 (offset 9)
001+21:32:48.761 N 28: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 01 3C 00 B3 (offset 13)
001+21:32:55.957 N 29: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x240), data: 02 (offset 9)
001+21:32:57.238 N 30: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
001+21:32:57.287 N 31: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
001+21:32:57.484 N 32: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 09 07
001+21:33:00.159 N 33: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x240), data: 01 (offset 9)
001+21:33:04.698 N 34: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 B3 21 00 00 00 00 25 01 3C 01 01 03 01 3C 00 B3 00 00 11 01 00 06 FD 00
001+21:33:04.980 N 35: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 97 02 FF 01 02 (offset 25)
001+21:33:05.265 N 36: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 1E 02 2A FF FF 00 25 01 E1 20 01 0F 05 00 00 02 21
001+21:33:05.590 N 37: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
001+21:33:05.830 N 38: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 28
001+21:33:07.434 N 39: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x240), data: 02 (offset 9)
```
Building type looks to be published in telegram 0x240. Data is 01:light, 02:medium, 03:heavy.

glitter-ball, 2020-10-29:
Playing around, I tried...
```
ems-esp:/# read 10 23A
001+21:40:48.927 N 53: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x23A), data: 00 68
```
68 hex is 104 decimal and my damped outside temperature currently is 10.4ºC. I will monitor and see if I can track a change.

glitter-ball, 2020-10-29:
OK - it's just increased by 0.1ºC and I have 69 hex now in telegram 0x23A. So... that's where the damped temperature is hiding. I probably need to see next if the thermostat pro-actively publishes it on a change. Could be a long wait...

glitter-ball, 2020-10-29:
OK. I've seen the damped temperature change (by checking read 10 23A) and the thermostat has not published it when it's changed. So, I may have to poll the thermostat to get it.

glitter-ball, 2020-10-29:
Looks like it's an error reporting telegram. First byte is the source device (10 - thermostat). Second byte is Product ID (158 in my case). Rest is zero as there are no errors currently. Maybe more detailed version of RCError 0xA2?

MichaelDvP, 2020-10-29:
Ok, let's summerize:
- 0x267 is a single message for all hc showing floordrying state and temperature
- 0x29B, ff as descriped from Norbert, settings per hc for optimization, window-detection, floordrying, do you need these parameters? read and write?
- 0x240 Settings, building in pos 9, we have this for RC35, so i add this. Other settings from this message?
- 0x23A damped outdoor temperture, i'll add it and fetch every minute.
- 0xBF error message, Do you have also the 0xA2, or is errorcode missing and we should read the errorcode from 0xBF

glitter-ball, 2020-10-29:
@MichaelDvP - thanks for all this investigation.

I would say that 0x267 and 0x29B are rarely changed, so would be low priority to implement.

0x240 - Do we know what else is in this telegram? I couldn't see it in Norbert's list.

0x23A - thanks!

0x2A / 0xBF : I have both. I guess 0xA2 is basic error info and 0xBF gives more detail if needed. Maybe 0xA2 is enough for most purposes? 0xBF might be low priority to implement then.
```
> ems-esp:/# read 10 A2
> 002+07:15:11.232 N 1: [emsesp] Thermostat(0x10) -> Me(0x0B), RCError(0xA2), data: 00 00 00 00 00
> ems-esp:/# read 10 BF
> 002+07:15:21.395 N 2: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0xBF), data: 10 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

glitter-ball, 2020-11-12:
Lots going on, re-visited this. I think we worked it all out and it needs code at some point:

> @MichaelDvP 
> 0x23A damped outdoor temperture, i'll add it and fetch every minute.

That should do it - thanks!

proddy, 2020-11-18:
@glitter-ball can we close this enhancement? I want to get 2.1.1 out

MichaelDvP, 2020-11-18:
>  I want to get 2.1.1 out

Oh, i have some smaller changes to boiler, thermostat, esp32-mqtt,.. tested the last days.
I'll do some PR for that, than you decide if it's better to ad this to next dev.

glitter-ball, 2020-11-18:
I think so. Michael has added - I still need to test. Let me do that in the next release and come back if there’s an issue. Thanks, both!

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/proddy/EMS-ESP/issues/583#issuecomment-729518179",
"url": "https://github.com/proddy/EMS-ESP/issues/583#issuecomment-729518179",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

glitter-ball, 2020-11-21:
Upgraded to 2.1.0 and damped temp is now available. Great - thanks! Happy for you to close if that's appropriate.

proddy, 2020-11-21:
Nice work Michael!

closed 2020-11-21 by proddy