emsesp/EMS-ESP #584: Question: (REST API) Getting Invalid device response for mixer device type

samvdb, 2020-10-28:
When trying to get the info for a mixing module i get "Invalid device" as a response.

Example call: /api?device=mixer&cmd=info&id=hc1
I have tried many different combinations with id=hc 1, hc2 0x20 etc...

Where can i see a list of valid device names that can be used and how to i query these mixing valves?

[image]

MichaelDvP, 2020-10-28:
There are some changes in the latest betas. One is renaming mixing to mixer. You are using a version with older `mixing`, so you have to use this in api too ` /api?device=mixing&cmd=info` and if you want to use the id, it's `id=1`.
Or update to latest version and use `mixer`.

samvdb, 2020-10-28:
Thank you so much! Worked instantly!

closed 2020-10-28 by samvdb