emsesp/EMS-ESP #587: Up arrow key in console recalls last command *8)

glitter-ball, 2020-10-29:
**Is your feature request related to a problem? Please describe.**
As a reflex action, I press up-arrow in EMS-ESP console to repeat the last command. I get '[A' "Command not found back." *8(

**Describe the solution you'd like**
In console, arrange for cursor up arrow key to recall previous command (like UNIX / DOS CLI)

proddy, 2020-10-29:
nice one Michael. This has been on my to-do list for a while

glitter-ball, 2020-10-29:
@MichaelDvP @proddy - thanks, both! Nice extra feature.

glitter-ball, 2020-10-29:
Is it then possible to use left and right arrows to move the insertion point through the line buffer and allow editing? I know this is suddenly asking a lot more, so no worries if not. The recall, alone, is worth having.

MichaelDvP, 2020-10-30:
There isn't a real cursor, the terminal sets a block at the end of the line, i have not found where it is set and how to write something behind this block. But i can do something similar: on left cursor shift the chars to a buffer and delete them from monitor, on right place them back.

glitter-ball, 2020-10-30:
We are spoilt to have editing on bigger hardware *8) and forget the overhead to implement even simple stuff like this!

Great to have, if the resources are there, but there might be more important things.

How long before Bosch/Buderus engineers start buying this cool kit (assuming they aren’t already ;-) )?!

proddy, 2020-10-30:
@MichaelDvP the left-right cursor is a bit strange. left deletes the character and right replaces it. I see what you're trying to do but it's not normal shell behaviour. Also the buffer should be cleared after each command. I would just remove that feature and stick to the cursor-up and then we can release 2.1

MichaelDvP, 2020-10-31:
@proddy i agree, remove it and release the v2.1. But please keep the removal of cursor chars like `[B`.

In comming v2.1.x betas we have more time to discuss, test, and modify experimental features.

proddy, 2020-10-31:
@MichaelDvP are you able to do it? You know exactly which lines you added

MichaelDvP, 2020-11-01:
Second try in the new dev. I've found the cursor controls, added esc-sequence-detection and as goodie added function keys: F1 for help, F2 show, F10 report, F9 starts `send telegram "0B "` for faster typing.

Please check if i have missed something.

proddy, 2020-11-13:
works fine over here.

closed 2020-11-13 by proddy