emsesp/EMS-ESP #588: Use NTP and EMS-ESP to correct EMS bus clock

glitter-ball, 2020-10-29:
**Is your feature request related to a problem? Please describe.**
My thermostat clock is pretty accurate and I can set an adjustment offset in seconds per week to keep it that way. But, could we do better?

**Describe the solution you'd like**
If the EMS time telegram is writeable, could an option be added to EMS-ESP to send a time sync telegram to the bus, say, once a day? This would only be transmitted if a healthy NTP session is available and subject to some checks (see below). It would appear as a tick-able option on the NTP settings page?

**Describe alternatives you've considered**
This might be yet more clutter in precious memory space. So, it might be better done as an automation in HA.

**Additional context**
Daylight saving and timezones could be tricky. Maybe the code reads the EMS bus clock and only applies a correction of up to +/- 1 minute at xx:30:00 to avoid these issues.

MichaelDvP, 2020-10-29:
`call thermostat datetime NTP` works if NTP is active.
Also after a powercut over longer time if the clock reports error, the NTP-time is set automatically. (That was the main reason to implement it)

glitter-ball, 2020-10-29:
Sorry to be a spoil-sport! There may be an issue with

`call thermostat datetime NTP`

When I tried it, it set the clock OK. But it set the date to 29 September 2020 and not 29 October 2020.

glitter-ball, 2020-10-29:
Nice feature, anyway. Worth a check box in the GUI to enable a daily automatic update? Or is there a better approach?

MichaelDvP, 2020-10-30:
NTP counts month 0-11, buderus 1-12, i fix this. 
I think if someone wants a daily update, it's better to trigger this over mqtt. The checkbox in GUI have to make sure that NTP is enabled and a thermostat is present, and where to set this box, to settings or to NTP?

glitter-ball, 2020-10-30:
@MichaelDvP - thanks. That explains the month problem.

MQTT sounds fine - I could set up an automation in HA to trigger it.

glitter-ball, 2020-11-21:
@proddy @MichaelDvP - Sorry, need to re-open this please.

Did a 'call thermostat datetime ntp' earlier and failed to notice that the date decode is not right yet (see above) - clock now set suddenly to Wed 21 October. No wonder the house was cold *8(. I am running 2.1.0b12 currently. Thanks. Shout if you want me to raise a new bug.

proddy, 2020-11-21:
think it's worth re-opening and taking another look at!

glitter-ball, 2020-11-21:
Thanks - there’s lots going on so it probably slipped through the net. Happy to raise as a new bug if that makes management easier.

susisstrolch, 2020-11-21:
Build: 963a08c
NTP works fine - showing Nov 21st
Set via Telnet...

MichaelDvP, 2020-11-21:
>  I am running 2.1.0b12 currently

Ähm, the fix was after b13.  30.10.2020 13:58
Surely the b12 does not work.

glitter-ball, 2020-11-21:
Can you point me to latest binary? Firmware branch isn't there any more... Thanks.

glitter-ball, 2020-11-21:
OK. Found 2.1.0

glitter-ball, 2020-11-21:
Upgraded to 2.1.0 and now works OK. Thanks. Closed.

closed 2020-11-21 by glitter-ball