emsesp/EMS-ESP #589: Use wwonetime...

glitter-ball, 2020-10-30:
**Question**
Struggling to get wwonetime working on my setup. Being a cautious sort (i.e. a coward!) most of my EMS-ESP use is read-only - watching stuff happen. But there are a couple of things I'd like to be able to do controlwise. One of them is kick off a one-time hot water re-heat. I thought I'd start simple and try to trigger this from the telnet interface. When I issue "call boiler wwonetime 1", nothing appears to happen. Am I doing something silly?

**Screenshots**
```
ems-esp:/# watch on
Watching incoming telegrams, displayed in decoded format
002+20:52:00.009 N 2: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 02 27 00 00 00 02 40 C0 80 00 01 A8 80 00 00 00 FF 00 00 00 CB 00 00 00
002+20:52:00.314 N 3: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 02 09 00 8B (offset 13)
002+20:52:00.530 N 4: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
002+20:52:00.752 N 5: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 01 A8 01 A8 01 00 00 03 00 00 98 2F 00 0E 6B 00 80 00
002+20:52:04.688 N 6: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 12 D9 2A
002+20:52:04.989 N 7: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 4B 64 00 04 F6 0A 01 03 64 50 04 00 00 00 00 00 00 00 00 00 00 00 00 28 00 3C
002+20:52:05.213 N 8: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 3C 00 (offset 25)
002+20:52:05.513 N 9: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 00 88 80 00 80 00 00 00 00 00 00 34 12 02 6A A0 00 00 00 01 D2 71 00 25 A7 80 00
002+20:52:05.712 N 10: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 80 00 (offset 25)
002+20:52:06.046 N 11: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 3C FB 00 28 00 00 46 00 FF FF 00
002+20:52:06.344 N 12: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 B8 21 00 00 00 00 25 02 09 01 01 03 02 09 00 8B 00 00 11 01 00 07 35 00
002+20:52:06.602 N 13: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 C6 04 FF 01 02 (offset 25)
002+20:52:06.935 N 14: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 1E 02 2A FF FF 00 25 01 E1 20 01 0F 05 00 00 02 21
002+20:52:07.058 N 15: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 02 25 00 00 00 02 40 C0 80 00 01 A8 80 00 00 00 FF 00 00 00 CB 00 00 00
002+20:52:07.412 N 16: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
002+20:52:07.700 N 17: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 28
ems-esp:/# call boiler wwonetime 1
002+20:52:09.386 N 18: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorWW(0x34), data: 28 01 A8 01 A8 01 00 00 03 00 00 98 2F 00 0E 6B 00 80 00
002+20:52:10.009 N 19: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 02 25 00 00 00 02 40 C0 80 00 01 A8 80 00 00 00 FF 00 00 00 CB 00 00 00
002+20:52:10.333 N 20: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 02 25 00 00 00 02 40 C0 80 00 01 A8 80 00 00 00 FF 00 00 00 CB 00 00 00
002+20:52:10.610 N 21: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlow(0x19), data: 00 88 80 00 80 00 00 00 00 00 00 34 12 02 6A A0 00 00 00 01 D2 71 00 25 A7 80 00
002+20:52:10.858 N 22: [emsesp] Boiler(0x08) -> All(0x00), UBAMaintenanceStatus(0x1C), data: 93 09 05 1D 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
002+20:52:11.129 N 23: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
002+20:52:11.377 N 24: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 01 A8 01 A8 03 00 00 03 00 00 98 2F 00 0E 6B 00 80 00
002+20:52:11.571 N 25: [emsesp] Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00
002+20:52:18.522 N 26: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBADevices(0x07), data: 0F
002+20:52:18.546 N 27: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00
002+20:52:20.059 N 28: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 02 23 00 00 00 02 40 C0 80 00 01 A8 80 00 00 00 FF 00 00 00 CB 00 00 00
002+20:52:20.279 N 29: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
002+20:52:20.527 N 30: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 01 A8 01 A8 03 00 00 03 00 00 98 2F 00 0E 6B 00 80 00
ems-esp:/# call boiler wwonetime 0
002+20:52:28.580 N 31: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorWW(0x34), data: 28 01 A8 01 A8 03 00 00 03 00 00 98 2F 00 0E 6B 00 80 00
002+20:52:30.008 N 32: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
002+20:52:30.058 N 33: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
002+20:52:30.294 N 34: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 B8 21 00 00 00 00 25 02 09 01 01 03 02 09 00 8B 00 00 11 01 00 07 35 00
002+20:52:30.502 N 35: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 01 A8 01 A8 01 00 00 03 00 00 98 2F 00 0E 6B 00 80 00
002+20:52:30.708 N 36: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 02 20 00 00 00 02 40 C0 80 00 01 A8 80 00 00 00 FF 00 00 00 CB 00 00 00
002+20:52:31.030 N 37: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
002+20:52:31.328 N 38: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 C6 04 FF 01 02 (offset 25)
002+20:52:31.527 N 39: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 09 07
002+20:52:31.878 N 40: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 01
002+20:52:31.924 N 41: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 28 (offset 3)
002+20:52:31.969 N 42: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 09 0A 1E 22 20 02 01 10 FF FE
ems-esp:/# watch off
```

**Device information**
```
ems-esp:/# show
EMS-ESP version 2.1.0b12

Boiler: Worcester Logamax Plus/GB192/Condens GC9000 (DeviceID:0x08 ProductID:208, Version:01.01)
  Heating active: off
  Warm water/DHW active: off
  Service code:
  Service code number: 203
  Warm water selected temperature: 60 °C
  Warm water set temperature: 40 °C
  Warm water disinfection temperature: 70 °C
  Selected flow temperature: 5 °C
  Burner selected max power: 0 %
  Burner current power: 0 %
  Pump modulation: 0 %
[... 55 lines cut ...]
    Summer temperature: 17 °C
    Summer mode: auto °C
    Mode: auto
    Mode type: eco

Dallas temperature sensors:
  Sensor 1, ID: 28-BE2A-7791-0402, Temperature: 39.1 °C

ems-esp:/# show devices
These EMS devices are currently active:

Boiler: Worcester Logamax Plus/GB192/Condens GC9000 (DeviceID:0x08 ProductID:208, Version:01.01)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA

Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:33.03) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x2A5 0x2B9 0x2AF 0x2A6 0x2BA 0x2B0 0x2A7 0x2BB 0x2B1 0x2A8 0x2BC 0x2B2 0x2F5 0x31B 0x31D 0x31E

Controller: ErP (DeviceID:0x09, ProductID:209, Version:01.03)

ems-esp:/#

```

**Additional context**
*Add any other context about the problem here.*

glitter-ball, 2020-10-30:
To complicate things... I have 2-level WW program in use. So 60ºC twice a day and 40ºC in between, off at night. Not sure if this has anything to do with it?

But... when I do one time HW cylinder charge from the thermostat, it always recharges to 60ºC, regardless of time program. Maybe I should capture telegrams when I call it from the thermostat...

proddy, 2020-10-30:
the syntax is correct. Can you try with logging set to debug. Then you should see in the console something like `[boiler] Setting warm water OneTime loading on` and we can also see if the Tx message was successful.

glitter-ball, 2020-10-30:
@proddy - thanks. Capture below - hopefully this is enough.

```
ems-esp:/# call boiler wwonetime 1
002+23:13:46.843 I 113: [boiler] Setting warm water OneTime loading on
002+23:13:46.852 D 114: [telegram] Sending write Tx [#61], telegram: 8B 08 35 00 22 CE
002+23:13:46.878 D 115: [emsesp] Last Tx write successful
002+23:13:46.878 D 116: [telegram] Sending post validate read, type ID 0x34 to dest 0x08
002+23:13:47.053 D 117: [telegram] Sending read Tx [#62], telegram: 8B 88 34 00 20 AC
002+23:13:47.102 D 118: [emsesp] Last Tx read successful
002+23:13:47.106 N 119: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorWW(0x34), data: 28 02 3B 02 3B 23 00 00 03 00 00 98 3C 00 0E 6C 00 80 00
002+23:13:47.106 D 120: [emsesp] Received UBAMonitorWW
002+23:13:47.118 D 121: [mqtt] Publishing topic ems-esp/boiler_data (#36848, attempt #1, size 519, pid 1)
002+23:13:47.319 D 122: [mqtt] Publishing topic ems-esp/boiler_data_ww (#36849, attempt #1, size 429, pid 1)
002+23:13:49.537 D 123: [mqtt] Publishing topic ems-esp/dallassensor_data (#36850, attempt #1, size 50, pid 1)
002+23:13:50.250 N 124: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 09 0C 1E 37 33 02 01 10 FF FE
002+23:13:50.250 D 125: [emsesp] Received RCTime
002+23:13:50.341 D 126: [mqtt] Publishing topic ems-esp/thermostat_data (#36851, attempt #1, size 263, pid 1)
002+23:13:53.460 N 127: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 DE 00 00 00 02 40 C0 80 00 02 3B 80 00 00 00 FF 00 00 00 CB 00 00 00
002+23:13:53.460 D 128: [emsesp] Received UBAMonitorFast
002+23:13:53.680 N 129: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
002+23:13:53.680 D 130: [emsesp] Received MC10Status
002+23:13:54.003 N 131: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 3B 02 3B 23 00 00 03 00 00 98 3C 00 0E 6C 00 80 00
002+23:13:54.003 D 132: [emsesp] Received UBAMonitorWW
002+23:13:54.563 D 133: [mqtt] Publishing topic ems-esp/dallassensor_data (#36852, attempt #1, size 50, pid 1)
002+23:13:59.591 D 134: [mqtt] Publishing topic ems-esp/dallassensor_data (#36853, attempt #1, size 50, pid 1)
002+23:14:02.636 N 135: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 01 7B 01 19 (offset 13)
002+23:14:02.636 D 136: [emsesp] Received RC300Monitor

```

MichaelDvP, 2020-10-30:
The onetime-function of the boiler works only if water temperature is more than 5 degree below setpoint. 
You have set temperatue 40°C and current temperature 42,5°C, the boiler will not heat on onetime-command. Also you have ww activated on, than the boiler will heat automatically if the temperature is 5° below setpoint. Onetime is a function to heat up if automatic is off. 

To get command confirmations as `Setting warm water OneTime loading on` you can also set `log info`.

glitter-ball, 2020-10-30:
OK. This isn't the behaviour I see using the thermostat. 

When I select one-time cylinder charging on the thermostat, it will fire the boiler anytime the water temperature is below 55ºC (60ºC setpoint minus 5ºC hysteresis) - regardless of the time program's temperature (e.g. 40ºC).

When I did the tests this morning, I could not get the boiler to fire from the telnet console even with the water below 55ºC but as soon as I selected one-time cylinder charging from the thermostat, the boiler started.

glitter-ball, 2020-10-30:
Did more tests. Cylinder temp 54.6C.

HW time program ‘reduced’ - 40C setpoint. Telnet call would not start boiler.

HW time program ‘off’ - 15C setpoint. Telnet call would not start boiler.

HW time program back to ‘reduced’ - 40C setpoint. Did one time cylinder charge from thermostat. Boiler fires and heats water until 60C setpoint reached.

glitter-ball, 2020-10-30:
Maybe there’s another one time charging command which mimics the thermostat HMI? Weird...

MichaelDvP, 2020-10-30:
Set `watch on` and start the onetime-heating on your thermostat. Then we can see which command is send.

glitter-ball, 2020-10-30:
Ok. Interesting - HA is showing onetime ww active when I trigger it from the thermostat. I’ll collect some logs later when the cylinder has cooled.

glitter-ball, 2020-10-30:
Is the wwonetime telegram being sent to the boiler? The logic to drive it is all in the thermostat in my system (enable/disable,  one time active period etc.) I don’t think my boiler knows anything about it. The thermostat runs the logic and then simply tells the boiler to make DHW and the setpoint. Again, hopefully that will show up in the logs.

glitter-ball, 2020-10-30:
Logs here...
```

ems-esp:/$ su
Password:
003+03:04:41.993 N 0: [shell] su session opened on console
ems-esp:/# watch on
Watching incoming telegrams, displayed in decoded format
003+03:05:02.233 N 1: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 94 02 00 (offset 13)
003+03:05:05.080 N 2: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 12 DA 9F
003+03:05:05.380 N 3: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 4B 64 00 04 F6 0A 01 03 64 50 04 00 00 00 00 00 00 00 00 00 00 00 00 28 00 3C
003+03:05:05.580 N 4: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 3C 00 (offset 25)
003+03:05:05.930 N 5: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 00 8C 80 00 80 00 00 00 00 00 00 34 14 02 6A B3 00 00 00 01 D2 71 00 25 A7 80 00
003+03:05:06.179 N 6: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 80 00 (offset 25)
003+03:05:06.415 N 7: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 3C FB 00 28 00 00 46 00 FF FF 00
003+03:05:06.742 N 8: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 BB 21 00 00 00 00 25 00 94 01 01 03 00 94 02 00 00 00 11 01 00 07 4D 00
003+03:05:07.015 N 9: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 C6 04 FF 01 02 (offset 25)
003+03:05:07.332 N 10: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 1E 02 2A FF FF 00 25 01 E1 20 01 0F 05 00 00 02 21
003+03:05:07.577 N 11: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
003+03:05:07.921 N 12: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 28
003+03:05:08.418 N 13: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 01
003+03:05:08.472 N 14: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 28 (offset 3)
003+03:05:08.976 N 15: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 65 00 00 00 02 40 C0 80 00 02 16 80 00 00 00 FF 00 00 00 CB 00 00 00
003+03:05:09.271 N 16: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
003+03:05:09.494 N 17: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 16 02 16 01 00 00 03 00 00 98 42 00 0E 6D 00 80 00
003+03:05:18.993 N 18: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 16 02 16 01 00 00 03 00 00 98 42 00 0E 6D 00 80 00
003+03:05:19.187 N 19: [emsesp] Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00
003+03:05:19.425 N 20: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 65 00 00 00 02 40 C0 80 00 02 16 80 00 00 00 FF 00 00 00 CB 00 00 00
003+03:05:19.702 N 21: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlow(0x19), data: 00 8C 80 00 80 00 00 00 00 00 00 34 14 02 6A B3 00 00 00 01 D2 71 00 25 A7 80 00
003+03:05:20.000 N 22: [emsesp] Boiler(0x08) -> All(0x00), UBAMaintenanceStatus(0x1C), data: 93 09 05 1D 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
003+03:05:20.221 N 23: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
003+03:05:28.945 N 24: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
003+03:05:29.218 N 25: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 16 02 16 01 00 00 03 00 00 98 42 00 0E 6D 00 80 00
003+03:05:29.450 N 26: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 64 00 00 00 02 40 C0 80 00 02 16 80 00 00 00 FF 00 00 00 CB 00 00 00
003+03:05:39.000 N 27: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 64 00 00 00 02 40 C0 80 00 02 16 80 00 00 00 FF 00 00 00 CB 00 00 00
003+03:05:39.270 N 28: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
003+03:05:39.518 N 29: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 16 02 16 01 00 00 03 00 00 98 42 00 0E 6D 00 80 00
003+03:05:41.825 N 30: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
003+03:05:41.877 N 31: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
003+03:05:42.072 N 32: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 09 07
003+03:05:48.968 N 33: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 16 02 16 01 00 00 03 00 00 98 42 00 0E 6D 00 80 00
003+03:05:49.174 N 34: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 64 00 00 00 02 40 C0 80 00 02 16 80 00 00 00 FF 00 00 00 CB 00 00 00
003+03:05:49.445 N 35: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
003+03:05:55.935 N 36: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 09 10 1E 2F 3A 02 01 10 FF FE
003+03:05:58.406 N 37: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 93 02 01 (offset 13)
003+03:05:58.970 N 38: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
003+03:05:59.293 N 39: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 02 16 02 16 01 00 00 03 00 00 98 42 00 0E 6D 00 80 00
[... cut ...]
003+03:06:19.727 N 72: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 01 B2 01 B7 80 00 00 80 00 80 00 80 00 00
003+03:06:20.049 N 73: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 3C 02 16 02 16 09 00 00 03 00 00 98 43 00 0E 6E 00 80 00
003+03:06:20.343 N 74: [emsesp] Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00
003+03:06:20.556 N 75: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 55 01 9F 64 48 0A 12 65 C0 80 00 02 16 80 00 01 0A FF 00 00 00 C9 00 00 00
ems-esp:/# watch off
Watching telegrams is off
ems-esp:/#
```

glitter-ball, 2020-10-30:
```
ems-esp:/# watch on 31d
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0x31D
```

*** Start WW onetime on thermostat ***

```
003+04:16:01.818 N 122: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 02 (offset 2)
003+04:16:03.795 N 123: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 01
003+04:16:44.168 N 124: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 01 00 02 07

```
*** Cancel WW onetime on thermostat ***

```
003+04:16:52.893 N 125: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00
003+04:16:54.494 N 126: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 09 (offset 2)
ems-esp:/# watch off
Watching telegrams is off
ems-esp:/#

```

MichaelDvP, 2020-10-31:
Hm, the command sets extra in pos 0, but pos 2 keeps 9=clock-mode, low temp. Maybe pos 2 is also writeable. Try to 
`send telegram "0B 10 FF 02 02 1D 02"`
if this does not work alone, set also the pos 0 to 1 (as before): `send telegram "0B 10 FF 00 02 1D 01"` 

or set both positions in one telegram `send telegram "0B 10 FF 00 02 1D 01 00 02"`

to cancel onetime  first try with only pos 0 `send telegram "0B 10 FF 00 02 1D 00"` and check if pos 2 changes.

glitter-ball, 2020-10-31:
@MichaelDvP - thanks for these ideas. Sadly, not much luck. None of the telegrams would start wwOneTime and the cancel one failed too!
```
ems-esp:/# watch on 10
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x10
003+21:14:01.847 N 1: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0A 0A 1F 37 3A 05 00 10 FF FE
003+21:14:07.380 N 2: [emsesp] Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
003+21:14:07.574 N 3: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 01 4E 00 38 (offset 13)
003+21:14:08.242 N 4: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 BA 21 00 00 00 00 25 01 4E 01 01 03 01 4E 00 38 00 00 11 01 00 07 49 00
003+21:14:08.491 N 5: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 C6 04 FF 01 02 (offset 25)
003+21:14:08.808 N 6: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 1E 02 2A FF FF 00 25 01 E1 20 01 0F 05 00 00 02 21
003+21:14:08.905 N 7: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
003+21:14:08.955 N 8: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
003+21:14:09.150 N 9: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 08 07
003+21:14:09.477 N 10: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
003+21:14:09.822 N 11: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 28
003+21:14:44.300 N 12: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 01
003+21:14:44.346 N 13: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 0F (offset 3)
003+21:15:02.846 N 14: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0A 0A 1F 38 3B 05 00 10 FF FE
003+21:15:06.828 N 15: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 01 4D 00 39 (offset 13)
003+21:15:08.040 N 16: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 BA 21 00 00 00 00 25 01 4D 01 01 03 01 4D 00 39 00 00 11 01 00 07 49 00
003+21:15:08.322 N 17: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 C6 04 FF 01 02 (offset 25)
003+21:15:08.607 N 18: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 1E 02 2A FF FF 00 25 01 E1 20 01 0F 05 00 00 02 21
003+21:15:08.932 N 19: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
003+21:15:08.999 N 20: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
003+21:15:09.050 N 21: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
003+21:15:09.286 N 22: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 BA 21 00 00 00 00 25 01 4D 01 01 03 01 4D 00 39 00 00 11 01 00 07 49 00
003+21:15:09.644 N 23: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 28
003+21:15:09.727 N 24: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 C6 04 FF 01 02 (offset 25)
003+21:15:09.926 N 25: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 08 07
ems-esp:/# send telegram "0B 10 FF 02 02 1D 02"
003+21:15:26.823 N 26: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWmode2(0x31D), data: 00 00 02 07
003+21:16:03.765 N 27: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0A 0A 1F 3A 00 05 00 10 FF FE
003+21:16:06.355 N 28: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 01 4C 00 3A (offset 13)
003+21:16:08.063 N 29: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 BA 21 00 00 00 00 25 01 4C 01 01 03 01 4C 00 3A 00 00 11 01 00 07 49 00
003+21:16:08.416 N 30: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 C6 04 FF 01 02 (offset 25)
003+21:16:08.703 N 31: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 1E 02 2A FF FF 00 25 01 E1 20 01 0F 05 00 00 02 21
003+21:16:08.774 N 32: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
003+21:16:08.824 N 33: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
003+21:16:09.020 N 34: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 08 07
003+21:16:09.603 N 35: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
003+21:16:09.899 N 36: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 28
ems-esp:/# send telegram "0B 10 FF 00 02 1D 01"
003+21:16:17.354 N 37: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWmode2(0x31D), data: 00 00 08 07
003+21:16:19.937 N 38: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x2F5), data: 00 00 00 00 00 00 00 00 (offset 12)
003+21:16:44.369 N 39: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 01
003+21:16:44.423 N 40: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 0F (offset 3)
ems-esp:/# send telegram "0B 10 FF 00 02 1D 01 00 02"
003+21:16:54.248 N 41: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWmode2(0x31D), data: 01 00 02 07
003+21:16:55.038 N 42: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBADevices(0x07), data: 0F
003+21:16:55.061 N 43: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
003+21:17:09.697 N 54: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 28
003+21:17:09.820 N 55: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 C6 04 FF 01 02 (offset 25)
003+21:17:10.019 N 56: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 08 07
ems-esp:/# send telegram "0B 10 FF 00 02 1D 00"
003+21:17:23.547 N 57: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWmode2(0x31D), data: 00 00 08 07
ems-esp:/# watch off
Watching telegrams is off
ems-esp:/#
```

glitter-ball, 2020-11-08:
I wonder whether this is going to be possible?

If we had someone with the Bosch internet gateway (MB LAN2 or MB LANi or KM200), then I wonder whether you can set wwonetime through that and capture telegrams? You'd like to think there is a way for the CW400 / Sense II to receive this request and start wwonetime.

MichaelDvP, 2020-11-09:
Yes, capturing the command from a working gateway is a simple way, but i think very few people using a geteway and ems-esp. 

I'm not sure that the last commands are wrong. Starting onetime from thermostat sets `RC300WWmode2(0x31D), data: 01 00 02 07` first byte to 1, third to 2.  Thats exactly what is set here:
```
ems-esp:/# send telegram "0B 10 FF 00 02 1D 01 00 02"
003+21:16:54.248 N 41: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWmode2(0x31D), data: 01 00 02 07
```
but after 15 sec the thermostat resets this telegram to 
`003+21:17:10.019 N 56: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 08 07`
Maybe the water was warm enough. 

After this the cancel command will do nothing, because there is nothing to cancel. 
```
ems-esp:/# send telegram "0B 10 FF 00 02 1D 00"
003+21:17:23.547 N 57: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWmode2(0x31D), data: 00 00 08 07
``` 

I think it's worth a second try. Start onetime on the thermostat, after the burner starts try the cancel command 
`send telegram "0B 10 FF 00 02 1D 00"` 
If this works you can try to start again by 
`send telegram "0B 10 FF 00 02 1D 01 00 02"`

glitter-ball, 2020-11-09:
@MichaelDvP - thanks. I've had another attempt.

So, cylinder at 45.2ºC, timed programme in reduced mode (set at 40ºC) but normal DHW temperature is 60ºC. Selected one-time cylinder charging from the thermostat. 3-way valve moved to HW and burner lit. Tap and flame symbols shown on LCD.

Issued the cancel telegram above and left for approx. 2 minutes. Nothing happened. Re-issued cancel telegram. Nothing happened.

Cancelled one-time cylinder charging from thermostat. 3-way valve moved to CH.

Sent start telegram above. Nothing happened. Sent again approx 40 seconds later. Nothing happened.

Frustrating! Trace below...
```
ems-esp:/# watch on 10
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x10
002+23:44:40.754 N 1: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 18 64 64
002+23:44:40.804 N 2: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
002+23:44:41.041 N 3: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 BA 21 25 18 00 25 1E 01 26 03 03 01 01 26 00 42 00 00 11 01 03 07 44 00
002+23:44:41.427 N 4: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 06 04 FF F7 00 00 FF 64 4B 00 86 05 FF 01 02 (offset 25)
002+23:44:41.625 N 5: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 09 07
002+23:45:10.718 N 6: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x06), data: 14 0B 11 09 24 3A 00 00 10 FF FE
002+23:45:10.909 N 7: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x2F5), data: FF (offset 11)
002+23:45:11.622 N 8: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 01
002+23:45:11.968 N 9: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 BA 21 25 18 00 25 1E 01 26 03 03 01 01 26 00 42 00 00 11 01 03 07 44 00
002+23:45:12.322 N 10: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 06 04 FF F7 00 00 FF 64 4B 00 86 05 FF 01 02 (offset 25)
002+23:45:12.631 N 11: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 1E 02 2A FF FF 00 25 01 E1 20 01 0F 05 00 00 02 21
002+23:45:12.698 N 12: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 3C (offset 3)
002+23:45:12.955 N 13: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 FF 00 00 00 00 00 00 00 00
002+23:45:13.245 N 14: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 28
002+23:45:13.326 N 15: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 02 (offset 2)
002+23:45:14.614 N 16: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 01 25 00 43 (offset 13)
002+23:45:18.761 N 17: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 20 (offset 2)
002+23:45:19.834 N 18: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 18 64 00
ems-esp:/# send telegram "0B 10 FF 00 02 1D 00"
002+23:45:22.955 N 19: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWmode2(0x31D), data: 00 00 02 07
002+23:45:41.009 N 20: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 18 64 00
002+23:45:41.059 N 21: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
002+23:45:41.255 N 22: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 01 00 02 07
002+23:45:54.076 N 23: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 01
002+23:45:54.106 N 24: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 3C (offset 3)
002+23:45:55.301 N 25: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBADevices(0x07), data: 0F
002+23:45:55.321 N 26: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00
002+23:46:10.884 N 27: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x06), data: 14 0B 11 09 25 3A 00 00 10 FF FE
002+23:46:11.550 N 28: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 BA 20 25 18 00 25 1E 01 25 03 03 01 01 25 00 43 00 00 11 01 03 07 44 00
002+23:46:11.847 N 29: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 06 04 FF F7 00 00 FF 64 4B 00 86 05 FF 01 02 (offset 25)
002+23:46:12.117 N 30: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 1E 02 2A FF FF 00 25 01 E1 20 01 0F 05 00 00 02 21
002+23:46:12.434 N 31: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 FF 00 00 00 00 00 00 00 00
002+23:46:12.674 N 32: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 28
002+23:46:14.140 N 33: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 01 24 00 44 (offset 13)
002+23:46:40.727 N 34: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 18 64 00
002+23:46:40.786 N 35: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
002+23:46:41.021 N 36: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 BA 20 25 18 00 25 1E 01 24 03 03 01 01 24 00 44 00 00 11 01 03 07 48 00
002+23:46:41.432 N 37: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 06 04 FF F7 00 00 FF 64 4B 00 86 05 FF 01 02 (offset 25)
002+23:46:41.630 N 38: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 01 00 02 07
002+23:47:10.795 N 39: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x06), data: 14 0B 11 09 26 3A 00 00 10 FF FE
[... cut ...]
002+23:48:41.629 N 81: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 06 04 FF F7 00 00 FF 64 4B 00 86 05 FF 01 02 (offset 25)
002+23:48:41.828 N 82: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 09 07
002+23:48:52.083 N 83: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: FF EE (offset 27)
002+23:48:55.368 N 84: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBADevices(0x07), data: 0F
002+23:48:55.392 N 85: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00
ems-esp:/# watch off
Watching telegrams is off
ems-esp:/#
```

glitter-ball, 2020-11-09:
mqtt publish error buried in there too at 23:48:16.968, in case that's relevant.

MichaelDvP, 2020-11-10:
Sad, the thermostat always reverts back the settings. Then this is only a monitor message and not for commands. 

There are 2F5 messages after the try, this is RC300WWmode (naming is fixed in dev). Can you `read 10 2F5` before and after starting the onetime. 

Or try direct writing to offset 11 as in  N7.
Start ww onetime:
`send telegram "0B 10 FF 0B 01 F5 FF"`
Cancel ww onetime:
`send telegram "0B 10 FF 0B 01 F5 00"`

glitter-ball, 2020-11-12:
@MichaelDvP  - I owe you a beer (or other drink)! You've cracked it. These telegrams work...

Start ww onetime:
send telegram "0B 10 FF 0B 01 F5 FF"
Cancel ww onetime:
send telegram "0B 10 FF 0B 01 F5 00"

How did you work them out? I see the 2F5 telegram above.
```
ems-esp:/# send telegram "0B 10 FF 0B 01 F5 FF"
005+21:08:06.751 N 2: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 FF 00 00 00 00 00 00 00 00
ems-esp:/# send telegram "0B 10 FF 0B 01 F5 00"
005+21:08:46.206 N 3: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
ems-esp:/# send telegram "0B 10 FF 0B 01 F5 FF"
005+21:11:42.553 N 4: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 FF 00 00 00 00 00 00 00 00
ems-esp:/#
```

There is one odd thing though. I get Warm water/DHW active, Boiler Warm water charging but no Boiler Warm water one time charging. It changed Off / Unknown / Off briefly.  Might be an HA issue. I did get Boiler Warm water one time charging when I triggered from thermostat.

Anyway - big step forward!

glitter-ball, 2020-11-12:
The Unknown state was me restarting HA - please ignore. I'll do some more tests on the onetime charging indication.

MichaelDvP, 2020-11-14:
The `wwonetime` command is now in the latest [dev](https://github.com/proddy/EMS-ESP/releases/tag/dev). Also some heatcurve parameters. Can you check please?

jakubjakubik, 2020-11-24:
Following up on #630 - I've tested sending the raw telegrams and this indeed works. The boiler still does not report it's own wwonetime state as enabled during this time, but the water heating cycle starts. I'll test the dev firmware later on and report back.

glitter-ball, 2020-11-24:
I should also do more tests.

jakubjakubik, 2020-11-24:
Noticed one more thing - this time, even though the one time charging was triggered by the thermostat, not only did the wwonetime sensor on the boiler remain off, if I sent the api command to trigger it on it didn't react.

MichaelDvP, 2020-11-25:
@jakubjakubik  Try again with new dev-build, api and console commands were broken.
As far as i see the onetime in thermostat do not trigger the boiler-onetime. It shows up in thermostat wwextra and sets the boiler wwtemp higher for some time.

Renifer1980, 2021-01-07:
> I wonder whether this is going to be possible?
>
> If we had someone with the Bosch internet gateway (MB LAN2 or MB LANi or KM200), then I wonder whether you can set wwonetime through that and capture [...]

I have Bosch Condens 9000 with Bosch MB LANi connected , can do some testing for you but will need small introduction how to proceed.

glitter-ball, 2021-01-09:
@Renifer1980 - thanks for your offer. I still need to get wwonetime built in HA and test it. Back on 12 Nov I got the boiler to fire up but some of the status indications didn't look quite right.

At some point, it would be good to capture the traffic in and out of the CW400 when one-time cylinder charging is started and stopped.

appliko, 2021-02-17:
I'm very interested on that too
I have a Logomax Plus GB152. I can ship a KM200 for tests if you need.
Not working on my Buderus. Using EMS-ESP with Domoticz