emsesp/EMS-ESP #590: Upgrade from 2.0.1 to 2.1.0 doesn't work with Upload firmware

rvanderven, 2020-11-01:
I am running 2.0.1 on a WemosD1 mini and am trying to upgrade to 2.1.0.
I tried Upload Firmware and selected the EMS-ESP-2_1_0-esp8266.bin file, but nothing happens.
Also tries the espota.py script but that results in an error 
** File "espota.py", line 6
**  <!DOCTYPE html>
**    ^
** SyntaxError: invalid syntax

Any idea how to solve this?

proddy, 2020-11-01:
did the upload via the web UI (2.0.1) not work? It can take around 2-3 minutes to upload, unpack and restart.

The espota.py script error - are you on Windows or Unix?

rvanderven, 2020-11-01:
> did the upload via the web UI (2.0.1) not work? It can take around 2-3 minutes to upload, unpack and restart.
> 
> The espota.py script error - are you on Windows or Unix?
I did wait for more than 5 minutes (using Chrome)
I am a MacOS user.
Just tried again: what am I supposed to see when the upgrade succeeded?

bbqkees, 2020-11-01:
If it succeeded you can refresh the page and it will show the new version number:
[image]

rvanderven, 2020-11-01:
The dashboard still shows 2.1.0

rvanderven, 2020-11-01:
I did a reboot as well, prior trying the 2nd time

rvanderven, 2020-11-01:
I managed to upgrade via espota.py. 
At 1st this didn't work, but I found out that I (stupily) saved the script via "save link as" which (of course) didn't download the python script itself, but the whole page which you see when you click on the link.
So I copied the python code and pasted it into the local espota.py file and managed to upload the code into my esp board.
This means the espota.py way works.
This of course still doesn't explain why the web based method didn't work.

closed 2020-11-01 by rvanderven

proddy, 2020-11-02:
I'll look into the Web upload again. We did have some issues prior to 2.0.1 with that feature.

Note the python scripts are also available in the /scripts folder of the EMS-ESP repo. https://emsesp.github.io/docs/#/Uploading-firmware?id=wirelessly-over-the-air-ota

rvanderven, 2020-11-02:
Hi,

That is where I found the code, but I couldn't find a download button and I am not that familiar with GitHub.
I thought I could prevent copy/paste by using the save as but that was a wrong thought.
So I ended up in selecting the contents and paste this in my local file.

I am glad that you have found the issues in 2.0.1 wrt the web upload. Might be useful to mention this on the support page and direct users to using the OTA option.

Nice product by the way 👍

Grt

Richard