emsesp/EMS-ESP #591: potential bug in MQTT output?

ReneHonig, 2020-11-02:
**Bug description**
I am running v2.1.1b0 and it looks like there is a mix-up with the MQTT output related to "heatingactive"

**Steps to reproduce**
- send an MQTT command to (de)activate the heating (using 'heatingactivated') or (de)activate the heating on the console. 
- observe the MQTT output. It looks like neither ems-esp/boiler_data {"heatingactive"} nor ems-esp/heating_active follow the change in setting.

**Expected behavior**
- ems-esp/boiler_data {"heatingactive"} to follow the current heating state
- ems-esp/heating_active to follow whether the heating is enabled

**Device information**
{
  "System": {
    "version": "2.1.1b0",
    "uptime": "000+00:27:43.558",
    "freemem": 56,
    "fragmem": 18
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 20,
    "publish_time_thermostat": 0,
    "publish_time_solar": 0,
    "publish_time_mixer": 0,
    "publish_time_other": 0,
    "publish_time_sensor": 0,
    "mqtt_format": 3,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 1,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 8543,
    "#read requests sent": 13,
    "#write requests sent": 0,
    "#incomplete telegrams": 54,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Nefit GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.07)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Controller",
      "name": "BC10 (DeviceID:0x09, ProductID:190, Version:01.04)",
      "handlers": ""
    },
    {
      "type": "Connect",
      "name": "OpenTherm Converter (DeviceID:0x18, ProductID:171, Version:02.00)",
      "handlers": ""
    }
  ]
}

closed 2020-11-03 by ReneHonig

proddy, 2020-11-03:
I see you closed this. Was it due to the mix up of the wording? There is indeed `heatingactive` which is EMS-ESP's own key/value and generated based on system codes and `heatingactivated` which comes direct from the telegram. They mean different things. Perhaps we should rename one of them.

ReneHonig, 2020-11-03:
It was indeed a mix up of the various 'heatingactivated' MQTT key/value pairs on my side. I don't think a rename is necessary, maybe we can point it out in the documentation. From what I understand:
- both /ems-esp/boiler_data {heating_active} and ems-esp/heating_active indicate that the burner is currently on for the purpose of heating;
- both /ems-esp/boiler_data {heatingActivated} indicates the boiler is enabled, so can heat on request.

Correct? 
And many thanks for all excellent improvements!