emsesp/EMS-ESP #593: gateway keeps resetting after HA update to 0.117.2 - no data in web interface and mqtt

lsw271, 2020-11-03:
**Bug description**
Today I did a HA update to version 0.117.2 (from 0.115.2). Data from the gateway stopped being sent by mqtt after updating to HA 117.2 - the system update time is the same as the last data sent. EMS-ESP in version v2.1.0b6. The mqtt broker works fine (normally receives data in the mqtt explorer for shelly for example). After two hours, I noticed the lack of current data and tried to reset the gateway, but to no avail. I can access via www while the device data is displayed briefly or not at all - even though the devices (boiler, cw400) are listed.
I am also unable to connect to telnet because the gateway keeps restarting. I don't know how to check any logs since I can't get to telnet. It's a bit weird but it looks like the HA update broke the gateway?

**Steps to reproduce**
update HA to 0.117.2

**Expected behavior**
working gateway

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api?device=system&cmd=report.*
http://192.168.5.31/api?device=system&cmd=report  return "Invalid cmd"

**Additional context**
One of error messages: NetworkError when attempting to fetch resource. I haven't tried going back to 0.115.2 yet - I'll try tomorrow if you have no idea and see if it helps but looks like a gateway issue. Only a strange coincidence of hours (HA update and no data)

proddy, 2020-11-03:
Sorry to hear that. So if I'm not mistaken you have been using 2.1.0b6 for a while now (this build is many weeks old btw) and it works with HA 0.115 but after the upgrade to HA 0.117.2 the EMS-ESP starts crashing and restarting? Other users have experienced similar crashes with HA which is baffling me.

Questions: 
- Did you take the firmware binary from GitHub or compile it yourself?
- Are you using the 'Home Assistant' setting as MQTT Format?

I'm on HA 0.117.2 and the latest dev build 2.1.1b1 and its looking stable. I wouldn't downgrade HA back to 0.115. 

I'll also check the Home Assistant release notes for any breaking changes with MQTT.

lsw271, 2020-11-03:
> Sorry to hear that. So if I'm not mistaken you have been using 2.1.0b6 for a while now (this build is many weeks old btw) and it works with HA 0.115 but after the upgrade to HA 0.117.2 the EMS-ESP [...]

Yes, it works great for few weeks. I had home renovation last 3weeks so I am not up to date with the changes  ;)

> 
> Questions:
> 
>     * Did you take the firmware binary from GitHub or compile it yourself?
binary from github

> 
>     * Are you using the 'Home Assistant' setting as MQTT Format?
yes

> 
> 
> I'm on HA 0.117.2 and the latest dev build 2.1.1b1 and its looking stable. I wouldn't downgrade HA back to 0.115.
> 
> I'll also check the Home Assistant release notes for any breaking changes with MQTT.

after i've added this issue I disconnected the jack gateway cable from the boiler and connect it 30minutes ago (ems-esp rested for more than 2 hours without connection) and... everythink works! its really wird... sorry for this issue but it really looked like gateway was broken after HA update (the same time stop data and update)

Tommorow I will update to latest ems-esp version

closed 2020-11-03 by lsw271

susisstrolch, 2020-11-03:
I couldn‘t see any evidence to HA. Problems are the same with and without HA. (I‘m using  FHEM as MQTT server since aeons.)
Problems are also independent  of MQTT setting (HA vs. structured). 
Only difference was doing a factory reset and falling back to b12...